zephyrproject-rtos/zephyr #22722: posix: redefinition of symbols while porting zeromq to zephyr

jensschroer, 2020-02-11:
**Describe the bug**
I am trying to port zeromq (libzmq) to Zephyr, using the posix subsystem.
I have patched libzmq to compile it as a module in Zephyr and created a small 'Hello World' app.
The compilation fails with several redefined symbols in the posix system.

**To Reproduce**
The easiest way to reproduce the issue is to clone my sample project.
I am using west for building the project. 
Assuming a newly initialized zephyr installation (`west init` followed by `west update` etc)

Steps to reproduce the behavior:
1. cd .. (starting in the directory above `zephyr` )
2. My sample app `git clone https://github.com/jensschroer/zmqapp.git`
3. My patched version of libzmq `git clone https://github.com/jensschroer/libzmq.git`
4. cd zmqapp; west build -b=frdm_k64f -o=-v
5. See error

**Expected behavior**
The build should continue without redefined symbols.

**Screenshots or console output**
As example, the following is the error output from one file that files, but also including `-H` output
```
. /home/schroer/code/blixt/blixt-breaker/bug-test/libzmq/src/precompiled.hpp
.. modules/libzmq/platform.hpp
.. /home/schroer/code/blixt/blixt-breaker/bug-test/libzmq/src/../include/zmq.h
... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/errno.h
.... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/errno.h
..... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/reent.h
...... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/_ansi.h
....... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/newlib.h
........ /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/_newlib_version.h
....... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/config.h
........ /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/machine/ieeefp.h
........ /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/features.h
...... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/lib/gcc/arm-zephyr-eabi/9.2.0/include/stddef.h
...... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_types.h
....... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/machine/_types.h
........ /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/machine/_default_types.h
....... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/lock.h
....... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/lib/gcc/arm-zephyr-eabi/9.2.0/include/stddef.h
... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/lib/gcc/arm-zephyr-eabi/9.2.0/include/stddef.h
... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/stdio.h
.... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/_ansi.h
.... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/cdefs.h
..... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/lib/gcc/arm-zephyr-eabi/9.2.0/include/stddef.h
.... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/lib/gcc/arm-zephyr-eabi/9.2.0/include/stddef.h
.... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/lib/gcc/arm-zephyr-eabi/9.2.0/include/stdarg.h
.... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/types.h
..... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/lib/gcc/arm-zephyr-eabi/9.2.0/include/stddef.h
..... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_stdint.h
..... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/machine/endian.h
...... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/machine/_endian.h
..... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/select.h
...... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_sigset.h
...... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_timeval.h
...... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/timespec.h
....... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_timespec.h
..... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/_pthreadtypes.h
...... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/sched.h
..... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/machine/types.h
.... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/stdio.h
... /home/schroer/code/blixt/blixt-breaker/bug-test/zephyr/lib/libc/newlib/include/stdint.h
.... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/lib/gcc/arm-zephyr-eabi/9.2.0/include/stdint.h
..... /home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/lib/gcc/arm-zephyr-eabi/9.2.0/include/stdint-gcc.h
.. /home/schroer/code/blixt/blixt-breaker/bug-test/libzmq/src/zmq_draft.h
. /home/schroer/code/blixt/blixt-breaker/bug-test/libzmq/src/macros.hpp
. /home/schroer/code/blixt/blixt-breaker/bug-test/zephyr/include/posix/unistd.h
.. /home/schroer/code/blixt/blixt-breaker/bug-test/zephyr/include/posix/posix_types.h
... /home/schroer/code/blixt/blixt-breaker/bug-test/zephyr/include/kernel.h
[... 549 lines cut ...]
/home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/include/c++/9.2.0/cstdio
/home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/include/c++/9.2.0/cstring
/home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/include/c++/9.2.0/cwctype
/home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/limits.h
/home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/machine/_endian.h
/home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/arm-zephyr-eabi/sys-include/sys/string.h
/home/schroer/code/blixt/zephyr-sdk/0.11.1/arm-zephyr-eabi/lib/gcc/arm-zephyr-eabi/9.2.0/include-fixed/syslimits.h
zephyr/include/generated/devicetree_fixups.h
zephyr/include/generated/devicetree_unfixed.h
zephyr/include/generated/kobj-types-enum.h
At global scope:
cc1plus: warning: unrecognized command line option '-Wno-tautological-constant-compare'

```

**Environment (please complete the following information):**
 - OS: Ubuntu 19.04
 - Toolchain Zephyr SDK 0.11.1
 - Zephyr commit 9dbc5eeb3b398fce2960d113f155c6f6dd0ea206

**Additional context**

I include the library libzmq in the app by using `ZEPHYR_EXTRA_MODULES`.
In order for west / cmake to add the zephyr posix include directories to the libzmq include paths,I had to explicitly add a `target_link_libraries` dependency on the `posix_subsys` in `libzmq/CMakefile.txt` (see libzmq/CMakefile.txt, search for `posix_subsys`).
I just wanted to point this out, as I am not sure if this is the correct way or if it even might contribute to this problem.

chrta, 2020-02-13:
I had also a similar problem when i tried out to port asio (asio is already ported for esp32 in esp-idf).
I would also like to structure the posix headers etc in a way so it is easier to port external libs.

pfalcon, 2020-02-13:
As someone who ported more than project to Zephyr POSIX subsystem, and made numerous changes to that subsys to accommodate such porting efforts, I certainly can understand there're many more changes to make before Zephyr will be an equal member among POSIX systems (if that ever happens).

ZMQ is definitely a hard case, because it's C++ and a large project overall.

Something which caught my attention in the original description is:

> I am using west for building the project.

There's no "west" in POSIX, so you're already not really using POSIX. I'd suggest to stick to POSIX tools when trying to work with POSIX subsystems, like Zephyr's one. At least, that would be my suggestion - please try to reproduce the issue without involving adhoc tools like west.

jensschroer, 2020-02-13:
> I had also a similar problem when i tried out to port asio (asio is already ported for esp32 in esp-idf).
> I would also like to structure the posix headers etc in a way so it is easier to port external libs.

Do you have a link to the PR? Maybe that could give me some additional ideas on what I need to do.

jensschroer, 2020-02-13:
Finally found the solution to this bug. Due to my own doing the CMakeLists.txt in libzmq defined -D_GNU_SOURCE which caused the redefines. 
I will close this. Thank you for the comments!

closed 2020-02-13 by jensschroer

pablogs9, 2020-07-30:
Hi, is there any official support for ZeroMQ on Zephyr or this @jensschroer is the only one solution?

jensschroer, 2020-07-30:
@pablogs9 As far as I am aware there isn't any official support for ZeroMQ on Zephyr. I discontinued my efforts on the direct port, as the library turned out to be too big for the MCU I was targeting. I am still interested in ZeroMQ on Zephyr, so I might start another try if time permits.
Which communication patterns do you need?

pablogs9, 2020-07-30:
Thanks for the answer @jensschroer, I'm looking a **broker-less** pub-sub (ideally also req-rep) library that can be embedded in MCU like STM32F4 or so (~150 KB RAM, ~1 MB Flash). 

I'm targeting systems with RTOS and some POSIX compliance (by now: Zephyr RTOS, Nuttx and FreeRTOS + POSIX labs extensions). And ideally, it should be C++ free... 

I'm checking right now this [nanomsg ng](https://github.com/nanomsg/nng). Are you aware of any other solution?

jensschroer, 2020-07-30:
@pablogs9 I have also looked at nanomsg-ng and tried to port it, after I got stuck with ZeroMQ. At the time I could not port nanomsg-ng as some systemcalls were missing. From what I remember eventfd was the main one - which has been added recently to Zephyr.
However, the reason why I decided against pursuing nanomsg-ng was the limited user and application base. 

I was considering to implement a subset of ZeroMQ in C in Zephyr, but have not found the time for it.
My communication patterns are similar to the ones you describe, but I would also like to add push and pull.

pablogs9, 2020-07-30:
I think that I'm going to try nanomsg-ng. I'll keep you updated if success.

lakshjaisinghani, 2024-04-19:
Any update on your progress @pablogs9?

pablogs9, 2024-04-19:
@lakshjaisinghani we developed a DDS embedded solution, that is FreeRTOS compatible, so I guess that is easy to port to Zephyr: https://safe-dds.docs.eprosima.com/