zephyrproject-rtos/zephyr #22731: Improve docker CI documentation

Lindenk, 2020-02-11:
**Is your enhancement proposal related to a problem? Please describe.**
There seem to be several (offical?) [docker images](https://hub.docker.com/r/zephyrprojectrtos/ci) on dockerhub used for CI. There is no documentation, examples, or repository link of any kind along with the image. While I could set my own build image up, I would rather use an offical image for CI if possible. 

**Describe the solution you'd like**
Document docker CI images with basic instructions on how to use them and a link to their repository(s).

**Describe alternatives you've considered**
Create my own docker image for CI purposes.

aescolar, 2020-02-11:
CC @nashif @galak

nashif, 2021-05-28:
we only have one docker image now and it is documented https://github.com/zephyrproject-rtos/docker-image

closed 2021-05-28 by nashif

windelbouwman, 2021-06-07:
This docker image looks useful to setup a gitlab runner for example. How can I use the (which?) docker image to build my zephyr project?