zephyrproject-rtos/zephyr #22748: Introduce pinctrl API to support pinctrl dts nodes

erwango, 2020-02-12:
## Introduction

In order to be able to use dts defined pin configurations, zephyr needs to propose a pinctrl API allowing to configure pins based on dts dt-bindings. This API will have to co-exist with newly introduced gpio API.

This issue is the follow up of https://github.com/zephyrproject-rtos/zephyr/issues/15611. It does not suggest solution, but is meant to launch discussions.

### Problem description

Following Linux description: "modules that control pin multiplexing or configuration parameters
such as pull-up/down, tri-state, drive-strength etc are designated as pin controllers".
A pinctrl API should allow device drivers to configure pins based on data available from dts pinctrl nodes, using pinctrl bindings (cf [Linux pinctrl-bindings](https://www.kernel.org/doc/Documentation/devicetree/bindings/pinctrl/pinctrl-bindings.txt)).
Pinctrl operation may be done by pinctrl client at boot time (initial device pin configuration), but also at run time (set pins to a power saving state).

**Things to consider**

Impacts of following points may vary across architectures, this list intends to be exhaustive (update if needed).

- Functional overlap:

Considering dt-bindings, pinctrl operations will have a certain overlap with gpio operations:
gpio: direction(i/o/disconnect), value, bias(pull_up/down), drive(push-pull, open-drain, ..), active level(active low/high), interrupts
pinctrl: muxing, bias, drive, input, slew rate (cf [Linux pincfg dt-binding](https://www.kernel.org/doc/Documentation/devicetree/bindings/pinctrl/pincfg-node.yaml))

- Shared resources:

gpio and pinctrl driver may require access to same pinctrl/gpio modules.
Aside from the pure conflict of access, impact on low power operations (controller clock switch off) should also be considered.

**Impact on implementation** 

Aside from the pure API perspective (what services to offer), on the architectures impacted by some of the above points, pinctrl API implementation will raise dilemmas to resolve ahead of implementations. 
Implementing pinctrl driver fully separately from gpio driver will have impact on memory footprint and make the resource sharing complex.
Other solutions could be considered: 
- gpio devices relying on pinctrl sevices or vice versa
- gpio and pinctrl relying on services of common driver.

erwango, 2020-02-12:
@carlescufi, @galak, @MaureenHelm, @pabigot, @mnkp, @b0661, @mbolivar

b0661, 2020-02-12:
**Things to consider**

- Runtime pinctrl data access:

The Linux pinctrl interface has a lot of functionality to query pinctrl configuration data. This functionality should be guarded by a config option (e.g. CONFIG_PINCTRL_RUNTIME_DTS), if implemented at all,  to enable pinctrl for very memory constraint devices.

- Pin controller device struct data

The way to generate the pinctrl configuration data for pin controller devices is key in the implementation of such a driver. Currently there is no adequate support in Zephyr. The tooling shall be clarified (gen_defines.py, another python script, ...).

b0661, 2020-02-13:
**Steps to implement**

Steps to implement and candidate PRs for the different aspects of implementation.

- Pin controller API (aka. include/drivers/pinctrl.h)
  - [ ] TBD

- Device tree pin configuration data encoding (aka. include/dts-bindings/pinctrl.h)
  - [ ] TBD 

- Pin controller device struct data format (struct config, struct data)
  - [ ] TBD

- Device tree pin configuration data extraction (tooling)
  - [ ] #22255, enhance edtlib.py for pin controller data extraction

- Pin controller device struct data generation (tooling)
  - [ ] TBD

- Pin controller device implementation for all SoCs
  - [ ] TBD

erwango, 2020-02-13:
@b0661 , out of curiosity, from what I understand you've already implemented pinctrl driver oot, could you share with us the solution you chose regarding gpio/pinctrl responsibilities partitioning?

b0661, 2020-02-13:
@erwango 

The solution is already in #6760. Gpio is using pinctrl as a backend for pin configuration and muxing.

galak, 2020-02-13:
Runtime Pin ctrl use model: https://github.com/zephyrproject-rtos/zephyr/issues/11918

galak, 2020-02-13:
https://github.com/zephyrproject-rtos/zephyr/issues/20012

carlescufi, 2020-10-06:
API meeting 6th October 2020:

@erwango describes the current state of the pinctrl API and bindings today.

Usecases:

- Runtime PIN control: to what extent?
- I2C and GPIO on the same set of pins to reset the bus

What do we want to achieve?

- A generic pinctrl API for drivers to map pins to functions. The mapping could come from Devicetree or be populated at runtime by the caller
```
pinctrl_pin_set(P0, 12, UART_TXD);
pinctrl_pin_group_set(.. group ..);
```

- An API to configure a pin (pull-up, pull-down, drive strength, etc)
```
pinctrl_pin_flags_set(pin, PULL_UP);
```

- Maintain, to the max extent possible, compatibility with the Linux Devicetree bindings

carlescufi, 2020-10-13:
API meeting 13th October 2020:

- @gmarull volunteers as long as it is a shared effort and it doesn't fall entirely on his shoulders
- @carlescufi to check with @erwango for availability
- @galak and @henrikbrixandersen also volunteer to help
- @mnkp best-effort

----

- In Linux, the pinctrl subsystem initializes the pins independently of the peripheral driver, before the driver is "probed". The driver can then change the state. In some cases however (at least power management), the driver does switch states
- In Linux there are a set of mutually exclusive predefined pin states that can be selected
- In Linux the pinctrl subsystem identifies pins using the identifiers in the actual package. This applies to things like GPIO extenders, where the pins are treated the same as the built-in SoC ones.

@mnkp asks:

- In the pinctrl API, should we use the absolute/global "package" pin numbers or have separate "namespaces" per-port.
- It is suggested that the encoding of the pin could be SoC-specific, and the API be generic enough that it can transport any representation of a pin number, typically a combination of a port and a pin.

gmarull, 2020-10-13:
For reference, an overview of pinctrl: https://wiki.st.com/stm32mpu/wiki/Pinctrl_overview

galak, 2020-10-15:
A few topics for thought:
* how pin params/flags (pull-up/down, slew, etc) are handled.  Via gpio api, via pinctrl api, via both - leave to implementations.
* How devicetree and api that is dev, port, pin, flags works for SoCs that have pinmux at the device level (SiLabs, Nordic).

jettr, 2020-10-16:
I think it will be confusing if both pinctrl and gpio API allow for us to configure the same information (e.g. pull up/down, drive strength, open-drain). We should pick one of them to do it (I would lean towards GPIO).

Should pinctrl focus more on the alternate mode and grouping aspects of pins? Pinctrl could be thought of as the controller for the pinmux device/operations.

---
On a related but different note, I noticed that the gpio API does not have a way to get the current gpio flags. It would be really useful to have a getter function to determine if the gpio was currently an input/output or push-pull/open-drain. (I can create a separate issue for this, but it does seem related to this overall discussion)

pabigot, 2020-10-16:
@jettr Please open an issue for fetching GPIO flags.  I've considered this before but because information can be lost in translating from the `gpio_flags_t` used to configure the peripheral into the peripheral-specific register settings it's not clear that the configuration can be reconstructed, or that it's worth caching the original configuration (4 bytes RAM per GPIO pin), or that doing so would be particularly useful.  Anything that configures a GPIO should be able to track out how it was configured, and it's not obvious that things that didn't configure the GPIO need to know.  But that can be discussed in the proposed issue.

nandojve, 2020-10-16:
The #28645 is an example that shows how to do it using pinctrl only. There is a variable length {0 - N} list for params. Users can complement that on board DTSI files or create their own. That shows how drivers can directly configure using pinctrl-0. Of course it is possible elaborate more and have a centralized entity to initialize all at same time. I think it is not so difficult to create a struct to store the data using DT_FOREACH like function, once params are bitfields and example shows how to handle it. It is possible inclusive store all pinctrl+flags (enabled and not at DT) that can be useful at runtime to change configurations.

henrikbrixandersen, 2020-10-18:
> I think it will be confusing if both pinctrl and gpio API allow for us to configure the same information (e.g. pull up/down, drive strength, open-drain). We should pick one of them to do it (I would [...]

I think we need to support pull up/down, drive strength, etc. via pinctrl. If we do not, how would one specify e.g. the drive strength for a non-GPIO pin?

erwango, 2020-10-19:
Here are few thoughts I'd like to share on pinctrl API. These could be seen as proposals to define API basis and are open to discussion.

> how pin params/flags (pull-up/down, slew, etc) are handled. Via gpio api, via pinctrl api, via both - leave to implementations.

The API should target on the services that should be provided to the users and that are required to implement portable applications and frameworks on top of it.
We should not mix API and SW architecture. Across various vendors, the hw implementation of pinctrl is quite heterogeneous.
IMHO, dealing with how the service is provided should be left implementation dependent. Each underlying HW blocks being different, the best way to provide the service will also vary from one HW to another. I think this should be left architecture specific

> I think it will be confusing if both pinctrl and gpio API allow for us to configure the same information (e.g. pull up/down, drive strength, open-drain). We should pick one of them to do it (I would lean towards GPIO).

Agreed, we should not have API with conflicting responsibilities. Here's my view :
- gpio api is in charge of individual pins configuration. It provides ways to control pin state directly from applications or frameworks:
  - We might need to clear some ambiguous references from hte gpio API. Cf: https://github.com/zephyrproject-rtos/zephyr/blob/3ce7afdaaa005dc0f46d721ad77d0c275ce0df9f/include/drivers/gpio.h#L454-L456
- pinctrl api is in charge of handling pin states of devices that directly control pins, without need for intermediate control code
  - In that view, pinctrl api provides a way to configure a set of pins in a state required to provide a function useful to a specific user device (eg: UART_TX/UART_RX).
  - The resulting pin configuration to achieve this function (eg: adequate pinmux setting plus pull-up/open drain for uart_tx on STM32F4) should be seen as opaque at the API level. Either this configuration is done internally by gpio driver is possible but left implementation dependent.

Default opt for simplicity:

Linux pinctrl api defines various API "levels", which FWIK, are not systematically implemented.
I think we should go through these and see which of them make sense and which don't or would be too much resource consuming. For instance, is that API responsibility to handle conflicting configurations? This would for sure be useful, but is the (resource-wise) cost of an embedded pin config sanitizing acceptable ?
For "niche" functionality, maybe we should consider if the service couldn't be provided at application level or dealt with on the user side.
And so on...

jettr, 2020-10-19:
I wanted to add two usecases when considering these APIs.

It is possible for i2c lines to have internal or external pull ups on the line. Depending on the hardware board layout, the SoC (this chip) should either enable or disable internal pull up resistors on the i2c pins when they are being use as the i2c alternate mode. I think the pinctrl api should be the one in charge of selecting alternate functions like I2C.  However when pin is selected to an alternate mode, e.g. I2C, the pull up/down control is still a valid function at the board/project level. I would think in that case you would use pinctrl api to select i2c, and gpio api to select the pull up/down.

Thinking of the very simple case were there are no alternate modes available for pins and only gpio is available, I would not want the application to have to use both gpio and pinctrl apis. I would want it to access everything through gpio api, even to set their pull up/down settings.

erwango, 2020-10-20:
> I wanted to add two usecases when considering these APIs.
>
> It is possible for i2c lines to have internal or external pull ups on the line. Depending on the hardware board layout, the SoC (this chip) [...]

>  I would think in that case you would use pinctrl api to select i2c, and gpio api to select the pull up/down.

Not necessarily. This can be done using only pinctrl api - today.
Let me take exemple of current STM32 dt_pinctrl api.
Let's say first board uses i2c1, which is using PB8: SCL, PB9: SDA:

```
board-1.dts
[...]
&i2c1 {
	pinctrl-0 = <&i2c1_scl_pb8 &i2c1_sda_pb9>;
	status = "okay";
};
```
with signals defined with default pull-ups  as:
```
dts/st/f4/stm32f429ziyx-pinctrl.dtsi
i2c1_scl_pb8: i2c1_scl_pb8 {
		pinmux = <STM32_PINMUX('B', 8, AF4)>;
		bias-pull-up;
		drive-open-drain;
};
i2c1_sda_pb9: i2c1_sda_pb9 {
		pinmux = <STM32_PINMUX('B', 9, AF4)>;
		bias-pull-up;
		drive-open-drain;
};
```
At driver level, pin configuration si done thanks to a single call to STM32 pinctrl API, which takes as argument a device cfg->pinctrl_list which is filled at compilation thanks to DT API:
```
drivers/i2c/i2c_ll_stm32.c
[...]
		stm32_dt_pinctrl_configure(cfg->pinctrl_list,
					   cfg->pinctrl_list_size);
[...]
static const struct soc_gpio_pinctrl i2c_pins_##name[] =		\
					ST_STM32_DT_PINCTRL(name, 0);	\
static const struct i2c_stm32_config i2c_stm32_cfg_##name = {		\
[...]
	.pinctrl_list = i2c_pins_##name,				\
	.pinctrl_list_size = ARRAY_SIZE(i2c_pins_##name),		\
[...]
};
```
If board-2 uses the same pins but with disabled pull-ups, you only need to redefine the pins at board dts level:
```
board-2.dts
[...]
&i2c1_scl_pb8 { /delete-property/ bias-pull-up; }
&i2c1_sda_pb9 { /delete-property/ bias-pull-up; }
&i2c1 {
	pinctrl-0 = <&i2c1_scl_pb8 &i2c1_sda_pb9>;
	status = "okay";
};
```
This has effect without any impact on i2c driver code.
 

> Thinking of the very simple case were there are no alternate modes available for pins and only gpio is available, I would not want the application to have to use both gpio and pinctrl apis. I would want it to access everything through gpio api, even to set their pull up/down settings.

This use case can also be dealt with using the only pinctrl API.
I'm taking the example of STM32F1 series that have a quite limited concept of alternate functions (REMAP), and configure some signals by gpio configuration. This is the case for usart_tx for instance, which is defined as:
```
dts/st/f1/stm32f103t(4-6)ux-pinctrl.dtsi
usart1_rx_pa10: usart1_rx_pa10 {
		pinmux = <STM32F1_PINMUX('A', 10, GPIO_IN, NO_REMAP)>;
};
```
No alternate is required to configure this signal, this is pure GPIO in.
This configuration is done at driver level with the same unique call to STM32 pinctrl API (which, also takes as argument  a device cfg->pinctrl_list which is filled at compilation thanks to DT API):
```
drivers/serial/uart_stm32.c
[...]
		stm32_dt_pinctrl_configure(config->pinctrl_list,
					   config->pinctrl_list_size);

```

I'm not saying that pinctrl API should be made from the STM32 one. But this shows that in both cases you proposed, it would be possible to perform pin configuration using only **a** pinctrl api.

galak, 2020-10-20:
I've started a "sub-issue" to capture discussion around how to represent devicetree data for pinmux on SoCs in which all the pinctrl is centralized (like STM32, NXP, etc).

https://github.com/zephyrproject-rtos/zephyr/issues/29369

galak, 2020-10-20:
based on API meeting - we might want an API that given a "signal" to get back the GPIO device pointer (if there is such a mapping available).

jettr, 2020-10-20:
> But this shows that in both cases you proposed, it would be possible to perform pin configuration using only **a** pinctrl api.

I agree that we could do everything with the pinctrl API. If that is the case, then we should get rid of the GPIO API as pinctrl API is a superset of the GPIO API, right? Or would the only thing in the gpio API be get and set level?

I don't think we want to end up in a situation where you can do almost all of the same things (e.g. setting pull up/down) in both the gpio and pinctrl APIs. Only one of them should be the "correct/recommend/available" way to do it.

erwango, 2020-10-21:
> I agree that we could do everything with the pinctrl API. If that is the case, then we should get rid of the GPIO API as pinctrl API is a superset of the GPIO API, right? Or would the only thing in the gpio API be get and set level?

GPIO API does things that have nothing to do with pinctrl, like IRQ handling for instance.

> I don't think we want to end up in a situation where you can do almost all of the same things (e.g. setting pull up/down) in both the gpio and pinctrl APIs. Only one of them should be the "correct/recommend/available" way to do it.

This is why I tried to provide a split of responsibilities [in this comment](https://github.com/zephyrproject-rtos/zephyr/issues/22748#issuecomment-711915688).
In short:
- gpio API would be used for bit SW pin handling (bit banging, get/set, ..) and gpio interrupts (buttons, led...)
- pinctrl API would be used for signals that should be used with dedicated hardware (i2c, adc, ...)

Depending on the underlying hardware there would be some overlap indeed, but not in a systematic way.
As example, on some hardware, ADC pins could be handled directly by ADC block, and in that case, usual pin electrical properties are not available (drive-, bias-, ..). This is more generally usually the case when pin are set to 'analog' function.
On the opposite, when a pin is used by gpio API, user expects these to be available.
Besides, on the pinctrl API, as opposed to gpio API, the flags would not be directly exposed to the API, but part of an opaque signal description ("pinctrl state") provided by device tree.

Please note that concept should be studied in the light of potential runtime pinctrl support.

jettr, 2020-10-22:
Do the follow examples fit in line which what you are proposing then?

**GPIO API**
* Register IRQ (e.g. level/edge)
* Get/Set raw level (e.g. high or low)

**PinCtrl API**
* Set pull up/down 
  * includes different resistor strength
* Set output driver characteristics
  * e.g strength, skew
* Set alternate function on pin
  * e.g. ADC, Timer output, I2C
 
**Timer API**
* Set output waveform and frequency

**I2C API**
* Set bus speed, etc.

**ADC**
* Set input read frequency

The Timer, I2C, and ADC APIs are just there for completeness of the example.

erwango, 2020-10-23:
> Do the follow examples fit in line which what you are proposing then?

For now, I'm not proposing any change in GPIO API. So whatever is done in GPIO API stands (PU/PD included).

For the proposed pinctrl API, there's indeed some overlap with GPIO API in the pin setting parameters (PU/PD for instance).
The distinction is:
- The destination of the API (sorry for being lazy, but I'm just copy/pasting what I proposed in earlier comment):
  - gpio API would be used for bit SW pin handling (bit banging, get/set, ..) and gpio interrupts (buttons, led...)
  - pinctrl API would be used for signals that should be used with dedicated hardware (i2c, adc, ...)
- Also, _in case we don't need run time configuration_ (which is still under investigation), the configuration would be done exclusively by the use of opaque 'states' defined in device tree, such as the ones you can find [here](https://github.com/zephyrproject-rtos/hal_stm32/blob/master/dts/st/f4/stm32f401c(b-c)ux-pinctrl.dtsi)

This being said, I'm not inventing anything as this mainly inspired from Linux.
The part on which I'd like to differ is the extensiveness of the original Linux API which covers lot of notions, which could be costly in terms of resource and of a disputable interest in our use cases (besides not systematically implemented on Linux platforms neither).

jettr, 2020-10-23:
> For now, I'm not proposing any change in GPIO API. So whatever is done in GPIO API stands (PU/PD included).

I do think it is worth looking at both gpio and pinctrl API together and see how we want them to work together long term. If we want to eventually move an API from gpio to pinctrl (e.g. pull up/down), then I think we should call that out. If we don't, then we will have two "competing" APIs that will cause end-user confusion.

Just to help me understand, does the following phrase
>gpio API would be used for bit SW pin handling (bit banging, get/set, ..) and gpio interrupts (buttons, led...)

include anymore more or less that then example GPIO API I mentioned in [post above](#issuecomment-714609204)? That is how I read it at least. I think we might be saying the same thing, but I want to make sure we aren't talking past each other.

Regarding run time configuration, our applications do switch configurations at run time based on external factors, but we could describe the full set of opaque states at build time in dts. At run time, we could just select the correct opaque state based on the external factors. I don't really think there are enough alternate modes for most MCUs to prevent the application developer from realistically describing all of the possible modes they expect a give pin (or set of pins) to use at build time.

henrikbrixandersen, 2020-10-25:
I still see use-cases for keeping the existing GPIO flags in the GPIO API along with introducing similar configuration via a pinctrl API. I will try to outline a few below:

1) Having to set a drive strength on a SoC PWM controller (e.g. for increasing the current in order to drive an LED using PWM). It would be odd having to go through the GPIO API for this. The PWM API flags could be extended to support this, but the pinctrl API would be more common and would be directly usable for other pin functions as well.

2) Having to set pin configuration (e.g. drive-strength, pull-up/pull-down, open drain, ...) on GPIO expanders (e.g. via SPI or I2C). This is closely related to the GPIO expander driver. Requiring a pinctrl driver for each GPIO expander would introduce a lot of overhead.

We could either choose to keep the existing GPIO controller code - or reimplement the flags handling to use the corresponding SoC pinctrl driver, when it becomes available.

erwango, 2020-10-26:
> If we want to eventually move an API from gpio to pinctrl (e.g. pull up/down), then I think we should call that out. If we don't, then we will have two "competing" APIs that will cause end-user confusion.

I understand your concern about user confusion, and agree we should take this into account. 
About the possibility of removing pin attributes from current GPIO API, this means that for configuring a button or a led, we'd have to call 2 different APIs, and I'm not sure this will make things easier in the end.

> include anymore more or less that then example GPIO API I mentioned in post above? That is how I read it at least. I think we might be saying the same thing, but I want to make sure we aren't talking past each other.

Yes, but I also meant keeping existing attributes (sorry if this was misleading).

> Regarding run time configuration, our applications do switch configurations at run time based on external factors, but we could describe the full set of opaque states at build time in dts. At run [...]

This was my initial though as well. Though, the use case targeted (https://github.com/zephyrproject-rtos/zephyr/issues/11918) tends to a wider understanding of run time configuration. I've added you to the slack channel so you can follow this particular topic.

jettr, 2020-10-26:
> About the possibility of removing pin attributes from current GPIO API, this means that for configuring a button or a led, we'd have to call 2 different APIs, and I'm not sure this will make things easier in the end.

If we did transition flags from gpio to pinctrl API, then we should do it slowly and with lots of documentation. If we can teach end-users the difference between the two APIs and when to use one or the other, then I think it will still end up being easier overall. 

The simple LED case may need two APIs, but then the user knows where to go when the drive strength of the pin isn't enough and they need to increase it. Or maybe they need to switch to PWM leds, and then they are already familiar with the pinctrl API to switch over from gpio to pwm. In the case of a really simple LED setup and the SoC sets gpio as the default pin configuration, the end-user may still only need to interact with the pin using the gpio API.

If we did keep duplicate API in gpio (which I would lean against), we should very clearly document it as a pass-through API to pinctrl for convenience only.

henrikbrixandersen, 2020-10-26:
> If we did keep duplicate API in gpio (which I would lean against), we should very clearly document it as a pass-through API to pinctrl for convenience only.

It wouldn't always be a pass-through API. Or do you envision us having to do pinctrl drivers for e.g. GPIO expanders on I2C/SPI busses?

jettr, 2020-10-26:
> It wouldn't always be a pass-through API. Or do you envision us having to do pinctrl drivers for e.g. GPIO expanders on I2C/SPI busses?

For GPIO expanders, I would imagine that they would chose either

1. Implement gpio and pinctrl API just like internal gpios. This means whether a gpio is internal or on an expander is transparent to applications using and configure them.
   * There still needs to be special care around getting and setting gpio levels on expanders in an interrupt context though
2. Implement a gpio_expander API (and not gpio or pinctrl). This gpio_expander API would advertise all of the functionality available on that IC, which would include things that are in both the gpio and pinctrl API.
   * This would makes it easier for the application developer to avoid calling gpio_expander_get_level in an interrupt context.
   * gpio expanders may have special HW features that we could add to this specialized API

I think both options have pros and cons. It is nice to be able to treat a expanded gpios as an internal gpios, _expect_ when you can't like interrupt context.

carlescufi, 2020-10-27:
I tried to summarize the runtime pin control in the body of this issue, with this text:

- Pin configuration at runtime. We can support this in two different levels:
  - Set or group-based: A single firmware image supports a finite set of board variants, with the selection taking place at boot time using a runtime variable (for example checking the level of a GPIO, which will be high on the a variant and low in another one). The flexibility on the differences among board variants is to be discussed, but at the very minimum it should support removing functions (eg. an external SPI flash present on a development board, but not on the final product)
  - Support the possibility of assigning functions to pins freely at runtime cf #11918, #20012.

gmarull, 2020-10-27:
I think `gpio` and `pinctrl` APIs should be considered orthogonal topics, even if
they have a few common things like the bias setting. I personally think about
the current gpio API as the "gpiolib" API available for Linux userspace apps,
that is, an API to manipulate I/O pin levels (get, set, IRQ...). It is quite
common to require bias when using a gpio (e.g. for a push button), so it is
useful that gpio API allows changing it.

Regarding `pinctrl`, I see it as an API to mainly configure _signals_, e.g. I2C
SDA, SPI SCK... My understanding of a `signal` is the output or input of a
certain hardware IP block (i.e. not general purpose). It is common to require
certain flags for a signal to operate in proper conditions, e.g. for I2C
drive-open-drain and possibly activate the pull-up. So I think these capabilities
need to be offered by a `pinctrl` driver. In most circumstances the consumer of
these signals (the driver) doesn't need to do anything else other than having
these signals configured prior to the device initialization. In more advanced
use cases, e.g. power management, the driver may also need to adjust the
_signal_ configuration when changing to another operation mode. Linux uses the
concept of _states_ to model that functionality. A certain device can hold
multiple pin groups, each representing a state. Some state names are
_standardized_, e.g. `default`, `sleep`, `init`...  A driver that has a single
state, i.e.  `default` will have no need to interact with `pinctrl` because this
state will be set before probing. In case it needs to change to another state at
runtime, it will call a `pinctrl` API providing the new desired state.

A DT representation example (STM32):

```
pinctrl {
    usart1_pins_a: usart1-0 {
        pins1 {
            pinmux = <STM32_PINMUX('A', 9, AF7)>,
                     <STM32_PINMUX('A', 10, AF7)>;
        };
    };

    usart1_sleep_pins_a: usart1-sleep-0 {
        pins1 {
            pinmux = <STM32_PINMUX('A', 9, ANALOG)>,
                     <STM32_PINMUX('A', 10, ANALOG)>;
        };
    };
};

usart1 {
    pinctrl-0 = <&usart1_pins_a &usart1_pins_sleep_a>;
    pinctrl-names = "default", "sleep";
};
```

One thing that we could think of is to be able to use `pinctrl` to configure
GPIOs at boot time, thus avoiding the need to call `gpio_configure`. But I think
that the state modeling for GPIOs is less clear.

Regarding runtime configuration:

I think that applications should not directly be in charge of configuring
pinctrl mappings. If we take the states modeling approach, it means we would
have two parts tweaking a state: not good. It does not mean that an application
should not be able to change the mappings, though. One possibility I am think of
is to have state groups, so an application could choose from a list of state
groups at runtime, then, the driver would be in charge of managing the state on
its own from the active state group. I think that would require, at least in
some cases, to be able to reload the driver.

nandojve, 2020-10-29:
I agree with @gmarull on almost every aspect he described, almost same things I'm thinking about both topics GPIO/pinctrl.

_It is quite common to require bias when using a gpio (e.g. for a push button), so it is useful that gpio API allows changing it._
I would like add another point:
1) Let's assume pin will need PM, it will require states like default and sleep, right? In this case, I think pinctrl can accommodate on a standard way the information and move from one state to another. If we keep GPIO flags to do it, it will need a new API, probably. I do prefer in this case only keep pinctrl.
2) Otherwise, If the pin don't need PM, pinctrl can be used to store and initialize the GPIO conditions (it don't change at runtime). So, GPIO API only need (get, set, int) things.
It seems that pinctrl can accommodate better instead share functionality with GPIO. 

GPIO expanders

I saw it same way as SoC pins with a transport mechanism. SoC may need a pinctrl driver to handle the API and another GPIO driver to execute (get, set, int). This means, that an GPIO expander should define a pinctrl and expose it, and drivers need accommodate the API. I don't know if Zephyr already offer the ways to cascade GPIO expanders. As I see, the fist GPIO expander is, in fact, already cascaded with SoC, having more, is just a consequence. Zephyr may explore the idea of having a virtual driver to allows cascade but it is another topic. The things is, we have ways to solve without create another API just for expanders.

Regarding runtime configuration

I was thinking about it and identified two distinct scenarios:

1) **Deep embedded**: This is the default scenario. A smartwatch will be a smartwatch, always. The user will never change their functionality, unless you are a hacker.
2) **Arduino like boards**: This is new to all of us. These boards bring to us some PC like capabilities and expose the generic use case. For those boards (not products), the runtime configuration makes sense.

Zephyr is moving to [1] direction and I hope continue on that direction. For [2], I see benefits having it, but for: tests, concepts, prototypes etc. Eventually a product like [2] will be ready and it will became [1]. Even using embedded python, once the final product became [1] it will never change. The most close example to a final product like [2] that it will need runtime configuration is a flight controller. There is a firmware there that user selects and configure pins. The problems is, once user complete the setup, it became [1] and never changes. Commercial products like drones, in general, don't allow it.

It makes more sense Zephyr provide the mechanisms to allow [2] from user space. It is important mention that the runtime configuration will provide similar functionality like the generic PC environment over time and, in this case, will bring the same security issues. If someone gets device control can, inclusive, harm people, share OAuth keys etc,

I'm not saying that we shouldn't have it, instead, reinforce that any runtime hardware change must be made with security on mind. Otherwise we can easily open the doors to threats. So, I do prefer have a controlled way to do it.

galak, 2020-11-05:
Dev-Review (Nov 5):

* pincfg APIs:
  - low level API that is opaque w/regards to function param (ie its SoC specific) for pin/func/etc config
  - provide pin group setting APIs around a few common use cases (default, power-mgmt, I2C bus reset)
    [ Intent is a finite set of pin data that can be "switched" at runtime based on specific use case (power mgmt state change, i2c bus reset)]

* The ability to generically identify pin as UARTn/TX is not something we intend to support
* General feeling is to leave GPIO API as is w/regards to pin configuration.  We'll have pincltr API also expose the ability to config pin parameters and leave it to implementations on how this map between the GPIO API version (ie GPIO API could call pinctrl version, or pinctrl version could call GPIO API, or no coupling, etc.)

carlescufi, 2021-04-27:
API meeting 27.04.2021

@mnkp describes the options available to us

a) Closer to the Linux approach
b) Closer to the current Zephyr approach

a) The pinctrl driver owns an array of `struct cfg` with all the possible configurations that the application may need. In order to select one of the configuration one would then call `pinmux_select(int select)`. In this case `struct cfg` contains a string that identifies the function for this particular configuration.
A UART driver could then call:
`pinmux_find_pin_cfg('uart0')`, which would return 7
`pinmux_select(7)` and that would configure the 4 UART pins 

b) `pinctrl_pin_configure(struct pinctrl_cfg *cfg)` and `pinctrl_pin_list_configure(struct cfg[])`. The contents of `struct cfg` would be SoC-family dependant (`pinctrl.h` would include SoC-specific files). Ownership of the `struct cfg` instance could be owned by the caller or by the callee (TBD). Accessing the fields in `struct cfg` would be done via macros that each SoC family would implement. The contents of `struct cfg` would be mainly the required pinmux data and the GPIO configuration.

mbolivar-nordic, 2021-04-27:
> a) Closer to the Linux approach

I was having audio problems when this was discussed, but I still cannot see how this can be made to work for Nordic devices in a space-efficient way.

If we require a full array of possibilities, then it will contain NPINS * NFUNCTIONS elements, which is simply impractical even if stored in flash, and anyway most of the data will never be used by 99% of applications, even the ones that want to dynamically switch between a few possibilities.

What am I missing?

erwango, 2021-04-28:
> > a) Closer to the Linux approach
> 
> I was having audio problems when this was discussed, but I still cannot see how this can be made to work for Nordic devices in a space-efficient way.
> 

I'm also concerned about "with all the possible configurations that the application may need". Is that all existing pinmux configurations (which could include all potential tx/rx combinatory) or only the ones configured in device tree by enabled nodes?

gmarull, 2021-04-28:
I think storing all possible combinations is definitely a no-go. In SoC like ST we'd be wasting a lot of ROM for nothing. I think we definitely should avoid something like https://elixir.bootlin.com/linux/latest/source/drivers/pinctrl/stm32/pinctrl-stm32mp157.c here. If I'm not wrong, what was discussed was to store all combinations listed in a board `-pinctrl.dtsi` file in a single place (now it is done per-driver). In fact, we should only store nodes referenced in `pinctrl-N` properties.

carlescufi, 2021-04-29:
@mbolivar-nordic and @erwango

To be clear, when I say "with all the possible configurations that the application may need", what I mean is "all pin configurations that are applicable to one particular board". So that means that you have to know **at build time** which pins are connected where, in some cases to more than one peripheral, on the board, and that's the only set of combinations that will be encoded and stored in the pinctrl driver. For example:

Pins 10 and 11 are physically wired to an FTDI chip that converts the UART signals to USB. But it turns out that there's also direct terminal access via raw UART provided in a port connector, to which pins 21 and 22 are wired. The software can, at runtime, decide when to route UART0 to either FTDI/USB or the UART connector.
It would then encode the two possible combinations in DT as such:

 ```
my-board-pins.dts:

	uart0_pins_a: uart0-0 {
			pinmux = <STM32_PINMUX('G', 10,  AF6)>; /* UART0_TX */
			bias-disable;
			drive-push-pull;
			slew-rate = <0>;
			pinmux = <STM32_PINMUX('B', 11, AF8)>; /* UART0_RX */
			bias-disable;
	};

	uart0_pins_b: uart0-1 {
			pinmux = <STM32_PINMUX('G', 21,  AF6)>; /* UART0_TX */
			bias-disable;
			drive-push-pull;
			slew-rate = <0>;
			pinmux = <STM32_PINMUX('B', 22, AF8)>; /* UART0_RX */
			bias-disable;
	};

mbolivar-nordic, 2021-04-29:
@carlescufi that sounds more reasonable to me, but I think it is not compatible with "the Linux approach" option proposed by @mnkp . Or did I misunderstand?

Anyway, I guess we'll see when we have a real API proposal, but it seems like we all agree that the Zephyr API needs to give the user (i.e. the board or application maintainer) full control over how much ROM to allocate to pin configuration data. If so, that addresses my concern.

carlescufi, 2021-04-29:
> @carlescufi that sounds more reasonable to me, but I think it is not compatible with "the Linux approach" option proposed by @mnkp . Or did I misunderstand?

That is, to my understanding, exactly the Linux approach. Except Linux doesn't really have "boards" so they define more combos than strictly required. We can narrow it down better than them because we know exactly for which physical PCB we are building for.

mnkp, 2021-04-29:
Just to reconfirm what @carlescufi has already stated. In case of option a) described here as "Closer to the Linux approach" the pinctrl would only store those combinations of pin configuration data that the application developer defined in the DTS for the given board they are developing their application for.

As it was already noted here, with modern hardware, it's not feasible to store all the possible combinations of different pin configurations that are supported by the hardware.

Option a) is actually more optimal regarding storage requirement since all the data are kept in one central place and made available to any interested party: driver or user application alike. In case of option b) if two different compilation modules switch pins between the same two peripherals each of the modules will keep local copies of the const struct that they will pass to `pinctrl_pin_configure` function.

mkschreder, 2021-05-12:
@erwango refering to the discussion on stm32l0 power management (#35183) and regarding the pinctrl related stuff for low power mode, what do you think of putting it into the stm32 pinmux driver (drivers/pinmux/stm32/pinmux_stm32.c)? 

Currently the driver doesn't define an instance of device with all the callback functions but perhaps it should? Then I'm thinking we could experiment with the idea of using pm callback to put individual port instances to sleep and use gpio pinmux dts nodes for configuring what to do with different pins. 

This would involve a config option for PM_GPIO that will enable GPIO power save functionality. When enabled, by default it would disconnect all GPIOs upon entering powersave - but the programmer can then set specific gpio pinmux modes for certain pins or set them to be ignored when entering power saving mode. 

I can see that nxp has for example these nodes:

`/dts/arm/nxp/nxp_ke1xf.dtsi:           porta: pinmux@40049000 {`

But I don't seem to find anything relevant for stm32. 

I think low power pin configurations are something that is closely related to gpio ports as they are configured on a specific board. It makes sense to put all configuration for low power deep sleep modes in one place. Perhaps even in power-states section in the board dts file. I think it would be way too complex to go through every single device driver and try to figure out how to configure low power modes for all the pins that it is using.

erwango, 2021-05-12:
> Currently the driver doesn't define an instance of device with all the callback functions but perhaps it should? 

That could be studied indeed and added if helps in pinctrl design.

> /dts/arm/nxp/nxp_ke1xf.dtsi: porta: pinmux@40049000 {
But I don't seem to find anything relevant for stm32.

Well, that's a bit different indeed on NXP and STM32 as on NXP (at least kxx) they have specific/distinct hw blocks for pinmux and gpio (please someone correct me if I'm wrong) and this is not the case on STM32 as "gpio" blocks are in charge both of pinmux and gpio.

>I think low power pin configurations are something that is closely related to gpio ports as they are configured on a specific board. It makes sense to put all configuration for low power deep sleep modes in one place. 

I'm concerned this would no fit well with [Device Power Management distributed method (PM_DEVICE_RUNTIME)](https://docs.zephyrproject.org/latest/reference/power_management/index.html?highlight=power%20management#distributed-method).

mkschreder, 2021-05-13:
@erwango interesting. I just thought of something. What about having pinctrl-<power-state> properties for all device nodes where <power-state> corresponds to the power state as specified in power-state-name. So for example: pinctrl-suspend-to-ram. 

The idea being that if these properties are present, then they would be automatically used. If not, then it is up to the device driver to manage power. This way the user has the choice to either use pinctrl or let the driver manage it - which works kindof like it does today with the pinctrl-0 property.

gmarull, 2021-06-01:
I'd like to mention a problem with approach (b) I faced a few days ago. Let me detail the example so that the problem can be better understood.

I have a custom driver called `svpwm` that generates a set of PWM signals that feed an inverter circuit (sometimes part of a shield). The "SV-PWM device" should be seen as the microcontroller PWM unit + the inverter. Some driver properties depend on the connected inverter circuit, for example, some circuits have enable/disable GPIOs, some others require PWM and PWM_N (inverted)....

The question now: can I use Zephyr shields? Let's see what happens.

In the current approach, a board is defined with the shield included, as if it was a single board:

```
&timers1 {
	svpwm: svpwm {
		compatible = "st,stm32-svpwm";
                currsmp = <&currsmp>;

                /* These depend on the shield, some shields may also require _N channels! */
		pinctrl-0 = <&tim1_ch1_pa8 &tim1_ch2_pa9 &tim1_ch3_pa10 &tim1_ocp_pa11>;

                /* These are also shield dependent, but have a solution */
		t-dead-ns = <1000>;
		t-rise-ns = <1050>;
		enable-gpios = <&gpioc 10 GPIO_ACTIVE_HIGH>,
			       <&gpioc 11 GPIO_ACTIVE_HIGH>,
			       <&gpioc 12 GPIO_ACTIVE_HIGH>;
	};
};
```

But we can't easily use different shields because we have this:
```
/* This depends on the shield, some may also require _N channels! */
pinctrl-0 = <&tim1_ch1_pa8 &tim1_ch2_pa9 &tim1_ch3_pa10 &tim1_ocp_pa11>;
```

If we had pinmux groups, as Linux, the board would be able to define the following groups:

```
&pinctrl {
    tim1_svpwm_single: tim1_svpwm_single {
        pins1 {
            pinmux = <STM32_PINMUX('A', 0, AF1)>,
                     <STM32_PINMUX('A', 1, AF1)>;
                     <STM32_PINMUX('A', 2, AF1)>;
        };
    };

    tim1_svpwm_full: tim1_svpwm_full {
        pins1 {
            pinmux = <STM32_PINMUX('A', 0, AF1)>,
                     <STM32_PINMUX('A', 1, AF1)>,
                     <STM32_PINMUX('A', 2, AF1)>,
                     <STM32_PINMUX('A', 3, AF1)>,
                     <STM32_PINMUX('A', 4, AF1)>,
                     <STM32_PINMUX('A', 5, AF1)>;
        };
    };
};

svpwm_single: &tim1_svpwm_single {};
svpwm_full: &tim1_svpwm_full {};
```

and then the shield could finish the device definition like this:
```
&svpwm {
    pinctrl-0 = <&svpwm_single>;

    t-dead-ns = <1000>;
    t-rise-ns = <1050>;
    /* use a connector with mappings */
    enable-gpios = <&my_standard_connector 0 GPIO_ACTIVE_HIGH>,
                   <&my_standard_connector 1 GPIO_ACTIVE_HIGH>,
                   <&my_standard_connector 2 GPIO_ACTIVE_HIGH>;
};
```

There may be other solutions to this problem, but I thought it was worth mentioning it.

henrikbrixandersen, 2021-06-15:
> There may be other solutions to this problem, but I thought it was worth mentioning it.

Another option could be to use a board-specific shield overlay.

nandojve, 2021-06-15:
I think @gmarull is a valid proposal and can usefull.

```
svpwm_single: &tim1_svpwm_single {};
svpwm_full: &tim1_svpwm_full {};

&svpwm {
    pinctrl-0 = <&svpwm_single>;
   ...
};
```

In this case, using approach [B], will Zephyr have only pinctrl-0 ? I mean, there is no necessity to have pinctrl-1...N, right?
I was wondering if I'm missing something related to above questions.

gmarull, 2021-06-15:
> Another option could be to use a board-specific shield overlay.

Definitely an option, even though it means duplicating the shield for every board.

gmarull, 2021-06-15:
> In this case, using approach [B], will Zephyr have only pinctrl-0 ? I mean, there is no necessity to have pinctrl-1...N, right?
> I was wondering if I'm missing something related to above questions.

`pinctrl-1..N` would still be needed to represent other states, e.g. low-power. The only difference is that `pinctrl-N` would contain a single phandle and not a list of phandles.

manoj153, 2021-09-08:
do we have any plan to support  boot-time gpio direction setting capability?

Possible benefits :
- the user does not have to invoke alike API `gpio_pin_configure_dt()` on each pin in the source code. 
- if a wrong operation is carried out,  returns an error, eg: invoking `gpio_pin_set()` on an `input` pin and vice-versa

while doing some google-ing found that linux might have such a feature covered, the term called `gpio-hog` 

https://www.kernel.org/doc/Documentation/devicetree/bindings/gpio/gpio.txt
peak from the source:
```
- input:      A property specifying to set the GPIO direction as input.
- output-low  A property specifying to set the GPIO direction as output with
	      the value low.
- output-high A property specifying to set the GPIO direction as output with
	      the value high.
```

gmarull, 2021-09-08:
> do we have any plan to support boot-time gpio direction setting capability?
>
> Possible benefits :
>
> * the user does not have to invoke alike API `gpio_pin_configure_dt()` on each pin in the source [...]

I guess it would be a matter of adding `pinctrl` entries to the GPIO node itself, e.g.

```
gpioa_default: ... {
    /* settings for GPIOA pins */
};

&gpioa {
    pinctrl-0 = <&gpioa_default>;
    pinctrl-names = "default";
};
```

adminy, 2023-07-31:
Not sure if this is the most relevant place to ask but with this can I use something like libgpiod library to list a microcontroller running zephyr via usb as a `gpiochip` ?