zephyrproject-rtos/zephyr #22751: STM32F407 I2C driver hangs

lowlander, 2020-02-12:
The STM32F407 I2C driver hangs from time to time where it waits on the semaphore in the driver.  When that happens the signal on the bus looks like this (SDA = yellow, SCL = green);

[image]

To work around it I added a timeout to the k_sem_take calls, but I have not found the real cause of the problem. On the scope is to see it must go wrong real early, because it generates a start-stop without any data. 

This is on the 1.14 branch, but I cherry-picked most commits (like the new timeout handling) from master. The target is a STM32F407 so it uses the V1 driver. Also not using IRQ's but polling causes the problem to happen less but it still happens.

erwango, 2020-02-12:
@lowlander even if happening from time to time, would you have some code to reproduce?

lowlander, 2020-02-12:
@erwango I will try to "port" the problem to a 407 dev board. This will take me at least until early next week.  As soon as I have more info I'll post it here.

pavlohamov, 2020-02-13:
@lowlander could you be so kind to try play with optimization? 
CONFIG_SIZE_OPTIMIZATIONS=n
CONFIG_NO_OPTIMIZATIONS/CONFIG_DEBUG_OPTIMIZATIONS=y 
and vise-versa

lowlander, 2020-02-14:
@pavlohamov I have tried every optimization setting and it doesn't seem to influence the problem, it happens with all settings.

erwango, 2020-02-14:
Setting prio low while we're waiting for way to reproduce

lowlander, 2020-02-20:
I have been trying to debug the problem, and the only thing I can say is the first IRQ (BS) happens, and the driver sets the DR register with the address, but that never lands on the bus. And no other IRQ's will be generated, so the driver "hangs".  

What is more worrying is that I found others that reported the same issue, but with different software; 
https://community.st.com/s/question/0D50X00009Xkhfn/stm32f2xx-i2c-not-sending-address-after-start

lowlander, 2020-02-24:
@erwango would it be possible to access the SDA and SCL as gpio pins inside the driver, so they can be used to "hard reset" the i2c bus in case a slave gets confused by the start-stop pulse and doesn't release the bus.

Because the ST errata just has as software workaround "reset the peripheral", and than there is a risk of leaving the bus in a wrong state.

So I think it would be good to have a real bus-reset function in the driver, so not everybody has to write their own hacks.

erwango, 2020-02-25:
@lowlander indeed, we're working on it via the introduction of pinctrl definition via device tree.
In this model, each peripheral driver would have access at pin definitions and would be able to use them for multiple purpose: reset, low power, ...
Topic has just started, but you could find some info here: https://github.com/zephyrproject-rtos/zephyr/issues/22748

lowlander, 2020-02-25:
@erwango but how to fix this for 1.14.X ? I can add the timeout and give the i2c hardware a reset, but I have no access to the GPIO pins the reset the bus in the worst case scenario.

erwango, 2020-02-25:
@lowlander 

> @erwango but how to fix this for 1.14.X 

My bad I forgot you were using 1.14. This would be a significant change (and not a simple fix), I don't know the policy on adding enhancement on LTS branch. @MaureenHelm ?

lowlander, 2020-02-26:
@erwango I think just a peripheral soft-reset is the only option, if it keeps failing the 1.14 user will have to build its own hard-reset via GPIO in its own application.

Your timeout patch can be back ported and a 100ms timeout on the mutex should make sure the driver comes back to "userspace" where the user than must check via GPIO if the SDA or SCL are still low and than reset the bus via a "fake" clock signal and a "fake" stop-condition.  

Let me try to make a PR and than move the discussion there.

lowlander, 2020-03-02:
OK after some more research it seems the the driver sets the STOP flag (not sure when/how), this flags stays pending until a data byte is finished or a START is generated, and than it will directly generate a STOP (and that is what I see on the bus). 

I now check and reset the STOP bit in CR1 before doing a START, and this seems to work.

Not really sure where it goes wrong, it has to be some race-condition or else it would always case a problem. 

The fix I have now needs some more long term (several days run time) testing before I'll create a PR.

lowlander, 2020-03-21:
@erwango I made an initial patch set #23663, that fixes my problems.  The ECM problem was very tricky because in only happened every 20 to 30 hours, until I figured out that it can be triggered by simply short circuit SDA to GND, that will directly hang the current driver.

PR #27291 (2020-08-12): stm32f4 i2c bug fixes for I2C hangs due to EMC problems (ported from 1.14 #23663)
This is a port of the I2C 1.14 patch (#23663) for master that fixes #22751. It adds timeouts to the read write functions, and resets the I2C device if a timeout happens. It is needed on my hardware (high EMC environment) to prevent hangs of the I2C bus due to EMC spikes.  

In user space we have extra code that checks the SDA/SCL pins on error and generates extra clock signals to make sure possible hung slave devices get back in a know state. This is hard to port to the driver because it doesn't really have acces [... cut]
closed 2020-08-12 by MaureenHelm