zephyrproject-rtos/zephyr #22758: RFC: Require system clock stability on startup

pabigot, 2020-02-12:
Currently, at least on Nordic, the system timer startup starts an underlying clock, but then starts the timer and continues with system startup without waiting for that clock to stabilize.  In some cases this may take several hundred milliseconds, with no guarantee on how the clock will behave during stabilization.

It's unclear whether this is a bug, or is allowed as an option for under-defined behavior.

I propose that z_clock_driver_init() be expected to wait internally until the clock has stabilized.  This may require support from the clock control API to signal that stability has been reached.

pabigot, 2020-02-12:
@nordic-krch @ioannisg

pabigot, 2020-02-18:
Supporting concern: https://github.com/zephyrproject-rtos/zephyr/issues/20712#issuecomment-581374974

carlescufi, 2020-03-17:
API meeting:

This applies only to the system clock that drives the kernel (ticks) time management.
The issue appears both in `POST_KERNEL` system initialization and also `main()`.

Suggestion (@carlescufi):
- Introduce a new `BOOT_WAIT_FOR_SYS_CLOCK_STABLE` that defaults to `=y` but can be set to `=n` by the user if it needs to operate very early on before the system clock crystal has settled
Reluctance among the audience to expose this to the users (since it is indeed dangerous). Discarded.

Suggestion (@nashif ):
- Add a weak function that waits for clock stabilization `clock_wait_stable()`, can be overridden.
- Potentially add an API that exposes the current stability of the clock?

nordic-krch, 2020-03-17:
I missed the meeting but have one additional note. Not sure how nordic specific is that but low frequency clock (system clock in nordic case) have two sources: RC (faster startup ~1ms, less stable) or XTAL (long startup time ~250ms, precise, needs external hardware). When XTAL source is picked then clock starts in 2 phases, first RC starts so that low frequency clock can run but is less stable, when XTAL is ready seamless switch occurs.

Due to that i would see 3 options:
1. no wait
2. wait for sys clock available (default)
3. wait for sys clock stable

For many vendors 2,3 will be the same.

pabigot, 2020-03-18:
My impression from the discussion was that the group consensus would be to make state 3 (sys clock stable) the default; at a minimum it's my preference.  The existence of an intermediate state is probably SOC specific so at the Zephyr level this should be a binary decision: wait for stable, or allow to execute without stable.

pabigot, 2020-07-07:
API 2020-07-07: Moved from triage to To-Do as desired functionality has been discussed and recorded above.

carlescufi, 2020-07-14:
API meeting 14th July:

Instead of the suggestion [here](https://github.com/zephyrproject-rtos/zephyr/issues/22758#issuecomment-600159720), let's expose to the user the 3 possible alternatives that make sense when a system has both an RC and an external crystal:

1. no wait, likely running on RC
2. wait for sys clock available, running on unstable crystal
3. wait for sys clock stable, running on stable crystal (**default**)

@nordic-krch please note that the default is 3., not 2. as per [your comment](https://github.com/zephyrproject-rtos/zephyr/issues/22758#issuecomment-600189618). Feel free to propose a Kconfig implementation of this, either via options or choice.

PR #27398 (2020-09-10): nRF: Add system clock stability policy
Pull request adds clock stability policy configuration to the system clock (see #22758) and adds support to Nordic platform. Additionally, spin waiting for stable or available clock has been reworked for Nordic to allow going to idle while waiting for clock readiness (it can take up to 250 ms). Added test dedicated for Nordic platforms which validates correctness of behavior for each configuration against various clock sources.

Fixes #22758.
Fixes #26280.
closed 2020-09-10 by carlescufi