zephyrproject-rtos/zephyr #22784: system hangs in settings_load() nrf52840 custom board

teijokk, 2020-02-13:
**Describe the bug**
When try to call settings_load() before bt_le_adv_start() system hangs in custom board.
This works fine in nordic nrf52840 dev board.

**To Reproduce**
Steps to reproduce the behavior:
basicly samples/bluetooth/peripheral example works in DK but not in custom boad

**Expected behavior**
bt_le_adv_start() should return Zero and start advertising

**Impact**

**Screenshots or console output**

**Environment (please complete the following information):**
 - Linux Ubuntu 7.4.0
 - Toolchain zephyr-sdk-0.11.1
 
**Additional context**
If i understant correctly settings_load() tries to load/write bt_id from flash. Could this be flash related problem? Do i have to do something to flash (because this is custom made pcb)

nvlsianpu, 2020-02-13:
@teijokk I don't know hat a SoC you are using, what kind of flash is used as the storage or How your storage partition looks like in the DTS.
I see not enough info for help with issue you registered.

teijokk, 2020-02-14:
Ok, i have consulted hw designer. soc is nrf52840_qi and there are no different memory or peripheral versions in that soc

my board dts flash declaration(default that comes in zephyr)

&flash0 {
	/*
	 * For more information, see:
	 * http://docs.zephyrproject.org/latest/guides/dts/index.html#flash-partitions
	 */
	partitions {
		compatible = "fixed-partitions";
		#address-cells = <1>;
		#size-cells = <1>;

		boot_partition: partition@0 {
			label = "mcuboot";
			reg = <0x000000000 0x0000C000>;
		};
		slot0_partition: partition@c000 {
			label = "image-0";
			reg = <0x0000C000 0x000067000>;
		};
		slot1_partition: partition@73000 {
			label = "image-1";
			reg = <0x00073000 0x000067000>;
		};
		scratch_partition: partition@da000 {
			label = "image-scratch";
			reg = <0x000da000 0x0001e000>;
		};

		/*
		 * The flash starting at 0x000f8000 and ending at
		 * 0x000fffff is reserved for use by the application.
		 */

        /* Storage partition will be used by FCB/NFFS/NVS if enabled. */
		storage_partition: partition@f8000 {
			label = "storage";
			reg = <0x000f8000 0x00008000>;
		};
	};
};

Laczen, 2020-02-14:
@teijokk, have you tried erasing the complete flash before uploading your program?

teijokk, 2020-02-14:
I have been managed to debug the code in our custom board. bluetooth/peripheral example . Seems that problem is not settings_load() or anything flash related. Program enters err = bt_le_adv_start(BT_LE_ADV_CONN_NAME, ad, ARRAY_SIZE(ad), NULL, 0);
then in 
err = set_advertise_enable(false); (hci_core.c )
but then when called
err = bt_hci_cmd_send_sync(BT_HCI_OP_LE_SET_ADV_ENABLE, buf, NULL);(hci_core.c)
  if (err) {
    return err;
  }

program running  k_cpu_idle(); forever and not returns in  bt_le_adv_start()

teijokk, 2020-02-14:
can anyone give me advice what is the correct way to investigate this problem?  bluetooth/beacon example works fine, but we need connectable ad in our project and that clearly not working.

aescolar, 2020-02-14:
@cvinayak can you help @teijokk ?

Laczen, 2020-02-14:
@teijokk, I would still start with disabling settings. In the prj.conf file comment the last 7 lines by inserting # before each line.

nvlsianpu, 2020-02-14:
@joakimtoe can you help?

teijokk, 2020-02-14:
yes, have disabled setting in prj file.
#CONFIG_BT_KEYS_OVERWRITE_OLDEST=y
#CONFIG_BT_SETTINGS=y
#CONFIG_FLASH=y
#CONFIG_FLASH_PAGE_LAYOUT=y
#CONFIG_FLASH_MAP=y
#CONFIG_NVS=y
#CONFIG_SETTINGS=y

 Code works in nrf52840_pca10056 dev board but not custom board. Same symptoms that before.

nvlsianpu, 2020-02-14:
@joerchan can you help?

teijokk, 2020-02-14:
Hey, somethings changed. After let say 30-60 seconds minicom prints -> Advertising successfully started and i can confirm in nordic app that advertise works!

joerchan, 2020-02-14:
After 30 seconds? Can you please enable CONFIG_ASSERT?

teijokk, 2020-02-14:
enable asserts
CONFIG_ASSERT=y
no changes or any assert prints. i have tested 5 times and seems to work. time out is 30s when advertise starts. btw enabled CONFIG_BT_DEBUG_LOG=y 

Have to test if this affect perormace of system or is this delay only for when bt is starting.

*** Booting Zephyr OS build v2.2.0-rc1-83-g4f9b9a4500d8  ***                    
[00:00:00.006,744] <inf> fs_nvs: 8 SecAdvertising successfully started          
tors of 4096 bytes                                                              
[00:00:00.006,744] <inf> fs_nvs: alloc wra: 0, ff0                              
[00:00:00.006,744] <inf> fs_nvs: data wra: 0, 0                                 
[00:00:00.008,789] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.008,789] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)               
[00:00:00.008,789] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0
[00:00:00.009,216] <inf> bt_hci_core: No ID address. App must call settings_loa)
[00:00:00.016,754] <inf> bt_hci_core: Identity: f2:a0:9e:23:6a:97 (random)      
[00:00:00.016,754] <inf> bt_hci_core: HCI: version 5.1 (0x0a) revision 0x0000, 1
[00:00:00.016,754] <inf> bt_hci_core: LMP: version 5.1 (0x0a) subver 0xffff

joerchan, 2020-02-14:
@teijokk Could you check and possibly switch your clock control source?
CLOCK_CONTROL_NRF_K32SRC_RC
CLOCK_CONTROL_NRF_K32SRC_XTAL
Is  the two I would like to try and check if anyone of those have trouble.

Also please enable CONFIG_BT_DEBUG_HCI_CORE and CONFIG_BT_DEBUG_HCI_DRIVER.

ToJIka4, 2020-02-14:
@teijokk Could you share schematic?

At the beginning I'd choose `nrf52840 pca10056` board configuration and started my project. For board modification I put `nrf52840_pca10056.overlay` file in project folder. I had some issue with my custom board because of power supply. In the end i found solution with `CONFIG_BOARD_ENABLE_DCDC=n`. My board have 3.3V supply.

So maybe you choose ready board that is closest to you and start modifying it?

teijokk, 2020-02-14:
i've enabled this CONFIG_CLOCK_CONTROL_NRF_K32SRC_RC=y
Now system boot with no delay and i tested my own code with all sensor codes etc. all seems to work. There is external oscilator in scematics.... are chip using internal rc oscilator if this above settings is on?

cvinayak, 2020-02-14:
@teijokk  if your custom board has long track to the XTAL and your load capacitors are not tuned to correct values, XTAL will take longer time or not settle. Also, power consumption during this setup time will be higher. I am guessing your 32KHz XTAL is not settling.

joerchan, 2020-02-14:
@teijokk It is what I suspected. The advertiser is waiting for the LFCLK to start.

Using RC is the nrf52 internal one. using the XTAL is the external one. I would look into if the design of the external XTAL is correct and within the requirements.

Using the RC is an alternative though.

teijokk, 2020-02-14:
Ok i talk with hw designer about this.. thanks for your time, i'm really grateful your help!

closed 2020-02-14 by teijokk

joerchan, 2020-02-14:
Close as root cause has been found and is not a zephyr issue.

@teijokk For further assistence with LFCLK please open up issue on https://devzone.nordicsemi.com/ to get in touch with the right people.