zephyrproject-rtos/zephyr #22786: SM/MAS/PROT/BV-01-C FAIL

agata-ponitka, 2020-02-13:
Describe the bug
After timeout Zephyr send pairing failed. 

To Reproduce
run auto-pts

Expected behavior
After timeout Zephyr should ignore pairing response.

[SM_MAS_PROT_BV-01-C_FAIL.zip](https://github.com/zephyrproject-rtos/zephyr/files/4197935/SM_MAS_PROT_BV-01-C_FAIL.zip)

aescolar, 2020-02-13:
CC @joerchan

jhedberg, 2020-02-13:
Based on our analysis on Slack: It looks like commit 6a71ebf6c53 from @Vudentz is to blame, since it adds the call to `smp_pairing_complete(smp, BT_SMP_ERR_UNSPECIFIED);` after setting `SMP_FLAG_TIMEOUT`, and then `smp_pairing_complete()` calls `smp_reset()` which in turn calls `atomic_set(smp->flags, 0);`

joerchan, 2020-02-13:
In my opinion the change could be safely reverted. smp_pairing_complete with an error status is only going to call the security_changed and pairing_failed callbacks.

PR #22813 (2020-02-14): Bluetooth: SMP: Fix failing to pass SM/MAS/PROT/BV-01-C
smp_pairing_complete does actually clears flags so setting
SMP_FLAG_TIMEOUT must come after that.

Fixes #22786
commit 11564129 (2020-02-14): Bluetooth: SMP: Fix failing to pass SM/MAS/PROT/BV-01-C

smp_pairing_complete does actually clears flags so setting
SMP_FLAG_TIMEOUT must come after that.

Fixes #22786
closed 2020-02-14 by jhedberg