zephyrproject-rtos/zephyr #22801: Bluetooth: Split LL: Reconnection problem

MarekPieta, 2020-02-13:
**Describe the bug**
Problem was replicated using Zephyr Bluetooth peripheral device. I switched to Zephyr SW Split LL (previously I used SW Legacy LL). If whitelisting is enabled, the Central is connecting and bonding, but it is unable to reconnect. The reconnection problem was not observed for SW Legacy LL.
When I disable the whitelisting, the reconnection works fine too.

**To Reproduce**
1. Use Peripheral device with Zephyr SW split LL.
2. Connect to host, after the disconnection Central is unable to reconnect.

**Environment:**
Zephyr commit SHA: 40175fd3bdc2771b688d67d6f29dcc7a574c30ed

cvinayak, 2020-02-14:
I tried with iPhone, I see no issue with reconnection.

With nRF Connect for Desktop v3.3.0 with Bluetooth Low Energy v2.3.2, reconnection fails with MIC failure. It seems the desktop application is not using correct keys.

[reconnection.zip](https://github.com/zephyrproject-rtos/zephyr/files/4203070/reconnection.zip)

cvinayak, 2020-02-14:
Reproduce and analyzed it to be porting regression while adding conditional compilations.

PR #22824 (2020-02-14): Bluetooth: controller: split: Fix regression in privacy port
Fix regression due to addition of conditional compilations
while porting the privacy feature from legacy to split
controller.

Fixes #22801.
commit 23c5003d (2020-02-14): Bluetooth: controller: split: Fix regression in privacy port

Fix regression due to addition of conditional compilations
while porting the privacy feature from legacy to split
controller.

Fixes #22801.
closed 2020-02-14 by carlescufi