zephyrproject-rtos/zephyr #22803: k_delayed_work_cancel documentation inconsistent with behavior

pabigot, 2020-02-13:
https://github.com/zephyrproject-rtos/zephyr/blob/c6d37eba7bf82df12e41b2c4b65cc17d739ea48d/include/kernel.h#L3117 inaccurately documents the behavior of `k_delayed_work_cancel()`.  Specifically, if the work item has already completed the function will return zero, rather than `-EINVAL`.  Assuming the states of a work item:
* never submitted
* delayed
* pending
* processing
* completed
* cancelled

the only time `-EINVAL` would be observed is in never-submitted, or if it transitioned from pending to processing during the cancellation attempt.

The intent may have been to implement to the documentation; in that case the `work_q` value must be set to null before or after processing, and possibly other conditions need to be reviewed.

This had an impact in #22756 where it is not possible to determine whether a pending operation has been completed without maintaining an external flag.

pabigot, 2020-02-13:
@joerchan

joerchan, 2020-02-13:
Thanks.

For my use-case I think a function similar to k_work_pending for k_delayed_work would have been very useful. Between the timeout running out (k_delayed_work_remaining_get) and the timeout handler submitting (or not processed) it's no possible to check if the delayed work has been submitted, but not completed.

Vudentz, 2020-02-13:
So I was with the right impression on this one and the documentation is wrong in one way or the other, the question is shall we change the implementation to reflect what is stated in the documentation? That would mean cancel would return an error if neither the timer or pending flag is set.

Something like this:

`diff --git a/kernel/work_q.c b/kernel/work_q.c
index 7023f70f07..d9b259b50b 100644
--- a/kernel/work_q.c
+++ b/kernel/work_q.c
@@ -65,7 +65,11 @@ static int work_cancel(struct k_delayed_work *work)
                        return -EINVAL;
                }
        } else {
-               (void)z_abort_timeout(&work->timeout);
+               int err = z_abort_timeout(&work->timeout);
+
+               if (err) {
+                       return err;
+               }
        }
 
        /* Detach from workqueue */
`

PR #22810 (2020-02-17): kernel: work: Return error if timeout cannot be aborted
This is aligned with the documentation which states that an error shall
be returned if the work has been completed:

  '-EINVAL Work item is being processed or has completed its work.'

Fixes #22803
commit 038d727c (2020-02-17): kernel: work: Return error if timeout cannot be aborted

This is aligned with the documentation which states that an error shall
be returned if the work has been completed:

  '-EINVAL Work item is being processed or has completed its work.'

Though in order to be able to resubmit from the handler itself it needs
to be able to distinct when the work is already completed so instead of
-EINVAL it return -EALREADY when the work is considered to be completed.

Fixes #22803
closed 2020-02-17 by jhedberg