zephyrproject-rtos/zephyr #22814: mcuboot doesn't build with zephyr v2.1.0

lawrence-king, 2020-02-13:
mcuboot is required to use the CONFIG_BOOTLOADER_MCUBOOT=y  feature of Zephyr. Cloning mcuboot from either https://github.com/JuulLabs-OSS/mcuboot.git or https://github.com/zephyrproject-rtos/mcuboot.git ends up with something that won't compile.

west build ends up with the error:

> ../include/sysflash/sysflash.h:6:24: fatal error: devicetree.h: No such file or directory
>  #include <devicetree.h>
> 

To reproduce, run the following steps:

> cd zephyrproject/zephyr
> git checkout v2.1.0
> west update
> cd ../..
> git clone https://github.com/zephyrproject-rtos/mcuboot.git
> cd mcuboot
> pip3 install --user -r scripts/requirements.txt
> west build -b nrf52840_mdk
> 

At this point the build error above appears.

**Expected behavior**
A clean compile

**Impact**
Cannot build mcuboot

**Workaround**
```
git revert e96b6873bd8a64185796ac47b941f56dd29b6e3f
```
This is the cause of the compile error:

> commit e96b6873bd8a64185796ac47b941f56dd29b6e3f
> Author: Ulf Magnusson <Ulf.Magnusson@nordicsemi.no>
> Date:   Mon Jan 13 12:06:44 2020 +0100

**Screenshots or console output**
```
lawrence@VM:~/workspace/rc-demo/mcuboot/boot/zephyr$ west build
-- west build: build configuration:
       source directory: /home/lawrence/workspace/rc-demo/mcuboot/boot/zephyr
       build directory: /home/lawrence/workspace/rc-demo/mcuboot/boot/zephyr/build
       BOARD: nrf52840_mdk (origin: CMakeCache.txt)
-- west build: building application
[1/207] Building C object CMakeFiles/app.dir/home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/loader.c.obj
FAILED: CMakeFiles/app.dir/home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/loader.c.obj 
ccache /usr/bin/arm-none-eabi-gcc -DBUILD_VERSION=zephyr-v2.1.0 -DKERNEL -DMBEDTLS_CONFIG_FILE=\"mcuboot-mbedtls-cfg.h\" -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../include -I../targets -I/home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/include -I/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/include -I/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/include/drivers -Izephyr/include/generated -I/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52 -I/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/ext/hal/cmsis/Core/Include -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/hal/nordic/nrfx -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/hal/nordic/nrfx/drivers/include -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/hal/nordic/nrfx/mdk -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/hal/nordic/. -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/crypto/mbedtls/include -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/crypto/mbedtls/configs -isystem /home/lawrence/workspace/rc-demo/zephyrproject/zephyr/lib/libc/minimal/include -isystem /usr/lib/gcc/arm-none-eabi/6.3.1/include -isystem /usr/lib/gcc/arm-none-eabi/6.3.1/include-fixed -Og -imacros/home/lawrence/workspace/rc-demo/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mthumb -mcpu=cortex-m4 -imacros/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/loader.c.obj -MF CMakeFiles/app.dir/home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/loader.c.obj.d -o CMakeFiles/app.dir/home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/loader.c.obj   -c /home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/loader.c
In file included from /home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/bootutil_priv.h:27:0,
                 from /home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/loader.c:38:
../include/sysflash/sysflash.h:6:24: fatal error: devicetree.h: No such file or directory
 #include <devicetree.h>
                        ^
compilation terminated.
[2/207] Building C object CMakeFiles/app.dir/home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/swap_misc.c.obj
FAILED: CMakeFiles/app.dir/home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/swap_misc.c.obj 
ccache /usr/bin/arm-none-eabi-gcc -DBUILD_VERSION=zephyr-v2.1.0 -DKERNEL -DMBEDTLS_CONFIG_FILE=\"mcuboot-mbedtls-cfg.h\" -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../include -I../targets -I/home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/include -I/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/include -I/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/include/drivers -Izephyr/include/generated -I/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52 -I/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/ext/hal/cmsis/Core/Include -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/hal/nordic/nrfx -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/hal/nordic/nrfx/drivers/include -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/hal/nordic/nrfx/mdk -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/hal/nordic/. -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/crypto/mbedtls/include -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/crypto/mbedtls/configs -isystem /home/lawrence/workspace/rc-demo/zephyrproject/zephyr/lib/libc/minimal/include -isystem /usr/lib/gcc/arm-none-eabi/6.3.1/include -isystem /usr/lib/gcc/arm-none-eabi/6.3.1/include-fixed -Og -imacros/home/lawrence/workspace/rc-demo/mcuboot/boot/zephyr/build/zephyr/include/generated/autoconf.h -ffreestanding -fno-common -g -mthumb -mcpu=cortex-m4 -imacros/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/include/toolchain/zephyr_stdint.h -Wall -Wformat -Wformat-security -Wno-format-zero-length -Wno-main -Wno-pointer-sign -Wpointer-arith -Wno-unused-but-set-variable -Werror=implicit-int -fno-asynchronous-unwind-tables -fno-pie -fno-pic -fno-strict-overflow -fno-reorder-functions -fno-defer-pop -ffunction-sections -fdata-sections -mabi=aapcs -march=armv7e-m -std=c99 -nostdinc -MD -MT CMakeFiles/app.dir/home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/swap_misc.c.obj -MF CMakeFiles/app.dir/home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/swap_misc.c.obj.d -o CMakeFiles/app.dir/home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/swap_misc.c.obj   -c /home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/swap_misc.c
In file included from /home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/bootutil_priv.h:27:0,
                 from /home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/src/swap_misc.c:24:
../include/sysflash/sysflash.h:6:24: fatal error: devicetree.h: No such file or directory
 #include <devicetree.h>
                        ^
compilation terminated.
[3/207] Building C object CMakeFiles/app.dir/flash_map_extended.c.obj
FAILED: CMakeFiles/app.dir/flash_map_extended.c.obj 
ccache /usr/bin/arm-none-eabi-gcc -DBUILD_VERSION=zephyr-v2.1.0 -DKERNEL -DMBEDTLS_CONFIG_FILE=\"mcuboot-mbedtls-cfg.h\" -DNRF52840_XXAA -D_FORTIFY_SOURCE=2 -D__PROGRAM_START -D__ZEPHYR__=1 -I../include -I../targets -I/home/lawrence/workspace/rc-demo/mcuboot/boot/bootutil/include -I/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/include -I/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/include/drivers -Izephyr/include/generated -I/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/soc/arm/nordic_nrf/nrf52 -I/home/lawrence/workspace/rc-demo/zephyrproject/zephyr/ext/hal/cmsis/Core/Include -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/hal/nordic/nrfx -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/hal/nordic/nrfx/drivers/include -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/hal/nordic/nrfx/mdk -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/hal/nordic/. -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/crypto/mbedtls/include -I/home/lawrence/workspace/rc-demo/zephyrproject/modules/crypto/mbedtls/configs -isystem /home/lawrence/workspace/rc-demo/zephyrproject/zephyr/lib/libc/minimal/include -isystem /usr/lib/gcc/arm-none-eabi/6.3.1/include -isystem /usr/lib/gcc/arm-none-eabi/6.3.1/include-fixed -Og -imacros/home/lawren [... cut ...]
 #include <devicetree.h>
                        ^
compilation terminated.
ninja: build stopped: subcommand failed.
ERROR: command exited with status 1: /usr/local/bin/cmake --build /home/lawrence/workspace/rc-demo/mcuboot/boot/zephyr/build
lawrence@VM:~/workspace/rc-demo/mcuboot/boot/zephyr$ 

```

**Environment (please complete the following information):**
 - OS: Ubuntu 19.10
 - Toolchain: ZEPHYR_TOOLCHAIN_VARIANT=gnuarmemb
    GNUARMEMB_TOOLCHAIN_PATH=/usr
 - Commit SHA or Version used: master

carlescufi, 2020-02-14:
MCUboot's master is only compatible with Zephyr's master. If you need backwards compatibility you need to use Zephyr's [MCUboot repository](https://github.com/zephyrproject-rtos/mcuboot) at the revision that corresponds to the zephyr release.

carlescufi, 2020-02-14:
See this discussion on the MCUboot GitHub for more info:
https://github.com/JuulLabs-OSS/mcuboot/pull/659

closed 2020-02-14 by carlescufi

lawrence-king, 2020-02-14:
I am using the Zephyr's MCUboot repository<https://github.com/zephyrproject-rtos/mcuboot>, however there does not seem do be a way to determine the mcuboot revision that corresponds to the zephyr revision.

Zephyr release tags are:
zephyr-v2.1.0
zephyr-v2.0.0
zephyr-v1.14.1
zephyr-v1.13.0
zephyr-v1.12.0
zephyr-v1.11.0
And so on

MCUBoot release tags are:
v1.3.1
v1.3.0
v1.2.0
v1.1.0
v1.0.0
v0.9.0
and so on

What is the algorithm for matching? It would be best if this was handled by ‘west update’ just like all of the other modules from third parties.

Lawrence King
Principal Developer
+1(416)627-7302

carlescufi, 2020-02-14:
@lawrence-king it's in zephyr's `west.yml`:

```
$ git show v2.1.0:west.yml
[...]
```

but I now realize that when we released 2.1.0 we still did not have a zephyrproject-rots MCUboot repo and thus no entry in `west.yml`, so apologies for that. In practical terms, just applying the commit in https://github.com/JuulLabs-OSS/mcuboot/pull/659 should work.