zephyrproject-rtos/zephyr #22819: #define _current in kernel_structs.h leaks into global namespace

jensschroer, 2020-02-14:
**Describe the bug**
While trying to port a library to zephyr I noticed that `#define _current` in kernel_structs.h leaks into the global namespace. This manifested itself in a compile error of a class that included the zephyr header `posix/sys/socket.h` and had a variable called `_current` - a legitimate name for a variable from my point of view.

**To Reproduce**
Steps to reproduce the behavior:
1. create a simple application with this main.c
```
#include <zephyr.h>
#include <sys/printk.h>

#include <net/socket.h>

void main(void)
{
  int _current;
  printk("Hello World! %s\n", CONFIG_BOARD);
}
```
2. west build -b=frdm_k64f -o=-v
3. See error an error similiar to
```
In file included from /home/schroer/code/blixt/blixt-breaker/zephyr-zmq/zephyr/include/kernel_includes.h:29,
                 from /home/schroer/code/blixt/blixt-breaker/zephyr-zmq/zephyr/include/kernel.h:17,
                 from /home/schroer/code/blixt/blixt-breaker/zephyr-zmq/zephyr/include/zephyr.h:18,
                 from ../src/main.c:7:
../src/main.c: In function 'main':
/home/schroer/code/blixt/blixt-breaker/zephyr-zmq/zephyr/include/kernel_structs.h:205:25: error: expected '=', ',', ';', 'asm' or '__attribute__' before '.' token
  205 | #define _current _kernel.current
      |                         ^
../src/main.c:14:6: note: in expansion of macro '_current'
   14 |  int _current;
      |      ^~~~~~~~
/home/schroer/code/blixt/blixt-breaker/zephyr-zmq/zephyr/include/kernel_structs.h:205:25: error: expected expression before '.' token
  205 | #define _current _kernel.current
      |                         ^
../src/main.c:14:6: note: in expansion of macro '_current'
   14 |  int _current;
```

**Expected behavior**
The code should just compile without any errors.

**Environment (please complete the following information):**
 - Ubuntu 19.04
 - Zephyr SDK 0.11.1

andrewboie, 2020-02-17:
> and had a variable called _current - a legitimate name for a variable from my point of view.

@jensschroer no it's not. Variables that are prefixed with an underscore are reserved for the system. This is documented in the C standard. Don't do this in your application.

closed 2020-02-17 by andrewboie

stephanosio, 2020-02-18:
Note that this "leakage" is intentional, as some public macros and inline functions make direct use of this variable.

For instance:
https://github.com/zephyrproject-rtos/zephyr/blob/8f1780f6dee29dc2b2db01744e328a1a4a20d0b8/include/sys/mutex.h#L140

As @andrewboie already mentioned, you should not be using underscore-prefixed identifiers in your application.

andrewboie, 2020-02-18:
Eventually we should finishing the re-naming of private stuff to convert _current to z_current() (don't know where we left off on that) but in any event -- underscore-prefixed identifiers at the application level are a no-no any way you look at it