zephyrproject-rtos/zephyr #22855: drivers: enc28j60: waits for wrong interrupt

thedjnK, 2020-02-15:
**Describe the bug**
The enc28j60 driver has this code setting up the interrupt:

        gpio_pin_interrupt_configure(context->gpio,
                                     config->gpio_pin,
                                     GPIO_INT_EDGE_TO_ACTIVE);

Which indicates it is waiting for a low to high transition on the pin, however the user guide available from http://ww1.microchip.com/downloads/en/devicedoc/39662e.pdf clearly states that the interrupt pin is inverse, i.e. that ground signifies that an interrupt has occured (and will continue to be low until all interrupt flags have been services)

**To Reproduce**
Build the telnet example for the nrf52840_pca10056 with the enp28j60 driver used on SPI1 (debugging set to debug) and try to ping the device.

**Expected behavior**
The ping to be received by the board and a debug event showing that it has received it

**Actual behavior**
The module never recieves any packets ever, apparently, the debug output never includes any enc28j60 interrupts.

**My fix**
I changed GPIO_INT_EDGE_TO_ACTIVE to GPIO_INT_EDGE_TO_INACTIVE and I begin seeing the debug output from Zephyr regarding the enp28j60 driver and it successfully responds to ping requests

**Impact**
The enp28j60 driver is entirely non-functional

**Screenshots or console output**
Output without my fix: Well there is no output to display
Output with my fix:

    [00:00:06.248,718] <dbg> eth_enc28j60.eth_enc28j60_rx: Received packet of length 98
    [00:00:06.249,267] <dbg> eth_enc28j60.eth_enc28j60_tx: pkt 0x20004bfc (len 98)
    [00:00:06.250,122] <dbg> eth_enc28j60.eth_enc28j60_tx: Tx successful
    [00:00:07.251,190] <dbg> eth_enc28j60.eth_enc28j60_rx: Received packet of length 98
    [00:00:07.251,708] <dbg> eth_enc28j60.eth_enc28j60_tx: pkt 0x20004bfc (len 98)
    [00:00:07.252,593] <dbg> eth_enc28j60.eth_enc28j60_tx: Tx successful
    [00:00:08.255,737] <dbg> eth_enc28j60.eth_enc28j60_rx: Received packet of length 98
    [00:00:08.256,286] <dbg> eth_enc28j60.eth_enc28j60_tx: pkt 0x20004bfc (len 98)
    [00:00:08.257,171] <dbg> eth_enc28j60.eth_enc28j60_tx: Tx successful
    [00:00:10.351,745] <dbg> eth_enc28j60.eth_enc28j60_rx: Received packet of length 60
    [00:00:10.352,172] <dbg> eth_enc28j60.eth_enc28j60_tx: pkt 0x20004bfc (len 42)
    [00:00:10.352,905] <dbg> eth_enc28j60.eth_enc28j60_tx: Tx successful

**Environment (please complete the following information):**
 - OS: Arch linux
 - Toolchain: Zephyr SDK 0.11.2
 - Branch: master
 - Commit SHA: bf3850c270924c81fc7d89f3e4be9a90e41ccf74

jfischer-phytec-iot, 2020-02-16:
> I changed GPIO_INT_EDGE_TO_ACTIVE to GPIO_INT_EDGE_TO_INACTIVE and I begin

It is not a bug and not "wrong interrupt", your [int-gpios](https://github.com/zephyrproject-rtos/zephyr/blob/master/dts/bindings/ethernet/microchip%2Cenc28j60.yaml#L16) is not correct defined. Something like `int-gpios = <&arduino_header 15 GPIO_ACTIVE_LOW>; /* D9 */` would fix your issue.

closed 2020-02-16 by jfischer-phytec-iot

thedjnK, 2020-02-16:
@jfischer-phytec-iot Ah thank you, it seems I overlooked that note on the documentation, all is working.