zephyrproject-rtos/zephyr #22865: drivers: enc28j60: sample: dumb_http_server: TX failed errors

thedjnK, 2020-02-16:
**Describe the bug**
With an enc28j60 connected to a nordic nRF52840 using SPIM and the dumb_http_server sample (this also happens in the multithreaded one), the application seems to report a lot of errors whilst trying to send data, however the web page loads fine in the browser, so something is amiss.

**To Reproduce**
Build dumb_http_server for nrf52840_pca10056 with enc28j60 enabled on SPIM, run and connect to it with a web browser

**Expected behavior**
No errors

**Impact**
Not related to any projects

**Screenshots or console output**
Output on shell:

    Connection #1 from 192.168.1.65
    
    Connection from 192.168.1.65 closed
    
    [00:02:40.934,936] <dbg> eth_enc28j60.eth_enc28j60_rx: Received packet of length 74
    [00:02:40.935,516] <dbg> eth_enc28j60.eth_enc28j60_tx: pkt 0x20003bbc (len 54)
    [00:02:40.936,187] <err> eth_enc28j60: TX failed!
    [00:02:40.937,866] <dbg> eth_enc28j60.eth_enc28j60_rx: Received packet of length 60
    [00:02:40.939,208] <dbg> eth_enc28j60.eth_enc28j60_rx: Received packet of length 405
    [00:02:40.939,819] <dbg> eth_enc28j60.eth_enc28j60_tx: pkt 0x20003bbc (len 54)
    [00:02:40.940,429] <err> eth_enc28j60: TX failed!
    [00:02:40.951,293] <dbg> eth_enc28j60.eth_enc28j60_tx: pkt 0x20003bbc (len 1506)
    [00:02:40.953,826] <err> eth_enc28j60: TX failed!
    [00:02:40.953,857] <dbg> eth_enc28j60.eth_enc28j60_tx: pkt 0x20003c00 (len 783)
    [00:02:40.955,413] <err> eth_enc28j60: TX failed!
    [00:02:40.957,794] <dbg> eth_enc28j60.eth_enc28j60_rx: Received packet of length 60
    [00:02:41.151,367] <dbg> eth_enc28j60.eth_enc28j60_tx: pkt 0x20003bbc (len 1506)
    [00:02:41.154,083] <err> eth_enc28j60: TX failed!
    [00:02:41.212,646] <dbg> eth_enc28j60.eth_enc28j60_rx: Received packet of length 60
    [00:02:41.213,165] <dbg> eth_enc28j60.eth_enc28j60_tx: pkt 0x20003c44 (len 54)
    [00:02:41.213,775] <err> eth_enc28j60: TX failed!
    [00:02:41.215,515] <dbg> eth_enc28j60.eth_enc28j60_rx: Received packet of length 60
    [00:02:41.216,094] <dbg> eth_enc28j60.eth_enc28j60_tx: pkt 0x20003c44 (len 54)
    [00:02:41.216,735] <err> eth_enc28j60: TX failed!
    [00:02:46.076,354] <dbg> eth_enc28j60.eth_enc28j60_rx: Received packet of length 60
    [00:02:46.076,782] <dbg> eth_enc28j60.eth_enc28j60_tx: pkt 0x20003c44 (len 42)
    [00:02:46.077,392] <err> eth_enc28j60: TX failed!

I've captured the trace using `tcpdump` and it shows the system send the HTTP request, then a 'TCP previous segment not captured' continuation and then a 'TCP retransmission' - the messages are out of order, the first is the second half the HTTP data and the second packet is the first half of the data.

I am testing this from a remote laptop, connected to a raspberry pi 4 via Wi-Fi, connected to a 100Mbps huawei modem on 1 port and the enc28j60  (at 10Mbps) on the other port.

**Environment (please complete the following information):**
 - OS: Laptop: arch linux, raspberry pi: arch linux arm 64-bit
 - Toolchain: Zephyr SDK 0.11.2
 - Branch: master
 - Commit SHA: bf3850c270924c81fc7d89f3e4be9a90e41ccf74

jita101, 2020-03-23:
Just to double this i.e. I'm also seeing the same issue with this sample and particle_xenon (nrf52840) and nrf52_pca10040 (nrf52832) board and two different generic enc28j60 shields.

That said I'm not having similar (on infact any) issues sending out millions of smaller TCP packets (MQTT) with these boards and enc28j60. The boards work reliably for days sending out packets full speed.

thedjnK, 2020-08-03:
Haven't tested again with enc28j60 but have just tried with an enc624j600 with the same MCU (nRF52840) and latest nRF connect SDK and I see no errors at all (also using fly-leaded wires - same length as was tested with enc28j60) so this might narrow down where the issue is, although someone might want to test on the current master and see if there are any errors at all anymore as it might have already been resolved.

carlescufi, 2020-09-08:
Closing this since it does not seem reproduceable anymore.

closed 2020-09-08 by carlescufi