zephyrproject-rtos/zephyr #22866: Amazon Web Services sample

md-jamal, 2020-02-16:
Hi ,
I dont see an example related to publishing our data to Amazon Web Services
https://docs.zephyrproject.org/latest/samples/index.html

If we have a sample it will be really helpful

rveerama1, 2020-02-19:
Right now we have MQTT samples for Google IoT and Azure cloud. https://github.com/zephyrproject-rtos/zephyr/tree/master/samples/net/cloud

md-jamal, 2020-02-20:
can we modify them and use it for AWS

rveerama1, 2020-02-21:
You can use those samples as a template for AWS. Authentication and Message publication might be different.

md-jamal, 2020-02-24:
Can we use "AWS IoT Device SDK for Embedded C" in Zephyr

md-jamal, 2020-02-29:
Hi,

I wrote this code:

```
/*
 * Copyright (c) 2019 Intel Corporation
 *
 * SPDX-License-Identifier: Apache-2.0
 */

#include <logging/log.h>
LOG_MODULE_REGISTER(aws_mqtt, LOG_LEVEL_DBG);

#include <zephyr.h>
#include <net/socket.h>
#include <net/mqtt.h>
#include <net/net_config.h>
#include <net/net_event.h>
#include <net/sntp.h>

#include <sys/printk.h>
#include <string.h>
#include <errno.h>
#include "dhcp.h"
#include <time.h>
#include <inttypes.h>
#include "globalsign.inc"

/* The mqtt client struct */
static struct mqtt_client client_ctx;
s64_t time_base;
/* MQTT Broker details. */
static struct sockaddr_storage broker;

#if defined(CONFIG_SOCKS)
static struct sockaddr socks5_proxy;
#endif
static struct pollfd fds[1];
static int nfds;

#define APP_MQTT_BUFFER_SIZE    128

static sec_tag_t m_sec_tags[] = {
#if defined(MBEDTLS_X509_CRT_PARSE_C) || defined(CONFIG_NET_SOCKETS_OFFLOAD)
	APP_CA_CERT_TAG,
#endif
#if defined(MBEDTLS_KEY_EXCHANGE__SOME__PSK_ENABLED)
	APP_PSK_TAG,
#endif
};
#define APP_CA_CERT_TAG 1
#define APP_PSK_TAG 2

/* Buffers for MQTT client. */
static u8_t rx_buffer[APP_MQTT_BUFFER_SIZE];
static u8_t tx_buffer[APP_MQTT_BUFFER_SIZE];
static bool connected;

#define MQTT_CLIENTID           "zephyr_publisher"

struct zsock_addrinfo *haddr;

time_t my_k_time(time_t *ptr)
{
	s64_t stamp;
	time_t now;

	stamp = k_uptime_get();
	now = (time_t)((stamp + time_base) / 1000);
[... 335 lines cut ...]
		return;
	}

	LOG_INF("DNS resolved for console.aws.amazon.com:80");
	show_addrinfo(haddr);
	do_sntp(haddr);
	mqtt_startup("agwgaprlb6xdl-ats.iot.us-west-2.amazonaws.com", 8883);	

}
```

It is failing at mqtt_input:

```
[00:00:07.219,000] <dbg> net_mqtt.client_connect: (0x20401bac): Connect completed
[00:00:07.684,000] <dbg> net_mqtt.mqtt_input: (0x20401bac): state:0x00000002
[00:00:07.693,000] <dbg> net_mqtt_rx.mqtt_read_message_chunk: (0x20401bac): [CID 0x20400254]: Connection closed.
[00:00:07.706,000] <dbg> net_mqtt_sock_tcp.mqtt_client_tcp_disconnect: (0x20401bac): Closing socket 0
[00:00:07.718,000] <err> aws_mqtt: MQTT connect failed -111
[00:00:07.726,000] <inf> aws_mqtt: Aborting connection

[00:00:07.734,000] <inf> aws_mqtt: try_to_connect: 0 <OK>

Can you please provide any input on this
```

luispatsko-trimble, 2021-11-18:
@md-jamal Have you succeeded in using "AWS IoT Device SDK for Embedded C" in Zephyr? We're planning to use it too, and I'd like to know how was your experience porting it to Zephyr.

PR #57635 (2023-05-10): samples: net: mqtt: Add a sample to connect to AWS IoT core
I would like to propose the addition of a new sample application to Zephyr RTOS, which implements an MQTT client that can publish messages to AWS IoT Core using the MQTT protocol. The sample showcases the implementation of key features such as establishing a TLS 1.2 connection with AWS IoT Core, subscribing to a topic, and publishing data. Additionally, this sample is designed to pass the test suite for the AWS Device Qualification Program (DQP).

This sample can be used as a starting point for developers who are interested in buildin [... cut]
commit e803b774 (2023-05-10): samples: net: mqtt: Add a sample to connect to AWS IoT core

This sample application demonstrates the implementation of an
MQTT client that can publish messages to AWS IoT Core using
the MQTT protocol.

Fixes #22866
closed 2023-05-10 by carlescufi