zephyrproject-rtos/zephyr #22870: Add Cortex-M4 testing platform

stephanosio, 2020-02-17:
Currently, there is no emulated testing platform available for the floating point-capable ARM M-profile devices (i.e. Cortex-M4 and above).

Since the QEMU 4.3.0 will add the `netduinoplus2` machine, which emulates the STM32F405 SoC that embeds the Cortex-M4F core, once released and integrated into the Zephyr SDK, add the `qemu_cortex_m4` to serve as the primary testing platform for the ARMv7-M devices.

In addition, once we add `qemu_cortex_m4`, remove `qemu_cortex_m3` as `qemu_cortex_m4` is a superset of it and `mps2_an385` already serves as the default testing platform for the Cortex-M3 devices.

Notes:
* `netduinoplus2` board definition:
https://github.com/qemu/qemu/blob/971b2a1e5b1a8cc8f597ac5d7016908f9fa880de/hw/arm/netduinoplus2.c
* QEMU Cortex-M4 emulation supports both VFPv4 and DSP instructions:
https://github.com/qemu/qemu/blob/971b2a1e5b1a8cc8f597ac5d7016908f9fa880de/target/arm/cpu.c#L2107-L2108

stephanosio, 2020-02-17:
cc @ioannisg @galak

ioannisg, 2020-02-17:
Nice to do for next release

stephanosio, 2020-02-25:
https://wiki.qemu.org/ChangeLog/5.0
https://wiki.qemu.org/Planning/5.0

stephanosio, 2020-02-27:
NOTE: Running STM32F405 Zephyr image on QEMU `netduinoplus2` requires some (not upstreamed yet) patches:

https://lists.gnu.org/archive/html/qemu-devel/2020-02/msg07841.html
https://lists.gnu.org/archive/html/qemu-devel/2020-02/msg07842.html

In summary, Zephyr links code and rodata in the flash at 0x8000000, which is aliased to 0x0. QEMU has a bug in which the ROM reset handler is executed after the CPU reset handler, and this causes the aliased memory data to be unavailable at the time of CPU vector table read and SP/PC initialisation- leading to boot failure.

stephanosio, 2020-03-01:
NOTE: Running STM32F405 Zephyr image on QEMU `netduinoplus2` requires some (not upstreamed yet) patches:

https://lists.gnu.org/archive/html/qemu-devel/2020-02/msg07841.html
https://lists.gnu.org/archive/html/qemu-devel/2020-02/msg07842.html

In summary, Zephyr links code and rodata in the flash at 0x8000000, which is aliased to 0x0. QEMU has a bug in which the ROM reset handler is executed after the CPU reset handler, and this causes the aliased memory data to be unavailable at the time of CPU vector table read and SP/PC initialisation- leading to boot failure.

In addition, the STM32Cube driver used by Zephyr expects the RCC and FLASH peripheral registers, which are currently not emulated on the QEMU, to function properly. The following patches add emulation support for these two peripherals:

https://lists.gnu.org/archive/html/qemu-devel/2020-02/msg08453.html
https://lists.gnu.org/archive/html/qemu-devel/2020-03/msg00001.html

nashif, 2021-05-28:
@stephanosio are you still looking into that?

stephanosio, 2021-05-28:
@ioannisg Have we managed to get an alternative for this? Or do we still need this?

ioannisg, 2021-05-29:
> @ioannisg Have we managed to get an alternative for this? Or do we still need this?

@stephanosio I have an open pr for qemu support on cortex-m33. So we will cover fpu testing on a slightly different ARCH though... Maybe worth keeping this open

stephanosio, 2021-05-30:
> @stephanosio I have an open pr for qemu support on cortex-m33. So we will cover fpu testing on a slightly different ARCH though... Maybe worth keeping this open

Comparing the QEMU emulation feature sets of Cortex-M4 and Cortex-M33:

**Cortex-M4**
```c
    set_feature(&cpu->env, ARM_FEATURE_V7);
    set_feature(&cpu->env, ARM_FEATURE_M);
    set_feature(&cpu->env, ARM_FEATURE_M_MAIN);
    set_feature(&cpu->env, ARM_FEATURE_THUMB_DSP);
    set_feature(&cpu->env, ARM_FEATURE_VFP4);
```

**Cortex-M33**
```c
    set_feature(&cpu->env, ARM_FEATURE_V8);
    set_feature(&cpu->env, ARM_FEATURE_M);
    set_feature(&cpu->env, ARM_FEATURE_M_MAIN);
    set_feature(&cpu->env, ARM_FEATURE_M_SECURITY);
    set_feature(&cpu->env, ARM_FEATURE_THUMB_DSP);
    set_feature(&cpu->env, ARM_FEATURE_VFP4);
```

As far as the QEMU emulation goes, Cortex-M33F can be thought of as a simple superset of Cortex-M4F, and considering the main reason for adding M4F support was to facilitate Cortex-M floating point testing in Zephyr, I am not sure if there is any major advantage in adding M4F support alongside your planned M33 FP support.

I don't think there is anything Cortex-M4F emulation can offer that Cortex-M33F emulation cannot. Do you agree? @ioannisg

ioannisg, 2021-05-31:
> I don't think there is anything Cortex-M4F emulation can offer that Cortex-M33F emulation cannot. Do you agree? @ioannisg

What I was thinking of is the combination of FPU along with the (legacy) MPU architecture of Cortex-M4. If we care about that, we could only test it on M4F (or M7F), but not M33F.

stephanosio, 2021-06-01:
> What I was thinking of is the combination of FPU along with the (legacy) MPU architecture of Cortex-M4. If we care about that, we could only test it on M4F (or M7F), but not M33F.

In that case, I will look into resurrecting this PR.

stephanosio, 2021-08-13:
Marking it low priority since #35381 already addresses the main problem at hand, which was lack of ARM hard FP testing in the CI.

stephanosio, 2022-05-04:
Although #35381 introduced the `mps2_an521_remote` target, which executes code on the CPU 1, a Cortex-M33F core that supports DSP and single-precision FPU, it has some limitations:

1. Its code memory is limited to 512K due to the TFM partitioning, which is not enough for some tests (e.g. CMSIS-DSP).

    https://github.com/zephyrproject-rtos/zephyr/blob/88ca4fb5336151f11857f966bb8ed1e11bf1962d/boards/arm/mps2_an521/mps2_an521_remote.dts#L104-L106

1. It does not support double precision FPU (Cortex-M33 core cannot be configured with double-precision support).

[QEMU introduced the MPS2-AN386 machine type](https://github.com/qemu/qemu/commit/897d27260a7eaccbf4ff01e49021205c5616c8ef), which includes a Cortex-M4 core, so consider adding the support for it.

Also there exists [MPS2-AN500](https://github.com/qemu/qemu/commit/6d4811c4b688de368748c8095250ba12c905a21c), which includes a Cortex-M7 core, so consider adding it also in order to support double-precision coverage.

stephanosio, 2022-05-04:
Closing this in favour of https://github.com/zephyrproject-rtos/zephyr/issues/45319

closed 2022-05-04 by stephanosio