zephyrproject-rtos/zephyr #22872: Hello world application for mps2_an521 board when build as a secure/non-secure with Trusted Firmware is crashing on qemu

musamaanjum, 2020-02-17:
I can build hello world in secure mode and run on the arm qemu(4.2) successfully.
```
west build -b mps2_an521 samples/hello_world
qemu-system-arm -M mps2-an521 -m 1G -kernel build/zephyr/zephyr.elf
```

But when I build a secure/non-secure hello world demo with Trusted Firmware and launch on the same qemu, it crashes. As given in the docs of this board, I've build Trusted Firmware M with ARMCompiler6.10.1 and when I'd tried copying the built resources from tfm/build/install directory to zephyr/src/ext as asked in the baord docs, I got the error that these directories doesn't exist. So I created src and then ext manually in the base zephyr directory. Then ran the following command again:
```
cp -r install/ $ZEPHYR_PROJECT/src/ext
cp $ZEPHYR_PROJECT/src/ext/install/export/tfm/veneers/s_veneers.o $ZEPHYR_PROJECT/src/ext
```
Then I ran:
```
west build -b mps2_an521_nonsecure samples/hello_world
qemu-system-arm -M mps2-an521 -m 1G -kernel build/zephyr/zephyr.elf
```
This time qemu crashed with the following coredump:
```
qemu-system-arm -M mps2-an521 -m 1G -kernel build/zephyr/zephyr.elf

qemu: fatal: Lockup: can't escalate 3 to HardFault (current priority -1)

R00=00000000 R01=00000000 R02=00000000 R03=00000000
R04=00000000 R05=00000000 R06=00000000 R07=00000000
R08=00000000 R09=00000000 R10=00000000 R11=00000000
R12=00000000 R13=ffffffe0 R14=fffffff9 R15=00000000
XPSR=40000003 -Z-- A S handler
FPSCR: 00000000
[1]    12909 abort (core dumped)  qemu-system-arm -M mps2-an521 -m 1G -kernel build/zephyr/zephyr.elf
```

**To Reproduce**
Steps to reproduce the behavior:
1. Build TFM
2. Run:
cp -r install/ $ZEPHYR_PROJECT/src/ext
cp $ZEPHYR_PROJECT/src/ext/install/export/tfm/veneers/s_veneers.o $ZEPHYR_PROJECT/src/ext
3. west build -b mps2_an521_nonsecure samples/hello_world
4. qemu-system-arm -M mps2-an521 -m 1G -kernel build/zephyr/zephyr.elf
5. See the crash

Environment
 - OS - Ubuntu
 - Toolchain - Zephyr SDK
 - Commit SHA : bf3850c27

microbuilder, 2020-02-25:
This is related to ongoing work in #19985 ... the TF-M binary needs some extra work to be run in qemu, which is specifically happening here: https://github.com/zephyrproject-rtos/zephyr/pull/19985/commits/04bae707e6fdb142cde105ac3ba72dd4640817df

That merged binary can then be run via `qemu-system-arm -M mps2-an521 -device loader,file=tfm_qemu.hex -serial stdio`

Once that is  updated and merged in, this issue can be closed (in theory). The remaining issue with the qemu binary is updating the `make run` command to call the corrrect qemu binary, rather than the default `zephyr.bin` (etc.) binary.

ioannisg, 2020-05-11:
The work in #19985 is now merged, so I wonder if this ticket is still valid.
@musamaanjum , could you revisit the issue?

microbuilder, 2020-05-11:
Trying to run the hello_world (or other non-TFM) sample example on the `_nonsecure` board, meaning any example outside `samples/tfm_integration/*` at present, will fail because the `mps2_an521_nonsecure` board target makes certain assumptions about TF-M being present on the S side as a post-build step that aren't being met: https://github.com/zephyrproject-rtos/zephyr/blob/master/boards/arm/mps2_an521/CMakeLists.txt#L34

The `_nonsecure` target should only be used with TF-M samples that manage the TF-M build themselves, for example: `west build -p -b mps2_an521_nonsecure samples/tfm_integration/tfm_ipc -t run`.

For **non-TFM** examples, the `mps_an521` target should be used, which does run out of the box with QEMU:

```
$ west build -p -b mps_an521 samples/hello_world/ -t run
...
[124/130] Linking C executable zephyr/zephyr_prebuilt.elf
Memory region         Used Size  Region Size  %age Used
           FLASH:       12064 B       224 MB      0.01%
            SRAM:        4352 B        16 MB      0.03%
        IDT_LIST:         200 B         2 KB      9.77%
[129/130] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: cortex-m33
qemu-system-arm: warning: nic lan9118.0 has no peer
*** Booting Zephyr OS build v2.3.0-rc1  ***
Hello World! mps2_an521
```

@ioannisg Should using the `_nonsecure` board definition on non-TFM samples and executing them in QEMU, which DOES result in the error mentionned here (I just tested), be considered a bug? Should we reject any attempts to use the `_nonsecure` board target when `CONFIG_BUILD_WITH_TFM` is not set?

closed 2020-07-31 by github-actions[bot]