zephyrproject-rtos/zephyr #22873: Bluetooth: RSSI Read command can be configured out even when mandatory

wopu-ot, 2020-02-17:
**Describe the bug**
When BT_CONN is enabled, the Kconfig option BT_CTLR_CONN_RSSI can be set to 'n', which disables support for the Read RSSI command. However, the Read RSSI command is mandatory if the Connection State is supported, leading to a configuration that cannot possibly pass qualification.

**To Reproduce**
Build with CONFIG_BT_PERIPHERAL=y and CONFIG_BT_CTLR_CONN_RSSI=n.

**Expected behavior**
The configuration system should not allow configurations that are obviously invalid. There might be a case to disable the Read RSSI command in combined host/controller builds if it is known to be unused. However, the Read RSSI command must be available if BT_CONN and BT_HCI_RAW are set.

**Impact**
Very low. Nothing prevents the user from selecting a valid configuration. However, users that set  CONFIG_BT_CTLR_CONN_RSSI=n may be surprised that the configuration is not qualifiable.

aescolar, 2020-02-17:
CC: @cvinayak @kruithofa @carlescufi @joerchan

cvinayak, 2020-02-17:
CONFIG_BT_CTLR_CONN_RSSI consumes both more current and code space for applications that do not need to poll/query RSSI using HCI command. The default is CONFIG_BT_CTLR_CONN_RSSI=y when CONFIG_BT_HCI_RAW is selected, and applications should be allowed to deselect.

joerchan, 2020-02-17:
I agree with vinayak. This would only be an issue if we are making an HCI product, and the user would have to select an HCI qualifiable configuration. For host+controller build this is not a problem

wopu-ot, 2020-02-17:
I am fine with having this configurable in combined builds. However, I believe that BT_HCI_RAW should enforce BT_CTLR_CONN_RSSI instead of just defaulting it to 'y'.

joerchan, 2020-02-17:
Is BT_HCI_RAW really the right option to force HCI conformant configurations?

@jhedberg Any comments or ideas?

jhedberg, 2020-02-17:
> Is BT_HCI_RAW really the right option to force HCI conformant configurations?

That sounds like a fair assumption to me. `BT_HCI_RAW` is essentially for building the controller as an independent (and independently qualified) module, with HCI as the interface to talk to this module. For combined builds we know exactly what kind of host is paired with what kind of controller, and will anyway not be qualifying HCI separately (even though it exists as an internal interface).

PR #26852 (2020-07-16): Bluetooth: controller: Enforce support for Read RSSI command in Kconfig
Enforce support for Read RSSI command if BT_CONN and BT_HCI_RAW are
selected. According to the Bluetooth specification, support for the
Read RSSI command is mandatory if the Connection State is
supported. BT_HCI_RAW indicates that the controller is an
independently qualifiable module and not part of a combined
host/controller build. This change avoids a configuration that cannot
be possibly qualified.

Fixes #22873
closed 2020-07-16 by carlescufi