zephyrproject-rtos/zephyr #22874: sanitycheck: when someone instance get stuck because of concurrent.futures.TimeoutErro exception, it always stuck

wentongwu, 2020-02-17:
**Describe the bug**
when someone instance get stuck because of concurrent.futures.TimeoutErro exception, it always stuck.

```
NFO    - 74/86 mps2_an385                tests/kernel/pending/kernel.objects                PASSED (qemu 2.189s)
INFO    - 75/86 mps2_an385                tests/kernel/lifo/lifo_api/kernel.lifo             PASSED (qemu 2.229s)
INFO    - 76/86 mps2_an385                tests/kernel/lifo/lifo_usage/kernel.lifo.usage     PASSED (qemu 2.218s)
WARNING - mps2_an385/tests/kernel/timer/timer_api/kernel.timer stuck?
WARNING - mps2_an385/tests/kernel/sched/preempt/kernel.scheduler.preempt stuck?
WARNING - mps2_an385/tests/kernel/interrupt/arch.interrupt stuck?
WARNING - mps2_an385/tests/kernel/critical/kernel.common.critical stuck?
WARNING - mps2_an385/tests/kernel/workq/work_queue_api/kernel.workqueue.api stuck?
WARNING - mps2_an385/tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing stuck?
INFO    - 77/86 mps2_an385                tests/kernel/stack/stack/kernel.stack.usage        PASSED (qemu 2.111s)
INFO    - 78/86 mps2_an385                tests/kernel/mutex/mutex_api/kernel.mutex          PASSED (qemu 2.096s)
WARNING - mps2_an385/tests/kernel/timer/timer_api/kernel.timer stuck?
WARNING - mps2_an385/tests/kernel/sched/preempt/kernel.scheduler.preempt stuck?
WARNING - mps2_an385/tests/kernel/interrupt/arch.interrupt stuck?
WARNING - mps2_an385/tests/kernel/critical/kernel.common.critical stuck?
WARNING - mps2_an385/tests/kernel/workq/work_queue_api/kernel.workqueue.api stuck?
WARNING - mps2_an385/tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing stuck?
INFO    - 79/86 mps2_an385                tests/kernel/mutex/sys_mutex/system.mutex          PASSED (qemu 2.072s)
INFO    - 80/86 mps2_an385                tests/kernel/mutex/sys_mutex/system.mutex.nouser   PASSED (qemu 2.089s)
WARNING - mps2_an385/tests/kernel/timer/timer_api/kernel.timer stuck?
WARNING - mps2_an385/tests/kernel/sched/preempt/kernel.scheduler.preempt stuck?
WARNING - mps2_an385/tests/kernel/interrupt/arch.interrupt stuck?
WARNING - mps2_an385/tests/kernel/critical/kernel.common.critical stuck?
WARNING - mps2_an385/tests/kernel/workq/work_queue_api/kernel.workqueue.api stuck?
WARNING - mps2_an385/tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing stuck?
WARNING - mps2_an385/tests/kernel/timer/timer_api/kernel.timer stuck?
WARNING - mps2_an385/tests/kernel/sched/preempt/kernel.scheduler.preempt stuck?
WARNING - mps2_an385/tests/kernel/interrupt/arch.interrupt stuck?
WARNING - mps2_an385/tests/kernel/critical/kernel.common.critical stuck?
WARNING - mps2_an385/tests/kernel/workq/work_queue_api/kernel.workqueue.api stuck?
WARNING - mps2_an385/tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing stuck?
WARNING - mps2_an385/tests/kernel/timer/timer_api/kernel.timer stuck?
WARNING - mps2_an385/tests/kernel/sched/preempt/kernel.scheduler.preempt stuck?
WARNING - mps2_an385/tests/kernel/interrupt/arch.interrupt stuck?
WARNING - mps2_an385/tests/kernel/critical/kernel.common.critical stuck?

WARNING - mps2_an385/tests/kernel/workq/work_queue_api/kernel.workqueue.api stuck?
WARNING - mps2_an385/tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing stuck?
WARNING - mps2_an385/tests/kernel/timer/timer_api/kernel.timer stuck?
WARNING - mps2_an385/tests/kernel/sched/preempt/kernel.scheduler.preempt stuck?
WARNING - mps2_an385/tests/kernel/interrupt/arch.interrupt stuck?
WARNING - mps2_an385/tests/kernel/critical/kernel.common.critical stuck?
WARNING - mps2_an385/tests/kernel/workq/work_queue_api/kernel.workqueue.api stuck?
WARNING - mps2_an385/tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing stuck?
WARNING - mps2_an385/tests/kernel/timer/timer_api/kernel.timer stuck?
WARNING - mps2_an385/tests/kernel/sched/preempt/kernel.scheduler.preempt stuck?
WARNING - mps2_an385/tests/kernel/interrupt/arch.interrupt stuck?
WARNING - mps2_an385/tests/kernel/critical/kernel.common.critical stuck?
WARNING - mps2_an385/tests/kernel/workq/work_queue_api/kernel.workqueue.api stuck?
WARNING - mps2_an385/tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing stuck?
WARNING - mps2_an385/tests/kernel/timer/timer_api/kernel.timer stuck?
WARNING - mps2_an385/tests/kernel/sched/preempt/kernel.scheduler.preempt stuck?
WARNING - mps2_an385/tests/kernel/interrupt/arch.interrupt stuck?
WARNING - mps2_an385/tests/kernel/critical/kernel.common.critical stuck?
WARNING - mps2_an385/tests/kernel/workq/work_queue_api/kernel.workqueue.api stuck?
WARNING - mps2_an385/tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing stuck?
WARNING - mps2_an385/tests/kernel/timer/timer_api/kernel.timer stuck?
WARNING - mps2_an385/tests/kernel/sched/preempt/kernel.scheduler.preempt stuck?
WARNING - mps2_an385/tests/kernel/interrupt/arch.interrupt stuck?
WARNING - mps2_an385/tests/kernel/critical/kernel.common.critical stuck?
WARNING - mps2_an385/tests/kernel/workq/work_queue_api/kernel.workqueue.api stuck?
WARNING - mps2_an385/tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing stuck?
WARNING - mps2_an385/tests/kernel/timer/timer_api/kernel.timer stuck?
WARNING - mps2_an385/tests/kernel/sched/preempt/kernel.scheduler.preempt stuck?
WARNING - mps2_an385/tests/kernel/interrupt/arch.interrupt stuck?
WARNING - mps2_an385/tests/kernel/critical/kernel.common.critical stuck?
[... 62 lines cut ...]
WARNING - mps2_an385/tests/kernel/timer/timer_api/kernel.timer stuck?
WARNING - mps2_an385/tests/kernel/sched/preempt/kernel.scheduler.preempt stuck?
WARNING - mps2_an385/tests/kernel/interrupt/arch.interrupt stuck?
WARNING - mps2_an385/tests/kernel/critical/kernel.common.critical stuck?
WARNING - mps2_an385/tests/kernel/workq/work_queue_api/kernel.workqueue.api stuck?
WARNING - mps2_an385/tests/kernel/sched/schedule_api/kernel.scheduler.no_timeslicing stuck?
WARNING - mps2_an385/tests/kernel/timer/timer_api/kernel.timer stuck?
WARNING - mps2_an385/tests/kernel/sched/preempt/kernel.scheduler.preempt stuck?
WARNING - mps2_an385/tests/kernel/interrupt/arch.interrupt stuck?
WARNING - mps2_an385/tests/kernel/critical/kernel.common.critical stuck?
WARNING - mps2_an385/tests/kernel/workq/work_queue_api/kernel.workqueue.api stuck?
```

**Screenshots or console output**
If applicable, add a screenshot (drag-and-drop an image), or console logs
(cut-and-paste text and put a code fence (\`\`\`) before and after, to help
explain the issue.

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows)
 - Toolchain (e.g Zephyr SDK, ...)
 - Commit SHA or Version used

**Additional context**
Add any other context about the problem here.

wentongwu, 2020-02-17:
And I'm sure these stuck labeled case has been executed successfully.

wentongwu, 2020-02-17:
```
        with BoundedExecutor(bound=self.jobs, max_workers=self.jobs) as executor:

            # start a future for a thread which sends work in through the queue
            future_to_test = {
                executor.submit(self.add_tasks_to_queue, self.test_only): 'FEEDER DONE'}

            while future_to_test:
                # check for status of the futures which are currently working
                done, pending = concurrent.futures.wait(future_to_test, timeout=1,
                    return_when=concurrent.futures.FIRST_COMPLETED)

                # if there is incoming work, start a new future
                while not pipeline.empty():
                    # fetch a url from the queue
                    message = pipeline.get()
                    test = message['test']

                    pb = ProjectBuilder(self,
                                        test,
                                        lsan=self.enable_lsan,
                                        asan=self.enable_asan,
                                        coverage=self.enable_coverage,
                                        extra_args=self.extra_args,
                                        device_testing=self.device_testing,
                                        cmake_only=self.cmake_only,
                                        cleanup=self.cleanup,
                                        valgrind=self.enable_valgrind,
                                        inline_logs=self.inline_logs
                                        )
                    future_to_test[executor.submit(pb.process, message)] = test.name

                # process any completed futures
                for future in done:
                    test = future_to_test[future]
                    try:
                        data = future.result()
                    except Exception as exc:
                        logger.error('%r generated an exception: %s' % (test, exc))
                        sys.exit('%r generated an exception: %s' % (test, exc))

                    else:
                        if data:
                            logger.debug(data)

                    # remove the now completed future
                    del future_to_test[future]

                for future in pending:
                    test = future_to_test[future]

                    try:
                        future.result(timeout=180)
                    except concurrent.futures.TimeoutError:
                        logger.warning("{} stuck?".format(test))

```

consider above code a while, didn't find anything wrong so far, do you have any idea? @nashif

wentongwu, 2020-02-17:
And by the way, why the timeout value is chosen 180 seconds?

```
                for future in pending:
                    test = future_to_test[future]

                    try:
                        future.result(timeout=180)
                    except concurrent.futures.TimeoutError:
                        logger.warning("{} stuck?".format(test))
```

nashif, 2020-02-18:
this is not a bug, it is just warning used for debugging tests that are taking too long, either by design or because something is really stuck, I suggest you ignore this while we work on resolving the real issues.

nashif, 2020-02-18:
> And I'm sure these stuck labeled case has been executed successfully in the final, but sanity-check still give warning it get stuck, never stop.

did you notice it is a warning and it is asking a question and not making a statement? This is just a message used to give the user some hint that there are jobs running that are taking longer than usual.

closed 2020-02-19 by wentongwu