zephyrproject-rtos/zephyr #22885: Sanitycheck timeout all test cases on mec15xxevb_assy6853 board.

chen-png, 2020-02-17:
**Describe the bug**
using sanitycheck to run test cases on mec15xxevb_assy6853 board, all test cases will be failed with timeout, but if **using the west build and west flash, it could get the correct result log from the serial.**

**To Reproduce**
using the samples/hello_world as an example
1. using west to run this test
    command:
            west build -b mec15xxevb_assy6853 -p auto samples/hello_world
           ./pre_script.sh  (used to pull down the pin level)
            west flash
           ./post_script.sh
    serial output:
           *** Booting Zephyr OS build v2.2.0-rc1-172-g93b78b7bf8bf  ***
           **Hello World! mec15xxevb_assy6853?**
 
2. using sanitycheck to run this test
     command:
             sanitycheck --hardware-map mec.yaml --device-testing -T samples/hello_world
     console output:
ERROR   - **mec15xxevb_assy6853   samples/hello_world/sample.basic.helloworld    FAILED: Timeout**
ERROR   - see: /home/maksim/zephyrproject/zephyr/sanity-out/mec15xxevb_assy6853/samples/hello_world/sample.basic.helloworld/handler.log
INFO    - Total complete: 1/   1 100%  skipped:  0, failed:   1
INFO    - 0 of 1 tests passed (0.00%), **1 failed, 0 skipped with 0 warnings in 115.68 seconds**
INFO    - In total 1 test cases were executed on 1 out of total 229 platforms (0.44%)

it seems that there are **some error log message output after the last line in serial output, like this "Hello World! mec15xxevb_assy6853?"** , **however, the sanitycheck script usually used the last line to check whether this test passed or failed,** so the sanitycheck couldn't get the correct log and waited until timeout.

**the related commit is ac2b117a**, if disable this patch, it's ok to run sanitycheck on mec board.
**add "CONFIG_SYS_POWER_DEEP_SLEEP_STATES=n" in the board/arm/mec15xxevb_assy6853/mec15xxevb_assy6853_defconfig**, then all tests can be passed.

INFO    - Total complete:    1/   1  100%  skipped:    0, failed:   0
INFO    - 1 of 1 tests passed (100.00%), 0 failed, 0 skipped with 0 warnings in 56.82 seconds
INFO    - In total 1 test cases were executed on 1 out of total 229 platforms (0.44%)
    

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.1
 - Commit ID: 93b78b7b

chen-png, 2020-02-17:
@wentongwu  hi, wentong, do you know the root cause of this issue, it's related to your patch ac2b117.

wentongwu, 2020-02-18:
@chen-png could you please try the PR#22886? Thanks.

chen-png, 2020-02-18:
>  try the PR #22886 

it works well to run sanitycheck on mec board now.

wentongwu, 2020-02-18:
@albertofloyd @scottwcpg it seems when system resume from suspend state(deep sleep), uart hardware doesn't work properly, additional chars show up in the console, but that's not intended by code(you can see that with hello world sample), so could you please take a look from SOC level? Thanks

wentongwu, 2020-02-19:
ping @dcpleung @albertofloyd @scottwcpg

dcpleung, 2020-02-21:
This is probably caused by the deactivation of UART during deep sleep transition.

When deep sleep enabled, `samples/hello_world` would not print the next line character at the end. So consecutive runs were like this:
```
*** Booting Zephyr OS build v2.2.0-rc1-274-g896e32b4149e  ***
Hello World! mec15xxevb_assy6853*** Booting Zephyr OS build v2.2.0-rc1-274-g896e32b4149e  ***
Hello World! mec15xxevb_assy6853*** Booting Zephyr OS build v2.2.0-rc1-274-g896e32b4149e  ***
Hello World! mec15xxevb_assy6853*** Booting Zephyr OS build v2.2.0-rc1-274-g896e32b4149e  ***
Hello World! mec15xxevb_assy6853
```

But after I commented out the call to `deep_sleep_save_uarts()` in `soc/arm/microchip_mec/mec1501/device_power.c`, console output was like this:

```
*** Booting Zephyr OS build v2.2.0-rc1-274-g896e32b4149e  ***
Hello World! mec15xxevb_assy6853
*** Booting Zephyr OS build v2.2.0-rc1-274-g896e32b4149e  ***
Hello World! mec15xxevb_assy6853
*** Booting Zephyr OS build v2.2.0-rc1-274-g896e32b4149e  ***
Hello World! mec15xxevb_assy6853
```

@scottwcpg Do you have any thoughts on this? It seems to be like the UART was deactivated before it was able to send out all bits.

scottwcpg, 2020-02-21:
Does the code transmitting messages through the UART insure the UART TX FIFO is empty before allowing a transition to deep sleep?  When deep_sleep_save_uart is called it assumes the UART is idle.  The function saves and clears the UARTs' activate register which gates off the clock to each UART.  
Also, the UART TX FIFO is 16 bytes.  If the message is broken up where the next line character does not fit in the FIFO, all the current TX FIFO bytes must be transmitted before the messaging code can write the last byte.  Can a deep sleep transition occur at this point causing the last byte to not be transmitted?  This is hard to check because 16650 compatible UART does not have a register field indicating the number of bytes currently in the TX FIFO.  It's either empty or not empty.

dcpleung, 2020-02-21:
I don't think we currently have any mechanism to ensure buffer is emptied before going into deep sleep mode. This might require CONFIG_DEVICE_POWER_MANAGEMENT to be enabled on the board which will require us to implement PM functions on all the drivers being used.

albertofloyd, 2020-02-25:
@dcpleung How are you doing these consecutive runs?  Resetting the board in between? 
This  seems to be a problem where we are entering deep sleep when is not supposed to occur, hence UART gets disabled too early.

Currently board minimum residency is 2s (CONFIG_SYS_PM_MIN_RESIDENCY_DEEP_SLEEP_1=2000) 
String cut-off means idle thread runs before printk is completed. Sounds to me more like power subsystem issue  @wentongwu why isn't the printk in application preventing deep sleep in this case?

wentongwu, 2020-02-25:
> Currently board minimum residency is 2s (CONFIG_SYS_PM_MIN_RESIDENCY_DEEP_SLEEP_1=2000)
> String cut-off means idle thread runs before printk is completed. Sounds to me more like power subsystem issue @wentongwu why isn't the printk in application preventing deep sleep in this case?

@albertofloyd we don't enable CONFIG_LOG_PRINTK, so there is no buffering for the string. Instread we poll out every char(uart_ns16550_poll_out), so there is no way idle thread run first than application thread which is doing printk.

dcpleung, 2020-02-25:
I just power-cycled the board to get the results.

My theory is that the character is put into the UART FIFO, and the hardware is shifting the bits out. In the meantime, the idle loop runs and decides to go into deep sleep since there are no more timeouts (or events) at the end of the test. In other word, the UART has not shifted all the bits out the lines before running the deep sleep function. Note that, if I disables the deep sleep states during testing programatically, this behavior is not observed.

scottwcpg, 2020-02-25:
The 16550 UART Line Status run time register has read-only bit[6] which is set to 1 when both the FIFO and the shift register is empty.  The driver could use this bit to know when TX HW is idle.

dcpleung, 2020-03-02:
@chen-png could you try it with master and see if this is still an issue?

chen-png, 2020-03-03:
> @chen-png could you try it with master and see if this is still an issue?

it worked well to run sanitycheck on mec board now.

closed 2020-03-03 by dcpleung