zephyrproject-rtos/zephyr #22888: Can't flash test image into iotdk board.

chen-png, 2020-02-18:
**Describe the bug**
can't flash test image into iotdk board, it **failed with "Error: checksum", and can't see anything from the serial output.**
**the related patch is #22602**, if i used the commit 3b60f095 before this patch, it's ok to flash image into iotdk board, if with this patch, it will be failed.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b iotdk -p auto samples/hello_world
2. west flash
3. See error

**Screenshots or console output**
Info : JTAG tap: arc-em.cpu tap/device found: 0x200444b1 (mfg: 0x258 (ARC International), part: 0x0044, ver: 0x2)
**Error: arc_mem_checksum NOT SUPPORTED IN THIS RELEASE.
Error: checksum mismatch - attempting binary compare
Error: arc_mem_checksum NOT SUPPORTED IN THIS RELEASE.
Error: checksum mismatch - attempting binary compare**
verified 8904 bytes in 0.019230s (452.174 KiB/s)

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.1
 - Commit ID: a9bfd28a

chen-png, 2020-02-18:
@abrodkin Could you take a look of this issue? It's related to #22602.

abrodkin, 2020-02-18:
@chen-png thanks for spotting this one.

That problem happens because we didn't instruct OpenOCD runner to load Elf file instead of binary.
You may either do it manually adding `--use-elf` to west's command line or see https://github.com/zephyrproject-rtos/zephyr/pull/22891.

chen-png, 2020-02-18:
it worked well on itodk with #22891.

closed 2020-02-18 by nashif