zephyrproject-rtos/zephyr #22890: IP networking does not work on ATSAME70 Rev. B

stephanosio, 2020-02-18:
**Describe the bug**
IP networking does not work on ATSAME70 Rev. B SoC (tested on `sam_e70_xplained` populated with the Rev. B chip).

**To Reproduce**
Steps to reproduce the behavior:
1. Go to `samples/net/dhcpv4_client`
2. `mkdir build; cd build`
3. `cmake -GNinja -DBOARD=sam_e70_xplained`
4. `ninja flash`

**Screenshots or console output**
```
[00:00:00.000,000] <inf> i2c_sam_twihs: Device I2C_0 initialized
[00:00:00.001,000] <inf> eth_sam: MAC: fc:c2:3d:0b:c3:26
[00:00:00.001,000] <inf> eth_sam: Queue 0 activated
[00:00:00.001,000] <inf> eth_sam: Queue 1 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 2 set to idle
[00:00:00.001,000] <inf> eth_sam_phy: Soft Reset of ETH PHY
[00:00:00.091,000] <inf> eth_sam_phy: PHYID: 0x221561 at addr: 0
uart:~$ *** Booting Zephyr OS build v2.2.0-rc1-133-g1736cb79c220  ***
[00:00:02.446,000] <inf> eth_sam_phy: common abilities: speed 100 Mb, full duplex
[00:00:02.451,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
uart:~$ net stats

Interface 0x20407640 (Ethernet) [1]
===================================
IPv4 recv      114      sent    5       drop    114     forwarded       0
IP vhlerr      0        hblener 0       lblener 0
IP fragerr     0        chkerr  0       protoer 0
ICMP recv      0        sent    0       drop    0
ICMP typeer    0        chkerr  0
UDP recv       0        sent    0       drop    110
UDP chkerr     0
Bytes received 43175
Bytes sent     0
Processing err 34
uart:~$ net iface

Interface 0x20407640 (Ethernet) [1]
===================================
Link addr : FC:C2:3D:0B:C3:26
MTU       : 1500
Ethernet capabilities supported:
        Virtual LAN
        10 Mbits
        100 Mbits
        Priority queues
IPv4 unicast addresses (max 1):
        <none>
IPv4 multicast addresses (max 1):
        <none>
DHCPv4 lease time : 0
DHCPv4 renew time : 0
DHCPv4 server     : 0.0.0.0
DHCPv4 requested  : 0.0.0.0
DHCPv4 state      : selecting
DHCPv4 attempts   : 3
```

Detailed debug log is available at the following address:
https://gist.github.com/stephanosio/c36e41ba3eecea1724079b8660c42284

**Environment (please complete the following information):**
 - OS: Ubuntu 18.04
 - Toolchain: Zephyr SDK 0.11.1
 - Commit SHA: 1736cb79c220f02e8a93c29f7c1ba3323aff17c1

stephanosio, 2020-02-18:
cc @mnkp @aurel32 @jukkar

tbursztyka, 2020-02-18:
Do you have another board to test the same network setup? To make sure it is not the zephyr dhcp client that does not work against your dhcp server. (Try also wireshark to see packet flow, if any)

stephanosio, 2020-02-18:
> Do you have another board to test the same network setup? To make sure it is not the zephyr dhcp client that does not work against your dhcp server. (Try also wireshark to see packet flow, if any)

I have `frdm_k64f` right next to it, everything works on it.

tbursztyka, 2020-02-18:
ok, and on wireshark you see no packet going out then?

stephanosio, 2020-02-18:
> ok, and on wireshark you see no packet going out then?

Correct, it seems like Ethernet frames are not getting transmitted for some reason. Receiving works though.

jukkar, 2020-02-18:
I wonder if this could help #12711

stephanosio, 2020-02-18:
> I wonder if this could help #12711

#12711 fixed the problem. It looks like we will need to merge #12711 ...

closed 2020-02-18 by stephanosio