zephyrproject-rtos/zephyr #22892: Kconfig warning when serial disable on PCA10059

pdunaj, 2020-02-18:
**Describe the bug**
When serial is disabled within application configuration file, the `CONFIG_UART_0_NRF_UARTE` option is no longer available. However since it is set to y in `boards/arm/nrf52840_pca10059/nrf52840_pca10059_defconfig` warning appears while compiling kconfig.

**To Reproduce**
Configure some application for pca10059 but disable serial related options.

**Expected behavior**
No warning during cmake processing.

**Impact**
Annoyance.

**Screenshots or console output**
```
warning: The choice symbol UART_0_NRF_UARTE (defined at drivers/serial/Kconfig.nrfx:33) was selected
(set =y), but no symbol ended up as the choice selection. See
http://docs.zephyrproject.org/latest/reference/kconfig/CONFIG_UART_0_NRF_UARTE.html and/or look up
UART_0_NRF_UARTE in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.
```

**Environment (please complete the following information):**
replicable with master

**Additional context**
N/A

pdunaj, 2020-02-18:
It would be great if configuration at board level was kept in bare minimum, assuming nothing about what driver/HW is enabled.

carlescufi, 2020-02-18:
@pdunaj I'll send a patch

PR #28478 (2020-09-18): boards: nrf52840dongle: Enable serial conditionally
Do not force SERIAL and UART_INTERRUPT_DRIVEN unconditionally.

Fixes #22892.
commit b8d13d6f (2020-09-18): boards: nrf52840dongle: Enable serial conditionally

Do not force SERIAL and UART_INTERRUPT_DRIVEN unconditionally.

Fixes #22892.
closed 2020-09-18 by carlescufi