zephyrproject-rtos/zephyr #22902: eth_mcux_phy_setup called before ENET clock being enabled causes CPU to hang

armandciejak, 2020-02-18:
**Describe the bug**
In `drivers/ethernet/eth_mcux.c:eth_0_init` `eth_mcux_phy_setup` is called before the ENET clock is enabled a few line below in `ENET_Init`. This causes the CPU to hang. Inside the `eth_mcux_phy_setup` function the SMI access cannot work without the ENET clock enabled.

IMHO `eth_mcux_phy_setup` should be be called after `ENET_Init` and `ENET_SetSMI`.

**Impact**
This prevent our board from booting.

**Environment (please complete the following information):**
 - OS: Linux (Ubuntu 18.04)
 - Toolchain: Zephyr SDK
 - Commit: 9f7cccc90f648953e0d198fffda22928edb277cf

**Additional context**
This problem occurs on a custom board equipped with the IMXRT1062 MCU but should occur as well with the MIMXRT1060-EVK.

MaureenHelm, 2020-02-24:
@agansari will you please have a look?

agansari, 2020-02-25:
@armandciejak `eth_mcux_phy_setup` was added in commit 2778e1d534845bc5e950528b4f858549134f51f1 it fixes an errata on early 1050EVK boards.
You are right about clock and SMI initialization.
Can you help me test #23068 pull on your board? I don't have an i.MX RT board right now.

closed 2020-02-25 by jhedberg

armandciejak, 2020-03-03:
@agansari I could test your change with success on my board. Thanks for your commit.

PR #23068 (2020-02-25): drivers: eth_mcux phy_setup moved after SMI init
SMI initialization is required to enable PHY communication.
PHY setups needs to run after SMI initialization.

Fixes #22902