zephyrproject-rtos/zephyr #22907: si7006 temperature conversion offset missing

anthonybrandon, 2020-02-18:
The si7006 sensor driver (./drivers/sensor/si7006/si7006.c) is missing an offset as described in the [datasheet.](https://www.silabs.com/documents/public/data-sheets/Si7006-A20.pdf) An offset of 46.85 should be subtracted in order to get the temperature in degrees Celsius. 

#19941 attempts to fix it along with some unrelated changes.

carlescufi, 2020-02-18:
@anthonybrandon would you mind reviewing and testing #19941 to see if we can move it forward?

anthonybrandon, 2020-02-18:
@carlescufi 

Sure, I have a board I can try it out on.
Do I just report it in a comment on the merge request once I have tried it, or did you mean something else?

carlescufi, 2020-02-19:
> @carlescufi
> 
> Sure, I have a board I can try it out on.
> Do I just report it in a comment on the merge request once I have tried it, or did you mean something else?

Yep, just approve the Pull Request and add a comment there.

MaureenHelm, 2020-04-22:
Fixed by #24563

closed 2020-04-22 by MaureenHelm