zephyrproject-rtos/zephyr #22927: (PTS)Bluetooth:  SM/MAS/SIP/BV-02-C-INCONCLUSIV

agata-ponitka, 2020-02-19:
Describe the bug
Zephyr set IO capability to KeyboardOnly instead of KeyboardDisplay

To Reproduce
run auto-pts

Expected behavior    
IO capability needs to be set to KeyboardDisplay

[SM_MAS_SIP_BV-02-C-INCONCLUSIV.zip](https://github.com/zephyrproject-rtos/zephyr/files/4223855/SM_MAS_SIP_BV-02-C-INCONCLUSIV.zip)

aescolar, 2020-02-19:
CC @jhedberg @joerchan

jhedberg, 2020-02-19:
@aescolar this sounds more like a test setup error than a stack bug, do you agree?

jhedberg, 2020-02-19:
Is the high priority really justified? I just took a look at `get_io_capa()` in `host/smp.c` and it *is* able to return KeyboardOnly, so this looks more like an auto-pts or tester app issue, and should therefore not be high priority (since the stack poses no obstacle for qualification in this particular case)

PR #23037 (2020-02-23): Bluetooth: tester: Fix 3 GAP qualification tests
Fix the following 3 failing tests:

- SM/SLA/SIP/BV-01-C
- SM/MAS/SIP/BV-02-C
- SM/MAS/PKE/BV-01-C

Fixes #22929.
Fixes #22927.
Fixes #22930.
closed 2020-02-23 by jhedberg