zephyrproject-rtos/zephyr #22928: (PTS)Bluetooth: SM/MAS/SIGN/BV-03-C, SM/MAS/SIGN/BI-01-C - INCONCLUSIV

agata-ponitka, 2020-02-19:
Describe the bug
Failed to distribute keys.

To Reproduce
run auto-pts

Expected behavior
IUT faild to distribute CSRK.

[SM_MAS_SIGN_BI-01-C-INCONC.txt](https://github.com/zephyrproject-rtos/zephyr/files/4223782/SM_MAS_SIGN_BI-01-C-INCONC.txt)
[SM_MAS_SIGN_BV-03-C-INCONC.txt](https://github.com/zephyrproject-rtos/zephyr/files/4223783/SM_MAS_SIGN_BV-03-C-INCONC.txt)

aescolar, 2020-02-19:
CC @jhedberg @joerchan

joerchan, 2020-02-19:
If I am reading the attached logs correctly it looks like there is an SMP error with SMP_UNSPECIFIED error code. Is it possible to get some more information or logs for these 2 test cases?

agata-ponitka, 2020-02-20:
Yes, I am sending logs in attachment. 

[SM_MAS_SIGN_BI-01-C.zip](https://github.com/zephyrproject-rtos/zephyr/files/4230941/SM_MAS_SIGN_BI-01-C.zip)

[SM_MAS_SIGN_BV-03-C.zip](https://github.com/zephyrproject-rtos/zephyr/files/4230805/SM_MAS_SIGN_BV-03-C.zip)

carlescufi, 2020-02-20:
@agata-ponitka could you please give us the exact command-line that you are using on the client side please?

agata-ponitka, 2020-02-20:
@carlescufi of course: ./autoptsclient-zephyr.py autopts-zephyr-210 ~/zephyrproject/zephyr/build/zephyr/zephyr.elf -i 192.168.56.101 -t /dev/ttyACM0 -b nrf52 -l 192.168.56.1 -c SM/MAS/SIGN/BI-01-C

carlescufi, 2020-02-20:
> @carlescufi of course: ./autoptsclient-zephyr.py autopts-zephyr-210 ~/zephyrproject/zephyr/build/zephyr/zephyr.elf -i 192.168.56.101 -t /dev/ttyACM0 -b nrf52 -l 192.168.56.1 -c SM/MAS/SIGN/BI-01-C

Thank you @agata-ponitka.

joerchan, 2020-02-20:
@agata-ponitka The PTS appears to be affected by a race condition in the HCI USB protocol. I have created a PR which has a workaround for this during testing, by simply sleeping for 10 ms before starting to distribute the keys.
The tests are still inconclusive though. I think some extra auto-pts changes might be needed to confirm that the data with wrong MAC was not accepted on the tester side.

jhedberg, 2020-02-20:
Downgraded to medium since this looks like a PTS & auto-pts issue

carlescufi, 2020-02-20:
@agata-ponitka this is now an autops/tester app issue with an "mmi" failure.

agata-ponitka, 2020-02-21:
I have added changes in autopts and retested with the latest master. The tests are pass now. Thank you.

aescolar, 2020-02-21:
Closing then. Thanks

closed 2020-02-21 by aescolar

carlescufi, 2020-02-22:
I've also re-run those locally and they pass for me as well. cc @joerchan