zephyrproject-rtos/zephyr #22929: (PTS)Bluetooth: SM/SLA/SIP/BV-01-C Error

agata-ponitka, 2020-02-19:
Describe the bug
MITM bit in Pairing Request cannot be set to cleared. Zephyr aplication doesn't support set BTP MITM off command.

To Reproduce
run auto-pts

[SM_SLA_SIP_BV-01-C-FAIL.zip](https://github.com/zephyrproject-rtos/zephyr/files/4223970/SM_SLA_SIP_BV-01-C-FAIL.zip)

aescolar, 2020-02-19:
CC @jhedberg @joerchan

carlescufi, 2020-02-19:
Lowering to medium since this is a tester issue.

carlescufi, 2020-02-21:
The BTP MITM command needs to be implemented.

carlescufi, 2020-02-21:
The test doesn't seem to bue using the MITM command (0x1b):
```
*** Booting Zephyr OS build v2.2.0-rc1-263-gcee1954ac394  **
[00:00:00.000,000] <dbg> bttester.tester_init: Initializing tester
[00:00:00.020,294] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.020,324] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.020,324] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.2 Build 0
[00:00:00.021,514] <inf> bt_hci_core: Identity: e6:37:9e:89:3f:ab (random)
[00:00:00.021,514] <inf> bt_hci_core: HCI: version 5.1 (0x0a) revision 0x0000, manufacturer 0x05f1
[00:00:00.021,514] <inf> bt_hci_core: LMP: version 5.1 (0x0a) subver 0xffff
[00:00:00.034,667] <dbg> bttester_gap.tester_handle_gap: opcode: 0x03
[00:00:00.360,900] <dbg> bttester_gap.tester_handle_gap: opcode: 0x10
[00:00:07.799,133] <dbg> bttester_gap.tester_handle_gap: opcode: 0x08
[00:00:07.804,321] <dbg> bttester_gap.tester_handle_gap: opcode: 0x0a
[00:00:08.037,933] <dbg> bttester_gap.tester_handle_gap: opcode: 0x11
```

PR #23037 (2020-02-23): Bluetooth: tester: Fix 3 GAP qualification tests
Fix the following 3 failing tests:

- SM/SLA/SIP/BV-01-C
- SM/MAS/SIP/BV-02-C
- SM/MAS/PKE/BV-01-C

Fixes #22929.
Fixes #22927.
Fixes #22930.
closed 2020-02-23 by jhedberg