zephyrproject-rtos/zephyr #22930: (PTS)Bluetooth: SM_MAS_PKE_BV-01-C-INCONCLUSIV

agata-ponitka, 2020-02-19:
Describe the bug

Bonding flag in Pairing Request cannot be set to cleared. Zephyr aplication doesn't support set BTP bondable off command.

To Reproduce
run auto-pts

[SM_MAS_PKE_BV-01-C-INCONC.zip](https://github.com/zephyrproject-rtos/zephyr/files/4224043/SM_MAS_PKE_BV-01-C-INCONC.zip)

aescolar, 2020-02-19:
CC @jhedberg @joerchan

carlescufi, 2020-02-19:
Lowering to medium since this is a tester issue.

carlescufi, 2020-02-21:
The test doesn't seem to be using the [Bondable command (0x09)](https://github.com/intel/auto-pts/blob/master/doc/btp_spec.txt#L278):

```
*** Booting Zephyr OS build v2.2.0-rc1-263-gcee1954ac394  **
[00:00:00.000,000] <dbg> bttester.tester_init: Initializing tester
[00:00:00.019,348] <inf> bt_hci_core: HW Platform: Nordic Semiconductor (0x0002)
[00:00:00.019,378] <inf> bt_hci_core: HW Variant: nRF52x (0x0002)
[00:00:00.019,378] <inf> bt_hci_core: Firmware: Standard Bluetooth controller (0x00) Version 2.2 Build 0
[00:00:00.020,568] <inf> bt_hci_core: Identity: e6:37:9e:89:3f:ab (random)
[00:00:00.020,568] <inf> bt_hci_core: HCI: version 5.1 (0x0a) revision 0x0000, manufacturer 0x05f1
[00:00:00.020,599] <inf> bt_hci_core: LMP: version 5.1 (0x0a) subver 0xffff
[00:00:00.025,390] <dbg> bttester_gap.tester_handle_gap: opcode: 0x03
[00:00:00.365,020] <dbg> bttester_gap.tester_handle_gap: opcode: 0x10
[00:00:09.241,973] <dbg> bttester_gap.tester_handle_gap: opcode: 0x0e
[00:00:09.487,121] <dbg> bttester_gap.tester_handle_gap: opcode: 0x11
[00:00:09.499,542] <dbg> bttester_gap.tester_handle_gap: opcode: 0x11
```

PR #23037 (2020-02-23): Bluetooth: tester: Fix 3 GAP qualification tests
Fix the following 3 failing tests:

- SM/SLA/SIP/BV-01-C
- SM/MAS/SIP/BV-02-C
- SM/MAS/PKE/BV-01-C

Fixes #22929.
Fixes #22927.
Fixes #22930.
closed 2020-02-23 by jhedberg