zephyrproject-rtos/zephyr #22931: GPIO callback is not triggered for tests/drivers/gpio/gpio_basic_api on microchip mec15xxevb_assy6853 board

225296, 2020-02-19:
**Describe the bug**
GPIO callback is not triggered for tests/drivers/gpio/gpio_basic_api on mec15xxevb_assy6853 board.

**To Reproduce**
Steps to reproduce the behavior:
1. using west to run this test
command:
-  West build -p -b mec15xxevb_assy6853 .
- ./pre_script.sh (used to pull down the pin level)
-  west flash
- ./post_script.sh
- see error

2. using sanitycheck to run this test
In the file: ” boards/arm/mec15xxevb_assy6853/ mec15xxevb_assy6853_defconfig” , Update to :  CONFIG_SOC_POWER_MANAGEMENT=n
command:
sanitycheck --device-testing --hardware-map my_new_hw_map.file -p mec15xxevb_assy6853 -T /home/kanjana/zephyrproject/zephyr/tests/drivers/gpio/gpio_basic_api -v -vconsole 

**Expected behavior**
GPIO callback should be triggered 

**Impact**
Not able to validate GPIO on Microchip board

**Screenshots or console output**
*** Booting Zephyr OS build v2.2.0-rc1-185-ga9bfd28a39f3  ***
Running test suite gpio_basic_test
===================================================================
starting test - test_gpio_port
Validate device GPIO240_276
Check GPIO240_276 output 1 connected to input 2
OUT 1 to IN 2 linkage works
- bits_physical
- pin_physical
- check_raw_output_levels
- check_logic_output_levels
- check_input_levels
- bits_logical
- check_pulls
PASS - test_gpio_port
===================================================================
starting test - test_gpio_callback_add_remove
FAIL - test_callback_add_remove@110. 1Callback1 triggered 0 times and Callback2 triggered 0 times
FAIL - test_callback_add_remove@111. not trigger callback correctly

    Assertion failed at ../src/test_callback_manage.c:245: test_gpio_callback_add_remove: test_callback_add_remove() not equal to TC_PASS

FAIL - test_gpio_callback_add_remove
===================================================================
starting test - test_gpio_callback_self_remove
FAIL - test_callback_self_remove@169. not trigger callback correctly

    Assertion failed at ../src/test_callback_manage.c:251: test_gpio_callback_self_remove: test_callback_self_remove() not equal to TC_PASS

FAIL - test_gpio_callback_self_remove
===================================================================
starting test - test_gpio_callback_enable_disable
FAIL - test_callback_enable_disable@214. not trigger callback correctly

    Assertion failed at ../src/test_callback_manage.c:257: test_gpio_callback_enable_disable: test_callback_enable_disable() not equal to TC_PASS

FAIL - test_gpio_callback_enable_disable
===================================================================
starting test - test_gpio_callback_variants
OUT init a01, IN cfg 34000, cnt 0
FAIL - test_callback@108. edge not trigger callback correctly

    Assertion failed at ../src/test_callback_trigger.c:131: test_gpio_callback_variants: (test_callback(GPIO_INT_EDGE_FALLING) not equal to TC_PASS)
falling edge failed
FAIL - test_gpio_callback_variants
===================================================================
starting test - test_gpio_deprecated
INT cfg 50000, cnt 0
FAIL - test_callback@124. edge not trigger callback correctly

    Assertion failed at ../src/test_deprecated.c:154: test_gpio_deprecated: (test_callback(GPIO_INT_EDGE | GPIO_INT_ACTIVE_HIGH) not equal to TC_PASS)
rising edge failed
FAIL - test_gpio_deprecated
===================================================================
Test suite gpio_basic_test failed.
===================================================================
PROJECT EXECUTION FAILED

**Environment (please complete the following information):**
OS : Ubuntu LTS 18.04
Toolchain: Zephyr SDK : 0.11.2
Commit ID: ga9bfd28a39f3

pabigot, 2020-02-19:
@dcpleung please look at this; I don't have the hardware.

dcpleung, 2020-02-20:
Could you try remove the jumpers on JP15? This header connects GPIO 50 and 51 to pull-up resistors which may cause issue when the GPIO pins cannot sink enough current. I just tried this on my board and the test was working correctly with the JP15 jumpers removed.

225296, 2020-02-24:
•	Removed jumper JP15 to disconnect pull-up resistor and used GPIO_050 and GPIO_051 , 
Test Execution is **PASSED**

closed 2020-02-24 by 225296