zephyrproject-rtos/zephyr #22933: k_delayed_work_submit_to_queue returns error code when resubmitting previously completed task.

LuxLuthor, 2020-02-19:
**Describe the bug**
k_delayed_work_submit_to_queue returns error code -EALREADY when submitting previously completed task. Tak is submitted successfully. This response is not covered in doc. This should probably be changed to 0 since error code is propagated from k_delayed_work_cancel. 

**Expected behavior**
0 result - work was submitted to queue. 
OR
response described in doc

carlescufi, 2020-02-19:
was this fixed by https://github.com/zephyrproject-rtos/zephyr/pull/22810?
@Vudentz @joerchan @pabigot

joerchan, 2020-02-19:
@LuxLuthor Can you specify which commit you are on?
Maybe it is a regression from #22810?

LuxLuthor, 2020-02-19:
I observe issue on 28298fa06a 
It seems that it is regression from #22810 
Code below causes setting err to -EALREADY although work is send to queue
`	
/* Cancel if work has been submitted */
	if (work->work_q == work_q) {
		err = work_cancel(work);
		/* -EALREADY indicates the work has already completed so this
		 * is likely a recurring work.
		 */
		if (err < 0 && err != -EALREADY) {
			goto done;
		}
	}
`

pabigot, 2020-02-19:
I propose we should revert #22810, fix #22803 by changing the documentation to reflect what the API actually does, and fixing #22756 with the alternative solution from #22769.

pabigot, 2020-02-19:
Clarification: I'm assuming that the bug here is a direct result of the change in API to match documentation by now returning an error from `k_delayed_work_cancel()` where before it returned success.  I take that as a sign that we shouldn't have changed the API in #22810, certainly not at this stage of the release cycle.  @jhedberg 

This particular problem might be fixed by overriding the error return in this case and make it return success value; I don't know, haven't looked closely.  But that won't address any other cases that are going to be discovered.

joerchan, 2020-02-19:
I think using the function added here in #22826 (we could just have it internal) is the way to solve it.
I cannot see a way to reset work->work_q to NULL. work_timeout would not be the right place it seems.

Tested using this approach and that solves it. Thoughts @pabigot @Vudentz ?

```
diff --git a/kernel/work_q.c b/kernel/work_q.c
index 845a2670c8..fba3f36a34 100644
--- a/kernel/work_q.c
+++ b/kernel/work_q.c
@@ -80,6 +80,12 @@ static int work_cancel(struct k_delayed_work *work)
        return 0;
 }
 
+static inline bool k_delayed_work_pending(struct k_delayed_work *work)
+{
+       return !z_is_inactive_timeout(&work->timeout) ||
+              k_work_pending(&work->work);
+}
+
 int k_delayed_work_submit_to_queue(struct k_work_q *work_q,
                                   struct k_delayed_work *work,
                                   s32_t delay)
@@ -94,7 +100,7 @@ int k_delayed_work_submit_to_queue(struct k_work_q *work_q,
        }
 
        /* Cancel if work has been submitted */
-       if (work->work_q == work_q) {
+       if (k_delayed_work_pending(work)) {
                err = work_cancel(work);
                /* -EALREADY indicates the work has already completed so this
                 * is likely a recurring work.
```

pabigot, 2020-02-19:
If we want a quick fix something like this might do it @Vudentz since #22810 left the non-zero `err` value to fall through causing the misbehavior:

```
diff --git a/kernel/work_q.c b/kernel/work_q.c
index 845a2670c8..96cd2f644f 100644
--- a/kernel/work_q.c
+++ b/kernel/work_q.c
@@ -99,7 +99,9 @@ int k_delayed_work_submit_to_queue(struct k_work_q *work_q,
                /* -EALREADY indicates the work has already completed so this
                 * is likely a recurring work.
                 */
-               if (err < 0 && err != -EALREADY) {
+               if (err == -EALREADY) {
+                       err = 0;
+               } else if (err < 0) {
                        goto done;
                }
        }
```

But I'm still not really happy about changing the return value of `k_delayed_work_cancel()` from success to error in this case.  We have no idea who else depends on the existing behavior.

PR #22943 (2020-02-20): kernel: work_q: fix return value in non-error case
A recent patch allowed an error code to be returned even though the
execution path treated it as a non-error condition.  Clear the code
before returning.

Fixes #22933
closed 2020-02-20 by jhedberg