zephyrproject-rtos/zephyr #22941: Refining Zephyr's Device Driver Model - Take II

tbursztyka, 2020-02-19:
(This issue is a replacement of #6293 )

### A deep revise and upgrade of the device driver model

Throughout its short existence, Zephyr has seen its kernel reworked, stabilized and made reliable with sound and concise APIs and objects. All still being designed within the scope of running on tiny targets, such as arduino_zero with 32k of RAM or bbc_microbit with its 16k of RAM for instance, but also providing enhanced features for fully fledged industrial boards with much more memory and CPU power.

Next to the kernel, the device driver ecosystem has experienced somewhat a different development.
The device driver APIs have been improved to cover most of identified generic use-cases, and are still being reworked, sometime deeply refined to adress new requirements. Surprisingly however, the core of this ecosystem: the device driver model, has gained little to no improvement at all. It does not make it void of issues, far from it actually. In fact, some of the issues, known from the beginning  sometime, have been a drawback for implementing advanced features. Worse, as time goes on with the device driver ecosystem growing everyday, these issues will tend to be more and more tedious to get fixed. It is not to say there are no attempts to fix some the known issues, there are. But I believe they are not generic nor concise enough, though some are definitely interesting test beds.

Hence a list of the known issues and a proposal to address these.

### Identified issues

1.  Power Management is broken:
    The lack of dependency tree support among devices makes it unusable. The burden of managing client devices and their parents within the power management context is pushed to the user which has been proven to be tedious, unreliable and thus made the PM subsystem nearly - if not fully - unused at all. Due to this most device drivers _do not_ implement the PM API.

2.  No support for cancelling on-going device API call:
     In PM context or else, it might be relevant to cancel an on-going call on a device. Stopping and/or re-initializing a user thread would require such feature for instance. A caller could also decide to cancel its own calls.

3.  No support for a timeout on each and every device API call:
    Being generic to all device driver API such feature is one of the important missing. It may happen, either because the system is overloaded or because the hw is not responding, that a call will never come to a closure. Thus locking up the device on the call as well as the caller waiting for something to happen. This prevents also the caller having the possibility to mitigate any error in case the call would be unsuccessful.

4.  No error storage when device initialization fails or gets into hw fault:
    In case of initialization failure, the device will just be inaccessible, and there would be no way to know what happened. In case of a run-time, non-blocking, hardware fault, only the user will get an error back. Localized logging cannot be sufficient in this case.

5.  No possibility to recover a device in case of initialization failure or hw fault:
    Such failure could be due to boot timing, temporary hw issue, cosmic rays, gremlins, etc... Currently, a device failing the initialization step will not be usable _at all_ afterwards. One can decide to reboot, but if the boot sequence is the culprit, the device will never be recoverable. It should be possible to call init again. Either to fix such booting issue but also at run-time where recovering a device may require to reinitialize it (after a locking hw fault for instance).

6.  Device instantiation:
    A device is uniquely allocated in device driver code through DEVICE_DEFINE() or DEVICE_AND_API_INIT() or DEVICE_INIT(). It requires as many calls of one of these macros as there are devices then. The link with DTS is non-existent unless the developer makes it happen using DTS generated config options, which is the current process.

7.  Each device is not constant (same for their config part):
    It was meant to be constant, but due to a hack on device's API setting (in case of init failure it is set to NULL at runtime) it is never constant, thus ends into RAM instead of staying into the ROM.

8.  The device name:
    It is only used to get the device bindings at runtime and thus reduce boot performances and takes ROM/RAM for nothing.

9.  Nameless device created by SYS_INIT():
    It does not make any use of the name, API, PM API and thus use too much memory for nothing.

10. device_pm structure:
    An obvious case of under optimized structure

11. Device's variable name:
    Is not generated from the device name (you cannot un-stringify on per-processor level),
    thus requiring a hard-coded unique and un-quoted name. That issue is due to the allocation macros (see issue 6).

Other know issues already recorded:

12. Device initialization order that respects device tree dependencies #22545

### Issues and intersections

Looking closely at the current device driver model architecture, it is rather obvious some issues find part of their own fix in other's.
We can therefore have such quick intersections list:

1.  2, 3, 4, 5, 6
2.  NA
3.  2, 4
4.  NA
5.  4
6.  NA
7.  4
8.  6
9.  NA
10. NA
11. 6
12. ~6

### An indirect solution to many of the issues
[... 111 lines cut ...]
   ```

   And the (infamous) device_get_binding() would just be a macro, replacing the "call" to device_get_binding(DT_LABEL_DEVICE_K) into device_k. It's in fact very trivial. A previous q&d patch I did back in the days was able to show off this feature, however not based on DTS at that time (it got barely introduced then) but on Kconfig. And it worked very well, the gain of ROM/RAM was interesting and the boot time was better as well, since device_get_binding() was not anymore retrieving a device by its name.

9. Nameless device created by SYS_INIT():

   A simple struct device, and its components, refactoring would be sufficient. Again, that was already prototype back in the days for the issue 8.

10. Optimizing device_pm structure:

   - the dev attribute can be removed (CONTAINER_OF() would just work)
   - enable and fsm_state could be just in same bit field and generalized (instead of book keeping "internal power state")
   - the lock could be generalized in sturct device, and used also for all calls for concurrent access.
   - work, event and signal usage should be revised. See on/off API for instance

11. Device's variable name

   Once issue 6 fixed, that would also directly fix this issue. See the PR #20253 attempting to generate such unique id "ZID", though the PR goal is wrong as there is no need for making device init and device_get DT aware per-se. Connecting strongly DTS to device (variable name, instances etc...) would anyway make the whole device driver model DT aware. (See issue 6)

12. Device initialization order that respects devicetree dependencies #22545

   That one would require also first to expose the software services to DTS. That's trivial imo. Then indeed sorting out the priority and levels out of the device tree would require some work on DTS side.
Let's sort the details out directly in relevant issue.

   Note that getting issue 6 fixed would greatly help that, in a sense no one would have to go all over the existing device drivers to fix the priority/labels there.

carlescufi, 2020-02-19:
CC @galak @pabigot @nordic-krch @erwango @MaureenHelm @nashif @anangl @henrikbrixandersen

andrewboie, 2020-02-20:
One more known (but relatively minor) issue that affects kernel footprint:

The model of placing function pointers in an API struct defeats gc-sections, as the functions will always have at least a reference in that API struct even if not referenced elsewhere. So if a driver is built, all of its API members will be pulled into the binary even none of them, or only a subset of them, are invoked.

At the moment I have no proposals to remedy this, but thought it worth noting.

pizi-nordic, 2020-02-20:
@tbursztyka: For Power Management you will need more complex dependency tracking, as you have several types of relation between devices. Examples:

- Parent/Child: Which controller handles given device?
- Device/GPIO: Which GPIO is used to select SPI Device? Which controller controls this GPIO?
- Device/Clock: What provides clocks for the device?
- Device/Power: What provides power for the device?

nordic-krch, 2020-02-20:
@tbursztyka thank you for taking that initiative. Few comments:

I would add following requirements/topics:
1. using devices with generic API's #22415 
2. device uninit
One use case is bootloader which should provide clean state before jumping into application. Second use case is nordic specific: we have couple of peripherals which are sharing same hw and address space thus cannot be used at once (e.g. SPI0 and I2C0) and currently in zephyr there is no way to use both in same application. I guess there might be other cases where user may want to uninit driver not because of PM reason.
3. manual/deferred initialization
Currently, all devices are initialized before reaching application. However, there might be cases where startup time is vital and application may want to defer initialization of some devices (because they might not be needed). One of the examples might be wake up from brown out reset when we want to do some kind of maintenance (e.g. flush some retained data to flash) and we don't want to have full blown system started.

> No support for a timeout on each and every device API call

I assume that this is applies only to synchronous calls.

tbursztyka, 2020-02-21:
@andrewboie Indeed, very good point. I wonder if an ld script could check if a device instance is being used, besides its presence in the device init section, and would give a warning about it. Though the variable name of the device pointer would probably be not really user friendly (it could be generated from the ZID, thus an indirect way to relate to DTS and then to the driver it belongs). 
But that sounds convoluted, there is probably a better way.

@pizi-nordic Most of this has to be solved in DTS side. The goal is to embed as less information as possible. Note that a node child may have 1+ parents (spi master and gpio in case of a spi device using a gpio cs, for instance.). The hierarchy of nodes would have to be generated by DTS. On DTS side, a clock source would need to expose whether it can be stopped (or under-clocked etc...), if it's fixed: no need to make it as a parent. Dealing with clock gate, I'd prefer the driver to embed that info directly, though I did not think much of that issue. Could be made generic, let's see.

@nordic-krch I still have to take some time on your proposal 1. I am not yet convinced this require a generic point into device. It would need more use case I think. About point 2, yes that's a very good one. On DW IP blocks for instance it may share spi and i2s. It is indeed a common use case. However, we have to come up with a transparent solution, i.e. something that would not require the user to do _anything_. Partial solution is about the device lock mechanism, in case of 2+ dev sharing the same resource, such lock will have to be unique for these 2+ devs. But then indeed we need to know: "is dev_x initialized?" and "since I just got the lock from dev_2, how to de-init dev_1 and init dev_2?" and "if I just used dev_2 last time, and reuse it again... I don't need to re-initialize it then?" etc... Point 3 seems valid as well, and it is crossing point 2: the way to know if a device has been initialized or not, and thus reacting transparently based on that info.

lgirdwood, 2020-02-21:
> > No support for a timeout on each and every device API call
> 
> I assume that this is applies only to synchronous calls.

Can this be explained in more detail.?

The danger here is that you are pushing device driver specific knowledge up the stack and this becomes unmanageable very quickly. i.e. consider a driver API calls that polls on a HW state change. It could poll forever, but it's the responsibility of the driver (and not the core) to manage this timeout (and return -ETIMEOUT or whatever up the stack).

tbursztyka, 2020-02-21:
@lgirdwood There is nothing the core has to do in this case. The idea it to bring a common object into struct device so the drivers will be able to use afterwards in a normalized way, instead of letting each and every device driver reinventing the wheel on their own.

vanti, 2020-02-25:
Adding a couple of ideas:

1. Expanding from the line of thought from @pizi-nordic, it'd be nice to have a concept of power domains in DTS. On TI SoCs and likely other ones out there, devices are grouped into power domains, and only when all devices in the domain are not needed then the entire domain can be shut off. 

2. Another use for "device uninit" is to employ it as a way for the application to say it doesn't intend to use a device for a long time, so the device can be shut off. With several device drivers currently the driver can only assume a device is not needed whenever there is no on-going data transfer, in which case it'd just turn on and off the device around the "transfer" function for example. But if the application wants to do back-to-back transfers, this can result in an unnecessary large overhead. If there were a "device uninit" then the driver would have the option to turn on the device during init and shut it off when "uninit" is called.

tbursztyka, 2020-02-27:
@vanti the 1 is indeed a good point, however let's put it aside for the moment as it is DTS/PM only issue (there is a _lot_ of issues with PM besides the dependency thing I mentioned. To put it mildly: it's a mess.). I think once we get the base fixed - issue 6 and 1 - we will be able to go further and fix PM completely. About 2, such use-case sounds just like suspend/resume to me. So I still don't see a proper use-case for "exposing" an uninit function. Note that "un-initializing" a device won't necessarily mean it will be shut down, from a power consumption perspective. And again, un-initiliazing a device which would be parent of other device... It's the issue 1.

b0661, 2020-03-01:
@tbursztyka, updated PR #10885 in case somebody wants to play around with code generation.

tbursztyka, 2020-03-03:
@b0661 Sure, I am evaluating it along with other solutions.

nashif, 2020-04-04:
deleted two duplicate comments, for the curious

tautologyclub, 2020-04-08:
So I get how iteratively enabling parents would be nice but iteratively disabling children seems sort of overkill. Wouldn't listing parents but only ref counting children be enough for pm?

endian-benjamin, 2020-04-08:
Another point regarding deferred init etc:

Perhaps these shortcomings are not strictly a matter of the API lacking features. It could be argued that it's more a matter of what's considered idiomatic. If drivers were in general structured like this:

```
int foodev_init(struct device *dev)
{
        // init mutexes, reserve memory, other setup-type tasks
        
#ifndef CONFIG_DEVICE_PM
        // this is NOT done in init() if CONFIG_DEVICE_PM=y
        foodev_do_poweron(dev); 
#endif /* CONFIG_DEVICE_PM */
}

int foodev_pm_control(struct device *dev, u32_t ctrl_command,
                      void *context, device_pm_cb cb, void *arg)
{
        // ...rather, it's done here:
        if (ctrl_command == DEVICE_PM_SET_POWER_STATE) {
                foodev_do_poweron(dev);
        }
}
```

then all "real" inits would be deferred until the user actually requested the device to be powered on. But NO (upstream) drivers are written like this to my knowledge.

Ofc, supporting this idiom would be a lot easier if the device could programmatically access its direct device dependencies...

pabigot, 2020-04-09:
Providing device dependencies is not difficult: we already have them, we just haven't emitted them yet, perhaps because we haven't identified the canonical representation of a device handle.  (The address of the `struct device` would work, but that's a lot of space when we could use a `u8_t` or `u16_t` handle value to identify dependent and depending nodes in any reasonable Zephyr application.)

While we can provide the dependency information for device tree nodes, that doesn't help with coordinating dependencies between devices and other initializers (#23407), whether or not those are also updated to be based on dependencies rather than hard-coded level plus configurable priority values (which I'm concerned can produce hard-to-identify failures if the priorities are mis-assigned).

I have some ideas about how to do this but it involves either grotesque linker script hackery; a special link that produces data that is processed to produce a dependency tree that gets embedded into the application; or both.  I don't think it's a viable path.

Aside: Maybe we should fork off a separate area to discuss specifically how to deal with initialization ordering, as there are probably other things related to this overview issue that will be obscured if we discuss too much here.

pabigot, 2020-05-26:
> 7. Each device is not constant (same for their config part):
It was meant to be constant, but due to a hack on device's API setting (in case of init failure it is set to NULL at runtime) it is never constant, thus ends into RAM instead of staying into the ROM. See #24873

Based on experience from #25208 I don't believe it will be possible to switch to using `const struct device *` everywhere: too many APIs pass or store device pointers in `void *`, which requires introducing casts that remove the `const` qualifier.  Comments to focus on:
* [MISRA violation in `const` removal](https://github.com/zephyrproject-rtos/zephyr/pull/25208#issuecomment-627246669)
* [Dev-review discussion](https://github.com/zephyrproject-rtos/zephyr/pull/25208#issuecomment-628721074)
* [Evidence it can be done](https://github.com/zephyrproject-rtos/zephyr/pull/25208#issuecomment-628898045) if we decide it's important enough.  See the commits in the PR for examples of the massive changes required, some of which cannot be automated.

wentongwu, 2020-06-01:
> Providing device dependencies is not difficult: we already have them, we just haven't emitted them yet, perhaps because we haven't identified the canonical representation of a device handle. (The [...]

@pabigot if the dependencies are only about:  1. get the last level's parent(s) of given `struct device` which will be used in device runtime pm where the number of reference children will be counted at runtime; 2. construct ordered list where parent will be at the front of children(devices will be suspended following this list and resumed in the converse order for the case of devices following system states), does that will simplify the implementation?

pabigot, 2020-06-01:
> does that will simplify the implementation?

@wentongwu That's about what I'd assumed we'd do, though dependencies are static (I don't see why they would be dependent on power state).  The blockers are representation of device handles, and how to manage dependencies between devices and non-device functions.  The latter may need to be done manually in the driver source.

wentongwu, 2020-06-02:
@pabigot Thanks, but what's the meaning of the `representation of device handles`? And I see your comments https://github.com/zephyrproject-rtos/zephyr/pull/24514#issuecomment-637208934, also new version pm subsystem is expected in 2.4 cycle, but the first we should give a solution for the device dependency. And above I just give the two possible use cases about the device dependencies from pm perspective to see if the only two use cases instead of wanting the whole dependency graph/tree can simplify the implementation

carlescufi, 2020-06-16:
API meeting 16th June 2020

Two options proposed:

1. #25208 converts all `struct device *` parameters to `const struct device *`
2. #26127 moves all the constant data to a separate structure, and keeps `struct device` non-const

The issues with 1.:
- All calls to `device_get_binding()` calls would now return a const pointer, requiring a change in the implementation of many applications
- Device drivers that use `struct device *` to pass it as a context to threads or HAL functions would have to down-cast from `const struct device *` to `void *`
*or* store a pre-cast pointer to a non-const `struct device *`  
*or*
have some sort of:
```
struct dev {
  struct device *d;
};
```
with an instance of that somewhere.

@tbursztyka suggests using `const struct device *dev` everywhere but storing a pointer to the `struct device` in the `driver_data` structure for drivers that need it.
For dynamic allocation, knowing if the instance is between the _start and _end of the driver linker section would be enough, in case the device driver actually needs to know whether it's in RAM and ROM.
Device drivers should not need to know whether the `struct dev` instance is in RAM or flash, ever.

pabigot, 2020-06-16:
https://github.com/zephyrproject-rtos/zephyr/issues/26072#issuecomment-645006182 has a summary of the impact of #25208 versus #26127 and a requested comparison (with limitations discussed there).

tbursztyka, 2020-06-17:
Some precision from my side:

> The issues with 1.:
>   - All calls to device_get_binding() calls would now return a const pointer, requiring a change in the implementation of many applications

I don't think we nacked all the huge API reworks (gpio, adc, spi, ... and on going) we have got so far, on the fact it would require changes in the implementations of many applications. So to me, this is a false argument. Moreover in this case, it's really a small change.

>  - Device drivers that use struct device * to pass it as a context to threads or HAL functions

Again that's not a big change. Only a fractions of drivers would require that, not even mentioning threads. Same previous answer applies here as well.

About numbers given by @pabigot, the number of files being touched does not matter as soon as one solution is clearly better than the other one (in this case, const dev is really the one to pick). It would indeed matter if it was fully comparable.
The gain of RAM/ROM might not look tremendous, but that's already a relevant one. It certainly does not play any role in boards with 1Mb and more, but it does on all the ones that have below. 

Last but no least, part of the gain made by such rework, will be useful to improve provided features to drivers (such as #24511 for instance, which is not going to eat up all that gain since many drivers implement such solution on their own currently, it's just a matter of normalizing the behaviors under a central API).

pabigot, 2020-06-17:
I've rebased #25208 on the same master commit as #26127 to improve comparability.  Because #25208 is closed I can't push an update, and	I don't want to submit another one, so if you want to reproduce the results you'll need to pull specific commits from my repository.  The links and specific commits used are documented in the results.

See https://gist.github.com/pabigot/d63ddfa24a30e7a95e902fe986cbbd89 for the script and latest results.

While I do not dispute that transition to `const struct device*` provides the maximum savings in RAM, my concern remains the impact of this change on existing in-tree and out-of-tree code.  This is not a small change, as you can see if you look at what had to change in the samples to eliminate compiler warnings.

pabigot, 2020-06-18:
Another potential resolution proposed in the [2020-05-14 dev-review](https://github.com/zephyrproject-rtos/zephyr/pull/25208#issuecomment-628721074) is to put all the immutable data directly into `struct device`, but rely on the linker scripts to put the `.device_PRIO` section into non-volatile memory.

This would mean all existing API using `struct device *` would be unchanged, but we'd not use RAM to store the referenced objects.  Best of both worlds.

But it would also eliminate any chance of language-level detection when somebody writes through the pointer.

On many targets such a write would silently do nothing.  On some it might cause a bus fault.  On others where RODATA is stored in a memory technology that is not write-protected (e.g. FRAM) it could be catastrophic.

It's still an option.

tbursztyka, 2020-06-18:
>  rely on the linker scripts to put the .device_PRIO section into non-volatile memory.

That was always the plan so far, to force the devices instances to be in ROM (and not anymore in this RAMABLE_REGION, ROMABLE_REGION at it is right now).

> But it would also eliminate any chance of language-level detection when somebody writes through the pointer.

That's a bad idea, you really want to know at compile time if you are using the APIs and the objects as they are meant to be used.

carlescufi, 2020-06-18:
Dev Review 18th June:

Original reason for moving the immutable part of the device information to ROM:
- Memory savings in RAM. This RAM is to be reused for other features like common thread safety and internal/external synchronization.

Reasons for passing around `const struct device*` instead of `struct device *`:
- Statement of intentions: Device drivers and users of `struct device *` are not to modify the structure directly, Instead, a common device driver infrastructure with a defined API would allow device drivers to use the common mechanisms for all, without requiring drivers to provide their own.
- Statement of intentions: Most device instances in Zephyr are defined at compile time and stored in ROM, so this is more truthful to the majority of cases

-

pabigot, 2020-06-19:
I've updated https://gist.github.com/pabigot/d63ddfa24a30e7a95e902fe986cbbd89 with new results comparing the alternatives.  I'd missed the commit that changed the linker scripts to place the device list in ROM instead of RAM.  This was obscured because `size(1)` does not accurately capture the true memory usage.  Instead I've switched to using the memory region summary produced by the linker and displayed when things are built.

The overall conclusions do not change, but the numbers are more justifiable.

teburd, 2021-11-18:
As maybe related to #30105 if redoing the device model it would be a great addition to avoid the function pointer table altogether in many cases by perhaps using _Generic and driver specific types with a macro wrapper like i2c_transfer(somedevptr), which is about as good as we can get in C for static dispatch. It might  let a linker GC more function symbols and reduce the size of the elf along with other performance benefits.

gregshue, 2022-08-22:
One of the features sorely lacking from Zephyr is a controlled, reverse dependency order shutdown of the RTOS.  One use case for this is when shutting down a multi-threaded instance of MCUBoot and transitioning to another executable (which could even be different instance of MCUBoot running from RAM instead of Flash).  The hardware state during this handoff from one executable to another needs to be well controlled, and not necessarily inactive (e.g., some GPIO hardware or graphics display buffers may need to stay active, hardware watchdog may need to stay running).  For a scalable, composable solution, this will need to be registration-based, perhaps as an extension to SYS_INIT().

galak, 2022-08-23:
Adding reference to new discussion thread on metadata tracking - https://github.com/zephyrproject-rtos/zephyr/issues/49431

carlescufi, 2022-09-20:
API meeting:

- Requirement from @bjarki-trackunit:
  -  To be able to call multiple API classes using the same `struct device *` (or equivalent) in each call

bjarki-trackunit, 2022-09-24:
I have summarized the discussions and conclusions regarding the last months of discussion regarding MFDs and the device model. It is a 2-3 minute read approximately.