zephyrproject-rtos/zephyr #22942: Missing TX power options for nRF52833

pdunaj, 2020-02-19:
**Describe the bug**
TX power higher then +4 dBm in not available for nRF52833.

**To Reproduce**
See `subsys/bluetooth/controller/Kconfig` or try enabling +8 dBm on mentioned SoC.

**Expected behavior**
All power configurations supported by HW should be made available.

**Impact**
Reduced functionality.

**Screenshots or console output**
N/A

**Environment (please complete the following information):**
N/A

**Additional context**
N/A

aescolar, 2020-02-19:
CC @carlescufi

pdunaj, 2020-02-19:
Hi @carlescufi , maybe instead of adding nRF5233 in the `depend on` it would be better to add some characteristic to SoC family that support higher TX power options? Otherwise this will be edited more frequently.

carlescufi, 2020-12-07:
Fixed in https://github.com/zephyrproject-rtos/zephyr/commit/3b4ead139fc2

closed 2020-12-07 by carlescufi