zephyrproject-rtos/zephyr #22955: tests/kernel/interrupt fails on qemu_cortex_m0

andrewboie, 2020-02-19:
**Describe the bug**
```
starting test - test_nested_isr
isr0 running !!
isr0 execution completed !!

    Assertion failed at ZEPHYR_BASE/tests/kernel/interrupt/src/nested_irq.c:91: isr0: (new_val not equal to old_val)
Nested interrupt is not working

FAIL - test_nested_isr
```

I am baffled as to why CI is not failing due to this.

**To Reproduce**

I'm running

```
sanitycheck -T tests/kernel/interrupt/
```

**Expected behavior**
Test passes.

**Impact**
I don't know if this is a problem with the test or the Cortex-M0 implementation.

galak, 2020-02-19:
Interesting, ran locally and it passes just fine for me.  What's your build.log look like?

galak, 2020-02-19:
I did do: ``sanitycheck -T tests/kernel/interrupt/ -p qemu_cortex_m0``

andrewboie, 2020-02-19:
@galak double-checking my environment and trying the build on some other machines, will follow up shortly.

andrewboie, 2020-02-19:
This appears to be some kind of timing-sensitive race. On further testing, whether it passes or fails seems to be intermittent and depends on how loaded the host machine is.

It also looks like there is already some kind of issue with k_busy_wait() on this platform. The test has some ifdefs:

```
void isr0(void *param)
{
	ARG_UNUSED(param);
	printk("%s running !!\n", __func__);
#if defined(CONFIG_BOARD_QEMU_CORTEX_M0)
	/* QEMU Cortex-M0 timer emulation appears to not capturing the
	 * current time accurately, resulting in erroneous busy wait
	 * implementation.
	 *
	 * Work-around:
	 * Increase busy-loop duration to ensure the timer interrupt will fire
	 * during the busy loop waiting.
	 */
	k_busy_wait(MS_TO_US(1000));
#else
	k_busy_wait(MS_TO_US(10));
#endif
	printk("%s execution completed !!\n", __func__);
	zassert_equal(new_val, old_val, "Nested interrupt is not working\n");
}
```

andrewboie, 2020-02-19:
@galak if I run this test outside of sanitycheck, it fails often for me. I think this needs deeper investigation, especially since I cannot find an open bug on k_busy_wait() on qemu_cortex_m0

PR #26189 (2020-06-25): Boards qemu cortex m0 timer rework
QEMU Cortex-M0 Board includes a system timer driver implementation based on the nRF TIMER peripheral, because the RTC peripheral is not supported in QEMU.

In this PR we rework the timer driver, to reflect all recent changes made to the standard nrf RTC timer driver. Various issues with the driver are fixed - including a couple of hacks to make this work with QEMU.

As a result of this rework, we can now re-enable all tests that were currently disabled for cortex-m0 QEMU, namely,
- tests/kernel/common
-  tests/kernel/tickless
- tests/kernel/timer_api (this on [... cut]
closed 2020-06-25 by andrewboie