zephyrproject-rtos/zephyr #22956: nios2: k_busy_wait() never returns if called with interrupts locked

andrewboie, 2020-02-19:
**Describe the bug**
Calling irq_lock(), and then k_busy_wait(MS_TO_US(1000)) hangs indefinitely on this arch.

The problem is that in altera_avalon_timer_hal.c, z_timer_cycle_get_32() returns only accumulated_cycle_count updated by the timer_irq_handler()...which is of course masked.

There's a lengthy comment about this:

```
u32_t z_timer_cycle_get_32(void)
{
	/* Per the Altera Embedded IP Peripherals guide, you cannot
	 * use a timer instance for both the system clock and timestamps
	 * at the same time.
	 *
	 * Having this function return accumulated_cycle_count + get_snapshot()
	 * does not work reliably. It's possible for the current countdown
	 * to reset to the next interval before the timer interrupt is
	 * delivered (and accumulated cycle count gets updated). The result
	 * is an unlucky call to this function will appear to jump backward
	 * in time.
	 *
	 * To properly obtain timestamps, the CPU must be configured with
	 * a second timer peripheral instance that is configured to
	 * count down from some large initial 64-bit value. This
	 * is currently unimplemented.
	 */
	return accumulated_cycle_count;
}
```

**To Reproduce**
Run tests/kernel/interrupt on qemu_nios2.

stephanosio, 2020-03-17:
@andrewboie I am seeing the same problem on `qemu_cortex_r5` which uses `xlnx_psttc_timer`.

My question is: is it a _requirement_ for `z_timer_cycle_get_32` to return the correct cycle count even when interrupt is disabled?

If not, wouldn't it make more sense to fix the `tests/kernel/interrupt` test, which relies on this behaviour?

andrewboie, 2020-03-17:
> @andrewboie I am seeing the same problem on `qemu_cortex_r5` which uses `xlnx_psttc_timer`.
> 
> My question is: is it a _requirement_ for `z_timer_cycle_get_32` to return the correct cycle count even when interrupt is disabled?

It's bugged. The intent of k_busy_wait() is to allow for maximal latency control for stuff like drivers, and the expectation is that it can operate in ISRs or with interrupts locked. 

> If not, wouldn't it make more sense to fix the `tests/kernel/interrupt` test, which relies on this behaviour?

Nope.

stephanosio, 2020-03-18:
> It's bugged. The intent of k_busy_wait() is to allow for maximal latency control for stuff like drivers, and the expectation is that it can operate in ISRs or with interrupts locked.

@andrewboie Fair enough, I think it would be helpful to mention that in the docs if it hasn't been mentioned already.

carlescufi, 2021-02-09:
@nashif do we want to keep this open? Is the nios2 arch maintained?

stephanosio, 2021-08-12:
This should not have been closed.

stephanosio, 2021-12-28:
Not stale

nashif, 2022-02-22:
Changing into an enhancement given the description above, a new timer implementation is needed here, the current timer implementation works as documented.

closed 2025-05-16 by nashif