zephyrproject-rtos/zephyr #22964: Define a consistent naming convention for device tree defines

galak, 2020-02-20:
We have a mish-mash of defines today and there are places that we can have possible naming conflicts.  So define a consistent name space for the various node defines that we generate.

Related to #21369

galak, 2020-02-20:
```
# For props not 'regs', 'interrupts', 'clocks', SPI 
DT_PATH_P_<PATH>_<PROP>:
				
				string	"foo"
				int		3
				boolean	1/0 (always defined)
				array	{ 1, 2, 3, 4 }
				uint8-array { 0x12, 0x34, 0x45 }
				string-array { "foo", "bar"}
				phandle	<PATH> to referenced node
				
				phandles - nothing
				phandles-array - nothing
				compound - nothing

/* bool */
#define DT_PATH_P_SOC_FLASH_CONTROLLER_40020000_FLASH_0_PARTITIONS_PARTITION_0_READ_ONLY 1
/* int */
#define DT_PATH_P_SOC_I2C_40066000_CLOCK_FREQUENCY                      100000
/* string */
#define DT_PATH_P_SOC_I2C_40066000_LABEL                                "I2C_0"
/* array */
#define DT_PATH_P_SOC_USART_4000C400_LOCATION_RX                        {4, 0, 0}
/* uint8-array */
#define DT_PATH_P_SOC_SPI_40004000_SSD16XXFB_0_LUT_INITIAL              {0x22, 0x55, 0xaa, 0x55}
/* string-array */
#define DT_PATH_P_SOC_RTC_4003D000_INTERRUPT_NAMES                      {"alarm", "seconds" }
/* phandle */
#define DT_PATH_P_SOC_GPIO_400FF080_NXP_KINETIS_PORT                    SOC_PINMUX_4004B000

				
regs:	
DT_PATH_X_REG_NUM_<n>_ADDRESS_<path>
DT_PATH_X_REG_NUM_<n>_SIZE_<path>
DT_PATH_X_REG_NAME_<name>_ADDRESS_<path>
DT_PATH_X_REG_NAME_<name>_SIZE_<path>
DT_PATH_X_REG_CNT_<path>

/* regs */

#define DT_PATH_X_REG_NUM_0_SOC_INTERRUPT_CONTROLLER_C000000_ADDRESS            0xc000000
#define DT_PATH_X_REG_NUM_0_SOC_INTERRUPT_CONTROLLER_C000000_SIZE               8192
#define DT_PATH_X_REG_NAME_PRIO_SOC_INTERRUPT_CONTROLLER_C000000_ADDRESS        0xc000000
#define DT_PATH_X_REG_NAME_PRIO_SOC_INTERRUPT_CONTROLLER_C000000_SIZE           8192
 
#define DT_PATH_X_REG_NUM_1_SOC_INTERRUPT_CONTROLLER_C000000_ADDRESS            0xc002000
#define DT_PATH_X_REG_NUM_1_SOC_INTERRUPT_CONTROLLER_C000000_SIZE               2088960
#define DT_PATH_X_REG_NAME_IRQ_EN_SOC_INTERRUPT_CONTROLLER_C000000_ADDRESS      0xc002000
#define DT_PATH_X_REG_NAME_IRQ_EN_SOC_INTERRUPT_CONTROLLER_C000000_SIZE         2088960
 
#define DT_PATH_X_REG_CNT_SOC_INTERRUPT_CONTROLLER_C000000                      3

interrupts:
DT_PATH_X_INTERRUPTS_NUM_<n>_ADDRESS_<path>
DT_PATH_X_INTERRUPTS_NUM_<n>_SIZE_<path>
DT_PATH_X_INTERRUPTS_NAME_<name>_ADDRESS_<path>
DT_PATH_X_INTERRUPTS_NAME_<name>_SIZE_<path>
DT_PATH_X_INTERRUPTS_CNT_<path>

phandles:
DT_PATH_X_PHANDLES_<n>_<path>_<prop>   <path>

/* phandles */
#define DT_PATH_X_PHANDLES_0_SOC_GPIO_400FF080_NXP_KINETIS_PORTS                SOC_PINMUX_4004B000
#define DT_PATH_X_PHANDLES_1_SOC_GPIO_400FF080_NXP_KINETIS_PORTS                SOC_PINMUX_4004C000

phandle-array:
DT_PATH_X_PHA_PHANDLE_<n>_<path>_<prop>
DT_PATH_X_PHA_<name from binding for cell>_<n>_<path>_<prop>

/* phandle array */
DT_PATH_X_PHA_PHANDLE_0_SOC_I2C_40066000_FXOS8700_1D_INT1_GPIOS	
DT_PATH_X_PHA_PIN_0_SOC_I2C_40066000_FXOS8700_1D_INT1_GPIOS
DT_PATH_X_PHA_FLAGS_0_SOC_I2C_40066000_FXOS8700_1D_INT1_GPIOS

array:
DT_PATH_X_ARRAY_NUM_<n>_<path>_<prop>
DT_PATH_X_ARRAY_CNT_<path>_<prop>

/* array */
#define DT_PATH_X_ARRAY_NUM_0_SOC_USART_4000C400_LOCATION_RX	4
#define DT_PATH_X_ARRAY_NUM_1_SOC_USART_4000C400_LOCATION_RX	0
#define DT_PATH_X_ARRAY_NUM_2_SOC_USART_4000C400_LOCATION_RX	0
#define DT_PATH_X_ARRAY_CNT_SOC_USART_4000C400_LOCATION_RX		3
```

galak, 2020-02-20:
Starts of my initial cut at a naming convention that is consistent and has limited conflicts.

galak, 2020-02-20:
Thinking for things like DT_ALIAS, DT_IDENT, DT_NODELABEL would just be 'PATH' defines as such:

```
#define DT_ALIAS_UART0 SOC_UART_4006D000
#define DT_NODELABEL_UART0 SOC_UART_4006D000
#define DT_INST_0_NXP_KINETIS_UART SOC_UART_4006D000
```

Than you can use macro magic to do something like:

``DT_ALIAS_PROP(UART0, LABEL) -> DT_PATH_P_SOC_UART_4006D000_LABEL``

pabigot, 2020-02-20:
I was hoping for something like this (draft status)
```
<name> ::= 'DT_' <node-id> [ '_' <suffix> ];

<init-id> ::= [A-Z]
<id-char> ::= [A-Z0-9]
<decimal> ::= [0-9]+ ;
<hexadecimal> ::= [0-9a-f]+ ;

<identifier> ::= <init-id> <id-char>* ;
<long-identifier> ::= <identifier> ( '_' <identifier> )* ;

<node-name> ::= <long-identifier> ;
<node-label> ::= <long-identifier> ;
<unit-address> ::= hexadecimal [ '_' <unit-address> ] ;

<property-name> ::= <long-identifier> ;

# An encoded vendor,model string from a "compatible" property
<compatible> ::= <long-identifier> ;

# A sequence meant to identify a specific node in the devicetree.
# Used as a prefix for node properties and to represent the node
# itself.
<node-id> ::= <node-alias> | <node-compat> | <node-inst> | <node-label> | <node-path> ;

# This name is used to reference a generic capability such as an LED
# or button.
<node-alias> ::= 'ALIAS_' + <alias> ;

# A property from the /aliases node
<alias> ::= <property-name>

# This name is used in driver instantiation
<node-inst> ::= 'INST_' <decimal> '_' <compatible> ;

# This name is used to reference by SOC-specific identifier
<node-label> ::= 'NODELABEL_' <long-identifier> ;

# This name is used to reference by full devicetree path
<node-path> ::= 'PATH_' <path> ;
<path> ::= ; # **TBD**

# Some old thing we probably want to get rid of.
<node-legacy> ::= [ <bus> '_' ] <compatible> '_' <unit_address> ;
<bus> ::= <long-identifier> ;

<suffix> ::= <property-suffix> ;

# Used to identify a property value (or value component)
<property-suffix> ::= <property-name> [ <ordinal> ] ;

```

galak, 2020-02-20:
Example of what we generate today for a register w/current PR from marti:

```
#define DT_PATH_SOC_UART_4006A000_BASE_ADDRESS                          0x4006a000
#define DT_PATH_SOC_UART_4006A000_SIZE                                  4096
#define DT_NXP_KINETIS_UART_4006A000_BASE_ADDRESS                       DT_PATH_SOC_UART_4006A000_BASE_ADDRESS
#define DT_NXP_KINETIS_UART_4006A000_SIZE                               DT_PATH_SOC_UART_4006A000_SIZE
#define DT_INST_0_NXP_KINETIS_UART_BASE_ADDRESS                         DT_PATH_SOC_UART_4006A000_BASE_ADDRESS
#define DT_INST_0_NXP_KINETIS_UART_SIZE                                 DT_PATH_SOC_UART_4006A000_SIZE
#define DT_NODELABEL_UART0_BASE_ADDRESS                                 DT_PATH_SOC_UART_4006A000_BASE_ADDRESS
#define DT_NODELABEL_UART0_SIZE                                         DT_PATH_SOC_UART_4006A000_SIZE
```

galak, 2020-02-20:
Based on what Peter had here's some examples of what PATH defines might look like using `_p_`:
```
#define DT_PATH_SOC_INTERRUPT_CONTROLLER_C000000_p_ADDRESS_idx_0
#define DT_PATH_SOC_INTERRUPT_CONTROLLER_C000000_p_ADDRESS_idx_1
#define DT_PATH_SOC_INTERRUPT_CONTROLLER_C000000_p_ADDRESS_name_PRIO
#define DT_PATH_SOC_INTERRUPT_CONTROLLER_C000000_p_ADDRESS_name_IRQ_EN
#define DT_PATH_SOC_INTERRUPT_CONTROLLER_C000000_p_ADDRESS_cnt_IRQ_EN

any array (string-array, uint8-array, array, phandles):

#define DT_PATH_SOC_GPIO_400FF080_p_NXP_KINETIS_PORTS_idx_0
#define DT_PATH_SOC_GPIO_400FF080_p_NXP_KINETIS_PORTS_idx_1
#define DT_PATH_SOC_GPIO_400FF080_p_NXP_KINETIS_PORTS_cnt

for phandle-array:

DT_PATH_SOC_I2C_40066000_FXOS8700_1D_p_INT1_GPIOS_idx_0		/* phandle */
DT_PATH_SOC_I2C_40066000_FXOS8700_1D_p_INT1_GPIOS_idx_0_pin
DT_PATH_SOC_I2C_40066000_FXOS8700_1D_p_INT1_GPIOS_idx_0_flags
DT_PATH_SOC_I2C_40066000_FXOS8700_1D_p_INT1_GPIOS_idx_cnt
```

mbolivar, 2020-02-21:
> I was hoping for something like this (draft status)
> <name> ::= 'DT_' <node-id> [ '_' <suffix> ];

As discussed in #22966, an ABNF for what we have now is here: https://github.com/mbolivar/zephyr/blob/doc-dt-improvements/doc/guides/dts/macros.bnf

This will hopefully be merged to master soon.

mbolivar, 2020-02-26:
@pabigot and any others interested in discussing.

I want to start with the grammar as defined in v2.2.0-rc2, and tackle `node-id` (used by `property-macro`) first as a warmup for the process:

https://github.com/zephyrproject-rtos/zephyr/blob/v2.2.0-rc2/doc/guides/dts/macros.bnf

I've created a [dt-macros-ng branch](https://github.com/mbolivar/zephyr/tree/dt-macros-ng) in my fork, which contains this commit at time of writing with proposed `node-id` changes that keep the old names around, but introduce new (hopefully better) ones as well:

https://github.com/mbolivar/zephyr/commit/3a7dc57f38afa4155f5f7a9cc4e4fdf777574a53

Using these conveniently would require some macro magic à la @galak.

Comment welcome. Here are some notes on collisions in the meantime:

mbolivar, 2020-02-27:
> Would have the following "path" prefixes (in the respective order they
> are defined in the DTS):
> 
>     * `DT_n_path_s_FOO_s_BAR_BAZ`
> 
>     * `DT_n_path_s_FOO_s_BAR_s_BAZ`
> 
>     * `DT_n_path_s_FOO_BAR_BAZ`

I went ahead and implemented some helper macros to make this style readable, and did the same for the other node identifiers. Examples are in the commit log for this patch:

https://github.com/mbolivar/zephyr/commit/019e18d5caad01549512065cc23135c77fc0aa4a

In particular:

```
Add helper macros which abstract the "true names" of each of the four
types of node identifier we intend to support:

- DT_ALIAS()
- DT_INST()
- DT_NODELABEL()
- DT_PATH()

Add a DT_PROP() macro which can be used to read the value of a
devicetree property given a node identifier from one of these four
other macros, and the uppercase-and-underscore name of the property.

Example macro expansions:

    DT_PROP(DT_ALIAS(UART0), CURRENT_SPEED)  ->
    DT_n_alias_UART0_p_CURRENT_SPEED

    DT_PROP(DT_INST(0, NORDIC_NRF_UART), CURRENT_SPEED) ->
    DT_n_inst_0_NORDIC_NRF_UART_p_CURRENT_SPEED

    DT_PROP(DT_NODELABEL(ARDUINO_SERIAL), CURRENT_SPEED) ->
    DT_n_nodelabel_ARDUINO_SERIAL_p_CURRENT_SPEED

    DT_PROP(DT_PATH(SOC, UART_40002000), CURRENT_SPEED) ->
    DT_n_path_sSOC_sUART_40002000_p_CURRENT_SPEED
```

As can be seen above, I think the basic form of a property macro can be `DT_n_NODEXYZ_p_PROPFOO`. We can keep the old macros around while we migrate to this new style.

I've looked at the other types of macros we currently generate and think this use of mixed case in the actual macros generated by scripts, along with helper macros that take arguments in devicetree.h to access them, is a workable and future-proof way forward.

One additional benefit is we can change the "true names" without big tree-wide problems as long as people use the helper API.

@erwango @mnkp @pabigot: I would like to discuss this approach tomorrow at the dev meeting. If it seems worth chasing, I will make the changes to gen_defines.py and handle the other special case macros as well.

We can use that to prepare to close this issue, along with #22554 and #22555. Of course we won't be able to really merge to topic-devicetree until @galak is back and can review, but it would be good to make progress meanwhile, and this proposal is along the same lines as his PR #22987.

erwango, 2020-02-28:
@mbolivar @pabigot, @mnkp, I've tried to go a little bit further in STM32 driver conversion to DT_INST macros. I've came across a few interesting points. As expected some driver will require DT_NODELABEL macros, but not always for the expected reasons.
You can have a check on my summary here: https://github.com/zephyrproject-rtos/zephyr/pull/22807#issuecomment-592481101.
I don't think there is anything blocking or that put in question what has been proposed above, but this could illustrate some points and show new examples that we didn't mentioned already.

ulfalizer, 2020-03-03:
Is this stuff that needs to be backwards compatible?

Otherwise, I think we should separate parts with two underscores. The macros we generate now are super hard to read (and pretty much unguessable if you don't already know how it works), since it's hard to see where parts start and end.

This is common in the Linux kernel too. Try this:

```sh
$ git grep '#define.*\w__'
```

Could have

```c
DT__ST_STM32_I2C_V2__40005C00__BASE_ADDRESS
```

instead of

```c
DT_ST_STM32_I2C_V2_40005C00_BASE_ADDRESS
```

for example.

Personally, that would've helped me enormously.

mbolivar, 2020-03-04:
> Otherwise, I think we should separate parts with two underscores.

We've been over this idea a couple of times before, and unfortunately we cannot do that, because @pabigot pointed out that double underscores are reserved in C++.

closed 2020-03-24 by galak