zephyrproject-rtos/zephyr #22965: 4 byte addressing in spi_nor driver for memory larger than 128Mb.

siltemc, 2020-02-20:
**Is your enhancement proposal related to a problem? Please describe.**
SPI memory devices with larger than 128Mb space require 4 byte addressing. The current spi_nor driver does not support 4 byte addressing. Some spi memory require a specific op code to be to switch from 3 byte to 4 byte. For example https://www.digikey.no/short/zqqwqz is a 1Gb nor memory which you need to send the code 0xB7 to enter 4 byte addressing and 0xE9 to exit. These op codes seem common on several devices but I am not sure if they are the standard code. 

**Describe the solution you'd like**
As suggested by @pabigot in slack:
The Nordic QSPI driver has a flag for 32 bit size addresses `address-size-32` in the devicetree. A similar flag for the standard spi_nor driver is probably the best way to specify the device for 32 bit addresses, which would require the driver to be updated.

**Describe alternatives you've considered**
I suggested also possibly adding to the driver the ability to switch and change between 3 byte and 4 byte as you need in your application. But I think this will require more work than making the above suggestion.

**Additional context**
Slack post in #general channel:
> silt  16 hours ago
I have a question about the basic spi_nor driver. Will it work with 4 byte addressing? If in spi_nor_write the addr you pass is a 4 byte address there will be no error? Also some memory devices need to enter 4 byte address mode through a command before the work with 4 byte addresses.
For example https://www.digikey.no/short/zqqwqz  is a 1Gb memory, this memory requires you to enter it into 4 byte addressing mode to be able to access the addresses outside the first 128Mb. It also looks like the spi nor driver op codes work with 4 byte addressing and this memory. But I don't see anywhere in the zephyr spi nor driver for zephyr to send a jedec op code to change to 4 byte address mode. This memory uses op code 0xB7 for enter and 0xE9 for exit. I am not sure if these are standard op codes but I have seen them repeated on different memories. I am interested in using this memory for a project.
Second question is it possible to add into a function like spi_nor_configure which if you flag you want 4 byte it sets it on, or add a function to enter/exit 4 byte address mode so you can change as you want? Then add a flash opcode into spi_nor.h? Would this be difficult to do?

> pabigot  14 hours ago
AFAIK there is no support for 4-byte addressing in the existing driver.  The Nordic QSPI driver has support for an address-size-32 flag in devicetree.   A flag like that for the standard driver is likely to be the best way to specify the device for 32-bit addresses; the driver would also have to be updated.

> pabigot  14 hours ago
I suggest starting by opening an issue with the material you have here, then if you wish create a pull request to add the support.

pabigot, 2021-03-03:
@sltemc if this is still of interest to you please see #32844

madscientist159, 2021-04-05:
Tested on Kestrel (POWER-based soft BMC) with a Micron N25Q512A.  Seems to function correctly in 4BA mode, and detects 4BA mode automatically (`CONFIG_SPI_NOR_SFDP_RUNTIME=y`).

PR #32844 (2021-04-27): drivers: spi-nor: support 4-byte addresses for flags with >128 MiBi address space
Add support for entering and using 4-byte addressing mode based on information derived from SFDP headers or devicetree.

Only the top four commits of this PR are relevant to this PR; the remainder are from #32680.

Fixes #22965
closed 2021-04-27 by carlescufi