zephyrproject-rtos/zephyr #22974: Add cancel function to onoff service

nordic-krch, 2020-02-20:
**Is your feature request related to a problem? Please describe.**

Currently, onoff service does not handle releasing when in transition from off to on. This makes it difficult to handle premature release scenario. There is an `onoff_cancel()` but for the last client it will return `-EWOULDBLOCK` and won't cancel the operation. Handling that case is not trivial when callback is used as notification and currently the best way to serve it is to have special case in request callback to call release from there.
There are and will be services which would be able to provide abort/cancel functionality to abort transition from off to on. One particular case is nordic clock where clock can be stopped also while  starting.

**Describe the solution you'd like**

Proposed solution:
Add cancel function to the structure with service functions. In case of last client and presence of cancel function call it. If cancel function returns 0, notify client with `-ECANCELED` error code.
Once this is added fail safe release procedure for service supporting cancel would look like:
```
err = onoff_release(srv, cli);
if (err == -EBUSY) {
  /* in transition to on, cannot release, try cancel */
  err = onoff_cancel(srv, cli);
  if (err == -EALREADY) {
    /* transition completed before cancel executed */
    err = onoff_release(srv, cli);
  }
}
```

**Describe alternatives you've considered**

Currently, without cancellation functionality release with premature support is more complex in case of nordic clock driver (immediate stop, asynchronous start) and looks something like:

```
static void started_cb(struct onoff_service *srv, struct onoff_client *cli,
		      void *user_data, int res)
{
  bool *release_requested = user_data;
  if (release_requested) {
    /* if flag is set it means that release was requested before transition to on completed */
    onoff_client_init_spinwait(&cli); /* assuming off is instant */
    err = onoff_request(srv, &cli);
    __ASSERT_NO_MSG(err >= 0);
   return;
  }
  /* handle started case */
}

/**************** function requesting and prematurely releasing ********/
/* requesting  */
bool release_req = false;
onoff_client_init_callback(cli, started_cb, &release_req);
err = onofff_request(srv, cli);

/* releasing using same client */
key = irq_lock();
/* Check if starting operation is already finished. */
if (onoff_client_fetch_result(cli, &err) < 0) {
    /* operation not finished yet so request shall be done from released callback */
    release_req = true;
} else {
   /* it's finished, we can release here */
    onoff_client_init_spinwait(cli);
}
irq_unlock(key);
if (!release_req) {
  err = onoff_release(srv, cli);
}
```

Solution is quite complex and additionally results in clock being stopped later (after being started) and not immediately, even tough hardware is capable of doing that.

pabigot, 2020-02-20:
I see some possibility for this, but we need #22419 merged before I can start on it.

pabigot, 2020-02-25:
Before going further I want to highlight that the design of this manager intentionally **requires** that there is always at least one client left listening to be notified of the completion of any in-progress transition.  This was done to be sure somebody's notified if the transition fails, placing the service into an error state.  Also, it is an error to initiate a service release that is not paired with a previously successful service request.

From the second point this proposed user code is incorrect, and suggests a possible misuse of the manager API:

```
err = onoff_release(srv, cli); /* A */
if (err == -EBUSY) {
  /* in transition to on, cannot release, try cancel */
  err = onoff_cancel(srv, cli);  /* B */
  if (err == -EALREADY) {
    /* transition completed before cancel executed */
    err = onoff_release(srv, cli);
  }
}
```

At point `A` a client structure is being passed to configure notification for the release.  This client structure **must not** be in active use by another operation.  Consequently you can't reference that structure at `B`.  Rather you would have to pass to `onoff_cancel()` the client structure that was used in the unresolved prior call to `onoff_request()`.

So let's take a step back and clarify the use case.  As I understand it, a module has invoked `onoff_request()` for a client, then wants to invoke `onoff_release()` for the same client before the original request completes.

This is very unusual: Zephyr doesn't generally support canceling in-progress operations.  The model for this service expects that the state machine always completes a transition before starting a new one. Breaking that expectation is not something to be done lightly.

But let's assume it's necessary and consider whether it can be handled.

First, you have to be able to detect this, which needs you need to synchronize with the notification point for `onoff_request()`.  The most natural way to check would be to start with `onoff_cancel()` passing the in-progress request client object.  This would produce `-EWOULDBLOCK` if the request is still in progress with no other active requesters; in all other cases something else is wrong or you can safely invoke `onoff_release()` which will not return `-EBUSY`.

Because `-EWOULDBLOCK` is stale by the time it gets back to the caller (the operation may have completed) the enhancement would have to be to extend `onoff_cancel()` with the ability to ask the service to return to the source state.  This has to be done under lock to be safe from asynchronous completion of the request.  But no other service transition is invoked under manager lock.  And the return transition must be synchronous because the only available notification mechanism is bound to the request operation.  But the return-to-source-state transition might itself put the service into an error state, and there's no way to distinguish that failure (which leaves the service in an error state) from a failure of the cancel operation (which doesn't).

That's just the start of the messiness.  The alternative code (which is not under manager lock) fails to handle interaction with other clients that issue requests that have not been cancelled, so it's not usable as a model for a solution: decisions (and transition abort) have to be made under the manager lock to be safe, and that might conflict with the steps necessary to complete the cancellation.

All this suggests the cancel-in-progress-transition solution is generally fragile, and users of the on-off manager should follow the expectation of waiting for a transition to complete before doing something else with the service.

I don't think this is something that can be done, though I'll keep it in mind.

nordic-krch, 2020-02-26:
> This is very unusual

I have following case: USB plug in triggers HF clock request, plug out release. I have not control when those events come. They might be in the interrupt context so request is asynchronous.

Note that without that feature it is a step back in functionality for clock control as previously clock could be stopped at any moment and onoff without cancellation makes it very complex.
  
I've prepared #23087 with proposed solution.

pabigot, 2020-02-26:
I'll take a look at that PR.  Where is your current implementation using the API as it currently exists?

nordic-krch, 2020-02-26:
Here is how i solve it without this functionality:
https://github.com/nordic-krch/zephyr/blob/clock_control_calib_onoff/drivers/usb/device/usb_dc_nrfx.c#L543

I use additional flag passed to the started callback for that. If it is set then it means that clock should be released in the callback:
```
static void hfxo_rdy_cb(struct onoff_service *srv, struct onoff_client *cli,
				void *user_data, int res)
{
	if (*(bool *)user_data) {
		int err;
		/* Meanwhile there was a request to stop. */
		onoff_client_init_spinwait(cli);
		err = onoff_release(srv, cli);
		__ASSERT_NO_MSG(err >= 0);
	}
}
```

Then stop function is starting by checking in lock if client completed. If not it means that starting did not complete so i set the flag to release it in the callback. If client completed then we can release it here. I assume here that onoff_release for clock is synchronous.

```
static int hfxo_stop(struct nrf_usbd_ctx *ctx)
{
	int err = 0;
	int key = irq_lock();

	if (onoff_client_fetch_result(&ctx->hfxo_cli, &err) < 0) {
		ctx->hfxo_do_release = true;
	}

	irq_unlock(key);

	if (ctx->hfxo_do_release == false) {
		onoff_client_init_spinwait(&ctx->hfxo_cli);
		err = onoff_release(ctx->hfxo_srv, &ctx->hfxo_cli);
	}

	return err;
}
```
It is a bit complex and it does not cover scenario like request followed by release before completion and followed by request before completion (not sure if it's realistic).

pabigot, 2020-06-29:
@nordic-krch Do you still need this, or has the subsequent rework of the onoff service satisfied your use cases?

closed 2020-06-30 by nordic-krch