zephyrproject-rtos/zephyr #22980: bluetooth: logging: Build assertion prevents immediate logging when using legacy ll

paulgavrikov, 2020-02-20:
**Describe the bug**
Setting `CONFIG_LOG_IMMEDIATE ` and `CONFIG_BT_LL_SW_LEGACY` aborts builds even if Bluetooth logging is disabled.

**Expected behavior**
Immediate logging has to be disabled for Bluetooth legacy ll logging, but it should be possible to use the option if Bluetooth logging is disabled.

**Impact**
annoyance

**Environment (please complete the following information):**
 - OS: (e.g. Linux, MacOS, Windows): Kubuntu 18.04
 - Toolchain (e.g Zephyr SDK, ...): Zephyr SDK 1.130
 - Commit SHA or Version used: cd9b940756c1c5f9f09311c1b889a47853430a4c

**Additional context**
As discussed with @carlescufi

carlescufi, 2020-02-20:
@paulgavrikov can you please attach the `build/zephyr/.config` file?

paulgavrikov, 2020-02-27:
[config.txt](https://github.com/zephyrproject-rtos/zephyr/files/4261161/config.txt)

cvinayak, 2020-03-01:
No immediate logging in or outside Bluetooth subsystem can be used, as it introduces large interrupt latencies.

nordic-krch, 2020-03-02:
@cvinayak yes, that will lead to failures but there might be cases when user still wants to use it. For example when debugging some crash at startup then switching to immediate logging may be welcomed. What about adding config option to bluetooth to disable this check with note that user should be aware of consequences.

cvinayak, 2020-03-03:
@nordic-krch The reporter can disable using Bluetooth controller if he is debugging startup. With Bluetooth state/role active and immediate logging, I do not think the reporter can reliably use his application.

closed 2020-07-30 by github-actions[bot]