zephyrproject-rtos/zephyr #22982: (PTS)Bluetooth: GATT/SR/GAS/BV-01-C,  GATT/SR/GAS/BV-07-C - BTP Error

agata-ponitka, 2020-02-20:
Describe the bug
After running hdl_wid_132 gatts_add_svc returns error.  

To Reproduce
run auto-pts

[GATT_SR_GAS_BV-01-C.zip](https://github.com/zephyrproject-rtos/zephyr/files/4231493/GATT_SR_GAS_BV-01-C.zip)
[GATT_SR_GAS_BV-07-C.zip](https://github.com/zephyrproject-rtos/zephyr/files/4231494/GATT_SR_GAS_BV-07-C.zip)

carlescufi, 2020-02-27:
#23137 Fixes `GATT/SR/GAS/BV-01-C`, but `GATT/SR/GAS/BV-07-C` still fails, likely due to an issue in BTP itself (see log below).
In both tests the auto-pts client seems to be trying to register the **same service** multiple times.

```
Traceback (most recent call last):
  File "/home/carles/src/auto-pts/ptsprojects/zephyr/gatt_wid.py", line 38, in gatt_wid_hdl
    handler = getattr(module, "hdl_wid_%d" % wid)
AttributeError: 'module' object has no attribute 'hdl_wid_137'
```

sjanc, 2020-02-28:
yeap, I think this issue can be closed
autopts work will be handled in https://github.com/intel/auto-pts/issues/366

aescolar, 2020-02-28:
Closing then (if anybody disagrees, please reopen)

closed 2020-02-28 by aescolar