zephyrproject-rtos/zephyr #22985: Check if Zephyr is affected by SweynTooth vulnerabilities

ceolin, 2020-02-20:
Researchers from Singapore University of Technology and Design published a list of 12 vulnerabilities towards seven Bluetooth SoC vendors.
The report can be read at https://asset-group.github.io/disclosures/sweyntooth/

Zephyr is not mentioned but they say that the list of affected SoC is not exhaustive, so we would be nice checking it.

As part of their report, there is a proof-of-concept to exploit these 12 vulnerabilities, it is based on Nordic nRF52840 Dongle and can be found in https://github.com/Matheus-Garbelini/sweyntooth_bluetooth_low_energy_attacks

jhedberg, 2020-02-20:
@carlescufi @cvinayak FYI

carlescufi, 2020-02-24:
This has been checked and no vulnerabilities have been found that affect Zephyr.

closed 2020-02-24 by carlescufi