zephyrproject-rtos/zephyr #23002: unknown type name 'class'

jasonedn, 2020-02-21:
When the following simple class is called in application, ir reported "unknown type name 'class'".

#pragma once;

In test.hpp
class ZrTest
{
    public:
    ZrTest();

    static void print();
 };

In test.cpp
ZrTest::ZrTest()
{}
void ZrTest::print()
{
    printk("print from ZrTest instance\r\n");
}

In main()
void main()
{
   ZrTest::print();
}

The CPP is enabled in prj.conf
CONFIG_CPLUSPLUS=y
CONFIG_NEWLIB_LIBC=y

Does anybody have idea on the issue?

stephanosio, 2020-02-21:
Probably because your `main` is inside a `.c` file and you included `test.hpp` from there.

jasonedn, 2020-02-21:
@stephanosio, I tried to rename main.c to main.cpp and test.hpp is also included, but it didn't fix it. Thanks

stephanosio, 2020-02-22:
> @stephanosio, I tried to rename main.c to main.cpp and test.hpp is also included, but it didn't fix it. Thanks

@jasonedn Please create a test branch and push it to your fork so that we can try reproducing the problem.

jasonedn, 2020-02-22:
The main.c was changed to main.cpp, but test.cpp was not added to
CMakeLists.txt. Now it worked when tect.cpp was added as below.
   target_sources(app PRIVATE src/main.cpp src/test.cpp)

Thanks a lot

closed 2020-02-23 by stephanosio