zephyrproject-rtos/zephyr #23004: Can't use west to flash test images into up_squared board.

chen-png, 2020-02-21:
**Describe the bug**
when using west flash to download image into up_squared, it met an error "KeyError: 'misc-flasher', 
it is that for up_squared, in the runner_config args list,  **the misc-flasher is NONE**, so when invoking "runner_config['args'][runner_name]", it will report an error.

**To Reproduce**
Steps to reproduce the behavior:
1. west build -b up_squared -p auto samples/hello_world/
2. west flash
3. See error

**Screenshots or console output**
[ztest@zserver12 zephyr]$ west flash ~/up_squared.sh
-- west flash: rebuilding
ninja: no work to do.
-- west flash: using runner misc-flasher
Traceback (most recent call last):
  File "/home/ztest/.local/bin/west", line 11, in <module>
    sys.exit(main())
  File "/home/ztest/.local/lib/python3.6/site-packages/west/main.py", line 583, in main
    args.handler(args, unknown)
  File "/home/ztest/.local/lib/python3.6/site-packages/west/main.py", line 334, in ext_command_handler
    command.run(args, unknown, topdir, manifest=manifest)
  File "/home/ztest/.local/lib/python3.6/site-packages/west/commands/__init__.py", line 118, in run
    self.do_run(args, unknown)
  File "/home/ztest/work/zephyrproject/zephyr/scripts/west_commands/flash.py", line 30, in do_run
    do_run_common(self, my_args, runner_args)
  File "/home/ztest/work/zephyrproject/zephyr/scripts/west_commands/run_common.py", line 170, in do_run_common
    **runner_config['args'][runner_name] +
KeyError: 'misc-flasher'**

**Environment (please complete the following information):**
 - OS: fedora28
 - Toolchain: zephyr-sdk-0.11.1
 - Commit ID: 82e006660

chen-png, 2020-02-21:
@mbolivar Could you take a look of this? it's related to commit" eb95bed55".

mbolivar, 2020-02-21:
Thanks for the report. I believe https://github.com/zephyrproject-rtos/zephyr/pull/23027 is a fix.

PR #23027 (2020-02-24): cmake: flash: three runners.yaml fixes
There are three problems with the way runners.yaml is being created:

1. The dictionary which contains the arguments for each runner is
   using the runner's name converted to a C identifier instead of the
   runner's name itself. That causes west flash to fail when the two
   are different, e.g. for 'misc-flasher' (runner name), which is
   different than 'misc_flasher' (runner name as C identifier)

2. We need to make sure that the dictionary key maps to an empty list
   if there are no arguments, which normally doesn't happen since the
   runner usuall [... cut]
commit e4479e2f (2020-02-24): cmake: flash: three runners.yaml fixes

There are two problems with the way runners.yaml is being created:

1. The dictionary which contains the arguments for each runner is
   using the runner's name converted to a C identifier instead of the
   runner's name itself. That causes west flash to fail when the two
   are different, e.g. for 'misc-flasher' (runner name), which is
   different than 'misc_flasher' (runner name as C identifier)

2. We need to make sure that the dictionary key maps to an empty list
   if there are no arguments, which normally doesn't happen since the
   runner usually [... cut]
closed 2020-02-24 by jhedberg