zephyrproject-rtos/zephyr #23019: logging: add LOG_RAW macro

koffes, 2020-02-21:
The logging system provides many useful features. 

In our system we get prints on the format: <timestamp><wrn_lvl><module><user msg>
This is fine most of the time, but sometimes we want to use ANSI escape characters for cursor positioning. With the current system we cannot e.g. save the cursor position in a desirable manner because <timestamp><wrn_lvl><module> has been printed before the command reaches the terminal.

Suggestion: Add a LOG_RAW macro which uses all the features of the logging system, but omits the timestamp, level and module.

@nordic-krch

carlescufi, 2020-02-21:
This could be useful for the Bluetooth monitor as well. CC @jhedberg

jhedberg, 2020-02-21:
In the case of the monitor protocol (btsnoop) it's all binary, whereas I think the suggestion here is for LOG_RAW to take a string.

koffes, 2020-02-24:
@jhedberg that is correct. May the existing feature LOG_HEXDUMP be close to what you need?
Perhaps LOG_BARE_RAW and LOG_BARE_STR be more suitable?

nordic-krch, 2020-02-24:
what level would that use. Do we implicitly assume something (e.g. info) or create `LOG_RAW_INF`?

koffes, 2020-02-24:
Since this feature would not use either color or any "<wrn>", "<err>" etc. label, do we need to define a level?