zephyrproject-rtos/zephyr #23021: logging: Sync timestamp between cores

koffes, 2020-02-21:
The logging system has a useful feature where a timestamp is prepended each log message.

This works well on single core MCU's, but in some situations you want to investigate multi-core applications. E.g. both cores crash, and you need to figure ut what took place first.
Suggest adding a config of type CONFIG_LOG_TIMESTAMP_SYNC. This will ensure that e.g. a LOG_DBG which is called on each core at the same point in time gets the same timestamp.

I realise that this is may well be far from trivial, as well as platform specific.
@nordic-krch

carlescufi, 2020-04-07:
This should probably be generalized to synchronization between cores of several data items, including timestamps.

koffes, 2020-08-17:
@nordic-krch: Is this something that is feasible? We often come across issues where figuring out which log message comes first is essential.

koffes, 2021-06-10:
Test

nordic-krch, 2021-09-13:
@koffes I've prepared PR which adds mechanism for synchronization. It's nrf53 specific. See #38495

PR #38495 (2022-01-19): soc: arm: nordic_nrf: nrf53: Add Synchronized RTC
The goal of this PR is to establish a mechanism which allows to get the offset between system clock on network and application core (assuming they both run from RTC). Offset can be applied to a function which returns timestamp for logging allowing logs synchronization from both cores (currently on independent outputs).

Add a module which is responsible for getting offset between RTCs
used for system clock on NET and APP cores. After getting an offset
between NET and APP clocks, it can be used for logging timestamping
on NET core to ensure that [... cut]
closed 2022-01-19 by carlescufi