zephyrproject-rtos/zephyr #23025: Make west aware of when "west update" may be needed

jhedberg, 2020-02-21:
It'd be nice if west tracked the last time "west update" was run and compared that against the last modification timestamp of west.yml. That way west could either automatically run "west update" or issue a suggestion to the user that "west update" may be necessary.

mbolivar, 2020-02-21:
I'm making this a Zephyr issue instead of a west issue based on the above discussion.

tejlmand, 2020-02-25:
A small thought on this.
We could consider to use CMake config package for this,

The config package doesn't have to be installed, as we can use the path provided by west list to look up package location.

Only risk I see on this is when project depends on a pre-release version, as example:
`find_package(nrfxlib 2.4.0)`, then all `2.4.99` will also match, even though `2.4.99` could be very different between two versions.

tejlmand, 2020-02-25:
Of course having n-number of modules will raise the chance that one of them is out-of-date and then user can be warned.

SebastianBoe, 2020-03-26:
I disagree that we should introduce versioning of every Zephyr
module. The paperwork needed to maintain the correct version will
be neglected, resulting in corrupted version checking.

We should solve this problem using data that already exists,
e.g. git revisions and/or file timestamps as they are not subject
to human-error neglect.

It's not clear to me where exactly the check should exist, it
could be in CMake. But for instance if we did timestamp checking
then we still need support in west to touch a file that records
the most recent "west update" invocation. So probably changes in
both west and CMake.

These checks will need to be suppressable because of course they
will not be perfect. No solution other than banning use of git
and requiring all version control to go through a single entry
point can be perfect.

mbolivar-nordic, 2020-03-27:
> The paperwork needed to maintain the correct version will
> be neglected, resulting in corrupted version checking.

It seems like this could be part of module pull request CI to me. What am I missing?

tejlmand, 2020-03-30:
Just did some more thinking on @SebastianBoe comment:
> Usually, my workflow is like this:
>
> ```
> git checkout new_zephyr_revision
> build
> # Get weird error
> west update
> build
> # See error is fixed
> ```

which basically means that the best we can do:
```
git checkout new_zephyr_revision
build
# CMake warning / error message: Zephyr project is outdated, please run `west update`
west update
build
# See error is fixed
```

so the best we can get is a faster fail with a better error message.

This also means the earliest place we can do this check, is in CMake, cause `west` was never invoked during update of manifest repo.

mbolivar-nordic, 2021-10-14:
> a mere warning "Maybe you should west update?" whenever `west` finds that the last-known-remote-state (= `manifest-rev`) is older than the timestamp of the manifest file does not seem complicated. Is it?

Not sure, want to try it and send a PR depending on the results?

Imports and non-SHA revisions tend to make everything trickier than they first appear if you're just considering the standard zephyr/west.yml style manifest full of SHAs. I encourage you to give it a try and see what you find.

marc-hb, 2021-10-15:
> > Of course it should perform the obvious optimization of not warning for SHA1s or tags in the manifests that still match manifest-rev.

> Imports and non-SHA revisions tend to make everything trickier than they first appear if you're just considering the standard zephyr/west.yml style manifest full of SHAs

You're right, these make the "obvious match" above complicated. This match is actually mandatory and not just "an optimization": if it were optional then the warning would appear all the time for slow moving repos.

I had a look at `git reflog --date=iso manifest-rev` and short of "spamming it" on every `west update`, this `reflog` is a very bad indicator of "the last time west update was run". So maybe "the last time west update was run" should just be an empty, per repo file touched by `west update` every time it runs; KISS. Resolving imports to find which manifest file applies to each repo would still be a problem :-/

> Not sure, want to try it and send a PR depending on the results?

Nope! Thanks for your patience and correcting me :-/

PS @0Grit: you asked in https://github.com/zephyrproject-rtos/west/issues/548 why submodule versions are embedded in the git history and not in plain text files / manifest. I think reducing such mismatches is one possible reason/advantage. Git submodules don't support branches either.

PPS: this sub-thread is not Zephyr-specific.

0Grit, 2021-10-15:
Surely another project out there has solved these problems?

I had always thought Buckaroo addressed this somehow.

https://github.com/LoopPerfect/buckaroo/wiki/Git-as-a-Package-Registry

Highly recommend looking at that tool for inspiration.

mbolivar-nordic, 2021-11-04:
Can you clear up how buck solves this problem? I read the link but it seems to say "solve the problem by using buck", which isn't helpful.

0Grit, 2021-11-05:
https://buck.build/concept/build_rule.html

https://buckaroo.pm/blog/approaches-to-c-dependency-managment

https://buckaroo.pm/blog/build-systems-should-use-hashes

https://buckaroo.pm/blog/lessons-learned-from-porting-300-c-libraries