zephyrproject-rtos/zephyr #23026: missing ISR locking in UART driver?

ZideChen0, 2020-02-21:
**Describe the bug**

When IO APIC working in fixed delivery mode to broadcast interrupts to all CPUs in SMP system, seems the UART may not work properly.

Running this app and will see that the console is not able to echo keyboard input.
$ west build -b qemu_x86_64 samples/subsys/console/echo/ && west build -t run

**To Reproduce**
Steps to reproduce the behavior:
1. Apply these commits:.
https://github.com/zephyrproject-rtos/zephyr/pull/22848/commits/3a447e65a68d59581dede00d402b2e534aac417f
https://github.com/zephyrproject-rtos/zephyr/pull/22848/commits/3ee41f80b55ebe766e2627286f270efc50ea878c

2. replace IOAPIC_LOWEST with IOAPIC_FIXED in drivers/interrupt_controller/intc_ioapic.c to make sure IO APIC is working in logical + fixed mode.

3. Run the test and it won't work.
$ west build -b qemu_x86_64 samples/subsys/console/echo/ && west build -t run

**Expected behavior**

Skip (2) and use IOAPIC_LOWEST to deliver IO APIC to one CPU only, and run the same test, the console will print whatever characters you input from keyboard.

This problem could be caused by UART driver doesn't have ISR locking, so that it will have problems if same interrupt is delivered to all CPUs.

nashif, 2020-03-06:
@jenmwms  can you please take this one, you have been dealing with this driver and the missing locking most likely was one of the main issues you have faced while working on the configure API patch

jenmwms, 2020-03-06:
@nashif Thanks! I'll take a look and update here.

jenmwms, 2020-03-06:
I can replicate the issue described - still investigating root cause. 

**Actual behavior (for qemu_x86_64 as reported):**
-- west build: build configuration:
       source directory: /home/jennife2/zephyrproject/zephyr/samples/subsys/console/echo
       build directory: /home/jennife2/zephyrproject/zephyr/build
       BOARD: qemu_x86_64 (origin: CMakeCache.txt)
-- west build: running target run
[3/4] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu64,+x2apic
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.01H:ECX.x2apic [bit 21]
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200214_052440-f7294c49af13-zephyr
)
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 8
*** Booting Zephyr OS build zephyr-v2.1.0-2617-g93d3a67905a1  ***
You should see another line with instructions below. If not,
the (interrupt-driven) console device doesn't work as expected:

**Expected behavior (for example, qemu_x86):**
-- west build: build configuration:
       source directory: /home/jennife2/zephyrproject/zephyr/samples/subsys/console/echo
       build directory: /home/jennife2/zephyrproject/zephyr/build
       BOARD: qemu_x86 (origin: CMakeCache.txt)
-- west build: running target run
[0/1] To exit from QEMU enter: 'CTRL+a, x'[QEMU] CPU: qemu32,+nx,+pae
SeaBIOS (version rel-1.12.1-0-ga5cab58-dirty-20200214_052440-f7294c49af13-zephyr
)
Booting from ROM..Optimal CONFIG_X86_MMU_PAGE_POOL_PAGES 7
*** Booting Zephyr OS build zephyr-v2.1.0-2617-g93d3a67905a1  ***
You should see another line with instructions below. If not,
the (interrupt-driven) console device doesn't work as expected:
Start typing characters to see them echoed back
FDAFDA

jenmwms, 2020-03-19:
@ZideChen0 Will this issue be resolved by your PR #23325?
From that description, it sounds like the patch yields successfully running samples/subsys/console/echo (test UART driver) on qemu_x86_64 (xapic, SP and SMP).

ZideChen0, 2020-03-20:
> @ZideChen0 Will this issue be resolved by your PR #23325?
> From that description, it sounds like the patch yields successfully running samples/subsys/console/echo (test UART driver) on qemu_x86_64 (xapic, SP and SMP).

Yes, with PR #23325, UART interrupt is supposed to be delivered to one CPU, so no ISR locking is needed.

jenmwms, 2020-03-23:
@ZideChen0 I will try your PR, too, but saw that it does not touch the uart driver itself. So I created a Draft PR of changes that might be needed to the uart driver in addition. But I still have the issue described here with my patch #23026 (WIP) so I can get feedback as it relates to this issue.

ZideChen0, 2020-03-23:
We are taking different approaches to address this issue. In my patches, I configured UART interrupts as lowest priority delivery mode, so it's delivered to one CPU only, and no need to modify UART driver. For example: https://github.com/zephyrproject-rtos/zephyr/pull/23325/commits/ab9d571aa7bb56a1891c93ed769fa421c39988cd#diff-5e7cae5167b76f9e0a04173995540696R65

In your PR, I think you are trying to add the spinlock, so even if same interrupt is delivered to multiple CPUs, it still can be correctly processed.

It seems to me, in the following cases, the spinlock doesn't have a chance to be released, thus the subsequent calls fail to grab the spinlock. In this particular use case (console/echo app), it seems driver/console/uart_console.c is a better place to add the spinlock, but this could impact all other UART drivers, and other UART use cases, like bluetooth is not taken care of ... so I don't know what is the best way to patch the UART driver.

https://github.com/jenmwms/zephyr/commit/7e87fc50254eca9de847a863cd84d4871312b100#diff-6f446bf9585fa3268d292c461074ae00R738

https://github.com/jenmwms/zephyr/commit/7e87fc50254eca9de847a863cd84d4871312b100#diff-6f446bf9585fa3268d292c461074ae00R690

jenmwms, 2020-03-23:
@ZideChen0 Thanks for clarifying the different approaches! Maybe we can bring this up at the API/dev meeting this week and get closure on these.

jenmwms, 2020-03-24:
> > the patch yields successfully running samples/subsys/console/echo (test UART driver) on qemu_x86_64 (xapic, SP and SMP).
> 
> Yes, with PR #23325, UART interrupt is supposed to be delivered to one CPU, so no ISR locking is needed.

I tried the PR #23325 and it resolves the bug behavior in the sample application described in the issue here. However, the other aspect of this issue reported here is to implement the missing ISR locking mechanism in the UART driver. So I will keep working on that part in #23693. I might use #23325 to confirm the application works with the spinlocks.

nashif, 2020-05-16:
raising priority, this is not blocking #25319

PR #23693 (2020-05-18): drivers: serial: uart_ns16550: add missing isr locking
ISR locking was missing in the UART driver for x86_64. This PR adds a spinlock mechanism to interrupt related services in the driver.

Fixes #23026
commit d2c74eb9 (2020-05-18): drivers: serial: uart: ns16550 add missing isr locking

The existing uart driver ns16550 did not have ISR locking that
effected IO APIC working in fixed delivery mode in SMP system
x86_64. This commit adds ISR locking mechanism using spinlock
for the interrupt related services.

The CONFIG_IPM_CONSOLE_STACK_SIZE is increased to lift
limitation of stack size experienced in IPM driver test with
this spinlock impelentation.

Fixes #23026
closed 2020-05-18 by nashif