zephyrproject-rtos/zephyr #23035: dhcpv4_client sample not working on sam e70

md-jamal, 2020-02-22:
Hi All,
I flashed dhcpv4_client on SAM E70 and it is not getting any ip address from DHCP Server which is my router.

```
[00:00:00.000,000] <wrn> soc: CIDR mismatch: chip = 0xa1020e01 vs HAL = 0xa1020e00
[00:00:00.000,000] <inf> i2c_sam_twihs: Device I2C_0 initialized
[00:00:00.001,000] <inf> eth_sam: MAC: fc:c2:3d:0b:fc:c7
[00:00:00.001,000] <inf> eth_sam: Queue 0 activated
[00:00:00.001,000] <inf> eth_sam: Queue 1 set to idle
[00:00:00.001,000] <inf> eth_sam: Queue 2 set to idle
[00:00:00.001,000] <inf> eth_sam_phy: Soft Reset of ETH PHY
[00:00:00.091,000] <inf> eth_sam_phy: PHYID: 0x221561 at addr: 0
[00:00:00.091,000] <dbg> eth_sam_phy.phy_sam_gmac_auto_negotiate: Starting ETH PHY auto-negotiate sequence
uart:~$ *** Booting Zephyr OS build zephyr-v2.1.0-1382-gcf89ba33eae9  ***
[00:00:02.215,000] <dbg> eth_sam_phy.phy_sam_gmac_auto_negotiate: PHY auto-negotiate sequence completed
[00:00:02.225,000] <inf> eth_sam_phy: common abilities: speed 100 Mb, full duplex
[00:00:02.225,000] <dbg> net_dhcpv4.net_dhcpv4_init: (0x20400d10): 
[00:00:02.231,000] <inf> net_dhcpv4_client_sample: Run dhcpv4 client
[00:00:02.231,000] <dbg> net_dhcpv4.net_dhcpv4_start: (0x20400d10): iface 0x20407600 state=init
[00:00:02.231,000] <dbg> net_dhcpv4.net_dhcpv4_start: (0x20400d10): wait timeout=1s
[00:00:03.231,000] <dbg> net_dhcpv4.dhcph4_manage_timers: (0x20400dc4): iface 0x20407600 state=init
[00:00:03.231,000] <dbg> net_dhcpv4.dhcpv4_enter_selecting: (0x20400dc4): enter state=selecting
[00:00:03.231,000] <dbg> net_dhcpv4.dhcpv4_send_discover: (0x20400dc4): send discover xid=0x5df19e0a timeout=4s
[00:00:03.231,000] <dbg> net_dhcpv4.dhcpv4_timeout: (0x20400dc4): Waiting for 4s
[00:00:03.231,000] <dbg> eth_sam.eth_tx: ETH tx
[00:00:07.231,000] <dbg> net_dhcpv4.dhcph4_manage_timers: (0x20400dc4): iface 0x20407600 state=selecting
[00:00:07.231,000] <dbg> net_dhcpv4.dhcpv4_send_discover: (0x20400dc4): send discover xid=0x5df19e0b timeout=8s
[00:00:07.231,000] <dbg> net_dhcpv4.dhcpv4_timeout: (0x20400dc4): Waiting for 8s
[00:00:15.231,000] <dbg> net_dhcpv4.dhcpv4_send_discover: (0x20400dc4): send discover xid=0x5df19e0c timeout=16s
[00:00:15.231,000] <dbg> net_dhcpv4.dhcpv4_timeout: (0x20400dc4): Waiting for 16s
[00:00:31.231,000] <dbg> net_dhcpv4.dhcph4_manage_timers: (0x20400dc4): iface 0x20407600 state=selecting
[00:00:31.231,000] <dbg> net_dhcpv4.dhcpv4_send_discover: (0x20400dc4): send discover xid=0x5df19e0d timeout=32s
[00:00:31.231,000] <dbg> net_dhcpv4.dhcpv4_timeout: (0x20400dc4): Waiting for 32s
[00:00:31.658,000] <dbg> eth_sam.queue0_isr: GMAC_ISR=0x00000002
[00:00:31.658,000] <dbg> eth_sam.queue0_isr: rx.w1=0x00004000, tail=0
[00:00:31.658,000] <dbg> eth_sam.frame_get: Frame complete: rx=0x204043a0, tail=2
[00:00:31.658,000] <dbg> eth_sam.eth_rx: ETH rx
[00:00:36.211,000] <dbg> eth_sam.queue0_isr: GMAC_ISR=0x00000002
[00:00:36.211,000] <dbg> eth_sam.queue0_isr: rx.w1=0x00004000, tail=2
[00:00:36.211,000] <dbg> eth_sam.frame_get: Frame complete: rx=0x204043a0, tail=4
[00:00:36.211,000] <dbg> eth_sam.eth_rx: ETH rx
[00:00:37.212,000] <dbg> eth_sam.queue0_isr: GMAC_ISR=0x00000002
[00:00:37.212,000] <dbg> eth_sam.queue0_isr: rx.w1=0x00004000, tail=4
[00:00:37.212,000] <dbg> eth_sam.frame_get: Frame complete: rx=0x204043a0, tail=6
[00:00:37.212,000] <dbg> eth_sam.eth_rx: ETH rx
[00:00:38.212,000] <dbg> eth_sam.queue0_isr: GMAC_ISR=0x00000002
[00:00:38.212,000] <dbg> eth_sam.queue0_isr: rx.w1=0x00004000, tail=6
[00:00:38.212,000] <dbg> eth_sam.frame_get: Frame complete: rx=0x204043a0, tail=8
[00:00:38.212,000] <dbg> eth_sam.eth_rx: ETH rx
[00:00:39.213,000] <dbg> eth_sam.queue0_isr: GMAC_ISR=0x00000002
[00:00:39.213,000] <dbg> eth_sam.queue0_isr: rx.w1=0x00004000, tail=8
[00:00:39.213,000] <dbg> eth_sam.frame_get: Frame complete: rx=0x204043a0, tail=10
[00:00:39.213,000] <dbg> eth_sam.eth_rx: ETH rx

```
It is only in discover stage and not coming out of it. Can anyone provide input on this, will be helpful

stephanosio, 2020-02-22:
This is a duplicate of #22890. 

It seems you are testing with the Rev. B chip and #12711 is required to get Ethernet working.

md-jamal, 2020-02-22:
Thanks so much saved my day.
After adding the commit and switching the revision to B it worked
`CONFIG_SOC_PART_NUMBER_SAME70Q21B=y`

closed 2020-02-22 by md-jamal