zephyrproject-rtos/zephyr #23052: nrf52840_pca10056: Spurious RTS pulse with hardware flow control disabled

carlescufi, 2020-02-24:
When running `samples/hello_world` with the default configuration on `nrf52840_pca10056`, `CONFIG_UART_0_NRF_FLOW_CONTROL` is disabled.

This means that the RTS line, which is an output from the nRF52840 perspective, should stay idle continuously. Instead, we see a pulse which then make the J-Link OB firmware believe that hardware flow control is enabled.

- [Segger J-Link wiki documentation](https://wiki.segger.com/J-Link_OB#Hardware_flow_control_support)
- [Nordic UART specification](https://infocenter.nordicsemi.com/topic/ps_nrf52840/uart.html?cp=4_0_0_5_32_1#concept_dly_2sw_sr)
- Logic analyzer capture:
[image]

carlescufi, 2020-08-04:
@anangl did we fix this at all?

anangl, 2022-02-14:
There is no bug here, actually. This is how the UART and UARTE peripherals work. As the specification [says](https://infocenter.nordicsemi.com/topic/ps_nrf52840/uarte.html?cp=4_0_0_5_33_4#concept_xsw_mtw_sr):

> If flow control is not enabled, the interface will behave as if the CTS and RTS lines are kept active all the time.

The driver initially configures the GPIO pin that is to serve as the RTS line to be an output with the high (inactive) level. Shortly after that, this pin number is written into the PSEL.RTS register of the peripheral and the peripheral is enabled. From this moment the peripheral starts driving the pin, and if the hardware flow control is disabled, the pin immediately becomes active. That is why the RTS pulse appears.
I don't think it is worth trying to align with JLink OB firmware expectations regarding the RTS line, so that it could perform its auto-detection of the hardware flow control, as this would mean quite a complication of the driver code (the RTS pin would need to be disconnected from the peripheral when the flow control is not enabled). Hence, I'm closing this issue.

closed 2022-02-14 by anangl