zephyrproject-rtos/zephyr #23063: CMSIS v2 osThreadJoin does not work if thread exits with osThreadExit

raidoz, 2020-02-25:
**Describe the bug**
Joining a CMSIS v2 thread that exited with osThreadExit returns osErrorResource. The join itself actually waits for the thread to exit, however `has_joined` is only set in zephyr_thread_wrapper and osThreadExit terminates the thread with k_thread_abort. Joining works, if the thread exits with a return, however this is counter to the RTOS v2 documentation and not allowed on other operating systems. Fix would seem to be to either also set the has_joined in osThreadExit or possibly redefining osThreadExit as return since the thread is already exiting voluntarily?

**Expected behavior**
According to RTOS v2 documentation: `The function osThreadExit terminates the calling thread. This allows the thread to be synchronized with osThreadJoin.`

andrewboie, 2020-02-26:
this shouldn't be hard to fix once #23028 goes in

andrewboie, 2020-05-06:
Spent some time looking at this. Integrating `k_thread_join()` into this scheme has issues due to Zephyr not having any concept of "detached" threads.

I'm using the following as a reference https://www.keil.com/pack/doc/CMSIS/RTOS2/html/group__CMSIS__RTOS__ThreadMgmt.html

About k_thread_join(): If some thread A calls `osThreadJoin()` on a joinable thread B, and while A is waiting, B calls `osThreadDetach()` on itself. The current behavior in Zephyr's CMSISv2 implementation is for `osThreadJoin()` called by A to return with `osErrorResource`. I think this is correct as the spec for `osThreadJoin()` documents its return values "`osErrorResource`: the thread is in an invalid state (ex: not joinable)." But if we use Zephyr's `k_thread_join()` implementation, I don't see a good way to wake up the caller early if the target thread detaches, since the OS itself doesn't know anything about CMSIS detached threads.

Now, it's not un-heard of for specific CMSIS implementations to have caveats. For example, FreeRTOS documents https://arm-software.github.io/CMSIS-FreeRTOS/General/html/tech_data.html. We could consider doing the same for Zephyr and just say that if you are blocked joining on a thread, and the thread detaches, the join operation will continue until that thread detaches.

Something else I noticed. In the spec, it says "The function `osThreadJoin` waits for the thread specified by thread_id to terminate. If that thread has already terminated, then `osThreadJoin` returns immediately." But there's a check `_is_thread_cmsis_inactive()` which returns `osErrorResource` if the check is true. This might also be a bug. I will add a test for this.

Finally, there's a test `test_thread_joinable_terminate()`. AFAICT, it enforces that if thread A calls `osThreadJoin()` on a joinable thread B, and later thread B is killed with `osThreadTerminate()`, that A's call with `osThreadJoin()` should return with `osErrorResource`. This doesn't make any sense to me, I would expect it would join normally. Indeed, the spec says "`osThreadTerminate` destroys non-joinable threads and removes their thread_id from the system. Subsequent access to the thread_id (for example `osThreadGetState`) will return an `osThreadError`. Joinable threads will not be destroyed and return the status `osThreadTerminated` until they are joined with `osThreadJoin`." So I'm thinking a bug here too.

> Fix would seem to be to also set the has_joined in osThreadExit.

Yeah, to me this looks like the shortest path to success.

I do not have much experience with CMSIS and would value suggestions/opinions.

closed 2021-01-12 by github-actions[bot]

coskunergan, 2023-07-19:
Has there been any progress on this issue, I get the same problem when I switched from a cmsis v2 with RTX migration to FreeRTOS?