zephyrproject-rtos/zephyr #23107: Convert SAM SoC drivers to DT_INST

nandojve, 2020-02-26:
**Is your enhancement proposal related to a problem? Please describe.**

This is to keep tracking of drivers to DT_INST related to Atmel SoCs. The worked started with #22806.

**Describe the solution you'd like**

Move/Create DT_INST for below driver list:

drivers/adc/adc_sam0.c
drivers/adc/adc_sam_afec.c
drivers/counter/counter_sam0_tc32.c
drivers/entropy/entropy_sam.c #22806
drivers/ethernet/eth_sam_gmac.c #23045
drivers/flash/flash_sam.c
drivers/gpio/gpio_sam0.c
drivers/gpio/gpio_sam.c
drivers/hwinfo/hwinfo_sam.c #22947
drivers/i2c/i2c_sam0.c
drivers/i2c/i2c_sam_twi.c
drivers/i2c/i2c_sam_twihs.c
drivers/pinmux/pinmux_sam0.c
drivers/serial/Kconfig.sam0
drivers/serial/uart_sam0.c
drivers/serial/uart_sam.c
drivers/serial/usart_sam.c
drivers/spi/spi_sam0.c
drivers/spi/spi_sam.c
drivers/usb/device/usb_dc_sam.c
drivers/watchdog/wdt_sam.c #22806

nandojve, 2020-02-28:
@galak

nandojve, 2020-03-05:
@carlescufi @benpicco @stephanosio

nandojve, 2020-03-06:
@jhedberg

galak, 2020-03-31:
@nandojve are you able to get back to look at conversions for SAM now that we have the new DT macro's in place.  Here's what I have (the # is count of DT_ references that need modification, or a rough estimate).

```
     10 drivers/adc/adc_sam_afec.c
      2 drivers/flash/flash_sam.c
     35 drivers/gpio/gpio_sam.c
      2 drivers/hwinfo/hwinfo_sam.c
     12 drivers/i2c/i2c_sam_twi.c
     18 drivers/i2c/i2c_sam_twihs.c
     35 drivers/serial/uart_sam.c
     24 drivers/serial/usart_sam.c
      5 drivers/spi/spi_sam.c
     28 drivers/usb/device/usb_dc_sam.c

     13 drivers/adc/adc_sam0.c
      9 drivers/counter/counter_sam0_tc32.c
     12 drivers/gpio/gpio_sam0.c
     35 drivers/i2c/i2c_sam0.c
     12 drivers/pinmux/pinmux_sam0.c
     46 drivers/serial/uart_sam0.c
     43 drivers/spi/spi_sam0.c
```

nandojve, 2020-03-31:
Hi @galak, I can take look at weekend and try update full list. I saw tons of DT_INST moving and probably lots of SAM related.

galak, 2020-04-01:
> Hi @galak, I can take look at weekend and try update full list. I saw tons of DT_INST moving and probably lots of SAM related.

Thanks.  All the DT_INST conversions should be complete.  This work now would be moving away from something like `DT_ATMEL_SAM0_SPI_...` to `DT_INST` and `DT_NODELABEL` (nodelabel form used for cases like pinmux.c that you need to know the specific instance).  The idea being that the node label in dts will match the SoC docs naming.

nandojve, 2020-04-25:
@galak I believe is almost done with Atmel SAM/SAM0.

Drivers converted however not define the global DT_DRV_COMPAT:
drivers/hwinfo/hwinfo_sam.c

Drivers without DT support:
drivers/dma/dma_sam_xdmac.c
drivers/i2s/i2s_sam_ssc.c

galak, 2020-05-08:
> @galak I believe is almost done with Atmel SAM/SAM0.
> 
> Drivers converted however not define the global DT_DRV_COMPAT:
> drivers/hwinfo/hwinfo_sam.c

I'm not sure it makes sense in this case, but ok if you want to submit a PR for it.

> Drivers without DT support:
> drivers/dma/dma_sam_xdmac.c
> drivers/i2s/i2s_sam_ssc.c

Done.

nandojve, 2020-05-10:
Atmel SoC DT_INST migration was concluded.

closed 2020-05-10 by nandojve